Xinyuan-LilyGO/LilyGo-Modem-Series #110: +CIPEVENT: NETWORK CLOSED UNEXPECTEDLY

PooruTorie, 2024-06-03:
Hi i am trying to create a tcp connection with an iot sim card but after NETOPEN
i get a NETWORK CLOSED UNEXPECTEDLY error

```cpp
if (!modem.restart(GSM_PIN)) {
  delay(10000);
  ESP.restart();
  return;
}

if (!modem.waitForNetwork()) {
  delay(10000);
  ESP.restart();
  return;
}

modem.gprsConnect(APN_HOST);

client.connect("postman-echo.com", 80);

client.println("GET https://postman-echo.com/ip HTTP/1.0");

client.readString();

client.stop();
```
```
AT+NETCLOSE

+NETCLOSE: 2

ERROR
AT+CGDCONT=1,"IP","iot.1nce.net","0.0.0.0",0,0

OK
AT+CIPMODE=0

OK
AT+CIPSENDMODE=0

OK
AT+CIPCCFG=10,0,0,0,1,0,75000

OK
AT+CIPTIMEOUT=75000,15000,15000

OK
AT+CGACT=1,1

OK
AT+NETOPEN

OK

+NETOPEN: 0

AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","postman-echo.com",80

OK

+CGEV: NW PDN DEACT 1

+CGEV: ME DETACH

+CGEV: EPS PDN ACT 1

+CIPOPEN: 0,0

+CIPEVENT: NETWORK CLOSED UNEXPECTEDLY

AT+NETOPEN?

+NETOPEN: 0

OK

PB DONE
AT+NETCLOSE

+NETCLOSE: 2

ERROR
AT+CIPSEND=0,40

+CIPERROR: 4

ERROR
AT+CIPSEND=0,2

+CIPERROR: 4

ERROR
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
```

felmue, 2024-06-04:
Hello @PooruTorie 

hmm, I am not seeing that. E.g. after opening the TCP connection with `AT+CIPOPEN` it stays open and only closes automatically after about 60 seconds. (But no unexpected close.) I used an A7600C1 modem, 1nce sim card and tried to connect to the same server.

Have you tried to open a TCP connection to a different server?

Thanks
Felix

PooruTorie, 2024-06-04:
I now have a working version build from a other example

now i need to find the difference to my code

```cpp
SerialMon.begin(115200);
delay(10);

pinMode(PWR_PIN, OUTPUT);
digitalWrite(PWR_PIN, HIGH);
delay(3000);
digitalWrite(PWR_PIN, LOW);

DBG("Wait...");

delay(2000);

SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

DBG("Initializing modem...");
if (!modem.init(GSM_PIN)) {
  DBG("Failed to restart modem, delaying 10s and retrying");
  delay(10000);
  ESP.restart();
  return;
}

modem.setNetworkMode(MODEM_NETWORK_AUTO);
modem.waitResponse(10000L);

DBG("Waiting for network...");
if (!modem.waitForNetwork()) {
  delay(10000);
  return;
}

DBG("Connecting to", apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
  delay(10000);
  return;
}

TinyGsmClient client(modem, 0);

client.connect("postman-echo.com", 80);

client.println("GET https://postman-echo.com/ip HTTP/1.0");

delay(10000);

String data = client.readString();
DBG(data);

client.stop();
```

PooruTorie, 2024-06-04:
I am now using the newest version of TinyGSM and the TINY_GSM_MODEM_SIM7600 setup it occasional gives the error but it works most of the time

PooruTorie, 2024-06-23:
It now works all the time but with the SIM7600 Modem and the newest tinygsm version

closed 2024-06-23 by PooruTorie