Xinyuan-LilyGO/LilyGo-Modem-Series #111: SD card mount failed

br4593, 2024-06-12:
Hi,

I inserted an SD card to the built in card slot of the T-7670G but when I run the SDCard example from the official github library (https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/), I get an "Card Mount Failed" error.

I have tried two different SD cards, neither one works:

San Disk 32GB SDHC with the following format:

[image]

San Disk 64GB SDXC with the following format:
[image]

I am running with the official SD card example with no change to the code, so I assume the SPI pinout is correct.
Any ideas?

lewisxhe, 2024-07-09:
Is the board selected correctly? Is the board selected LILYGO_T_A7670?

lewisxhe, 2024-07-09:
[image]

br4593, 2024-07-21:
> ![image](https://private-user-images.githubusercontent.com/22990954/346980217-158bcb35-1c68-4d60-988a-b1abc7cddc29.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE1NDY0NDUsIm5iZiI6MTcyMTU0NjE0NSwicGF0aCI6Ii8yMjk5MDk1NC8zNDY5ODAyMTctMTU4YmNiMzUtMWM2OC00ZDYwLTk4OGEtYjFhYmM3Y2RkYzI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIxVDA3MTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYwOTMwNGJlMmVhN2IwYTgxODU2ZDlhNDZiN2EyYTExNjI2MTQwZDAwY2MwM2M5Y2UwZDViYjhiZGE2OWI2ZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.paMNOmcj1yxF8-XnblIiDg58p50eclLE9cOyg0La5zw)

Yep. Here is how it looks like:

[image]

lewisxhe, 2024-08-13:
If replacing the SD card does not work, then it is certain that there is a problem with the card slot. Contact customer service for after-sales service.

closed 2024-09-10 by lewisxhe