Xinyuan-LilyGO/LilyGo-Modem-Series #114: GPSShield Example + A7670E will not recognize GPS Module

dashDotter, 2024-07-08:
Hello there,

I got the A7670E Module with the optional GPS Shield (backside shows etecl25t6a-n3-v1). I tried to use the GPSShield example. The module is plugged into the GNSS antenna slot on the board.

I tried the baud rates 115200 and 9600 as specified in GPSShield.ino:
` SerialGPS.begin(9600, SERIAL_8N1, BOARD_GPS_RX_PIN, BOARD_GPS_TX_PIN);`
Unfortunatley, `if (SerialGPS.available()) {
        Serial.println("Serial GPS available");
    }` 
does not show up when monitoring the serial output(baud 115200 as well as 9600). Therefore `displayInfo()` will not be executed. 
Might there be a problem with the example, or should I get some specific GPS Antenna? Right now, its not possible for me to test it with an other antenna.

lewisxhe, 2024-07-09:
The GPSShield example is only available with A7670G + GPDShield, as noted above the sketch.
Please check the example descriptions in the README to see which examples are available

lewisxhe, 2024-07-09:
[image]

dashDotter, 2024-07-12:
Thank you for the answer. Than this is no issue and the error lies on my behalf for not reading thoroughly. 

One last question, before it gets closed: 
Since the 7670E is bundled (and sold) with a GPS Antenna, is there an example which utilizes the Expansion?

papsngbayan, 2024-07-31:
> Thank you for the answer. Than this is no issue and the error lies on my behalf for not reading thoroughly.
> 
> One last question, before it gets closed: Since the 7670E is bundled (and sold) with a GPS Antenna, is there an example which utilizes the Expansion?

Am also interested in this follow up question. I have been trying to get the examples to work with the A7670E with GPS antenna but have had no luck so far.

lewisxhe, 2024-08-14:
> Thank you for the answer. Than this is no issue and the error lies on my behalf for not reading thoroughly.
> 
> One last question, before it gets closed: Since the 7670E is bundled (and sold) with a GPS Antenna, is there an example which utilizes the Expansion?

GPSShield in the example

lewisxhe, 2024-08-14:
> Am also interested in this follow up question. I have been trying to get the examples to work with the A7670E with GPS antenna but have had no luck so far.

You need to make sure whether the GPS function is included in your purchase

dashDotter, 2024-09-02:
> GPSShield in the example

> You need to make sure whether the GPS function is included in your purchase

Then why is the A7670E bundled with a GPS Antenna (and sold) in the first place if it is not possible to get the GPS Running with the E-Version?

lewisxhe, 2024-09-06:
All versions of A7670* have GPS and non-GPS functions, for example, A7670E has GPS and non-GPS functions, but A7670G definitely does not have GPS function

I don't understand what you mean by bundled sales

closed 2024-09-10 by lewisxhe