Xinyuan-LilyGO/LilyGo-Modem-Series #115: Can´t Receive or make calls on LTE

Sig2018, 2024-07-31:
I own a A7670SA working fine with MQTT connection.
I'm also able to receive and send SMS.
But voice call doesn't work on LTE(4G).
The only way to make it work is to force module to GSM(2G).
I guess there is some problem with network handshake when it request the CSFB (Circuit Switch Fallback) since VoLTE is not supported.
Can anyone help with any way to debug?
I've tried with different SIMs from different carriers, TinyGSM and AT commands with same result: NO call when in LTE 
Tank you in advance for any help.

lewisxhe, 2024-08-13:
First confirm whether the sim card supports volte, put the volte-supported card into the phone and try to see if it can browse the web when making calls. After confirming that it is OK in the phone, put it into the board for testing. VOLTE does not require special settings, as long as the card supports volte, it will automatically select

Sig2018, 2024-08-13:
Thank you.
No,  Sim doesn't support VoLTE.
As I said I'm looking for the modem to fall back to 2G.
Is that possible?

lewisxhe, 2024-08-13:
SIMCom technicians told me that all A7670 series support VOLTE and no additional settings are required.

Sig2018, 2024-08-13:
But I-m not asking about VoLTE.
I-m asking about fallback to 2G when an incoming call is received.

lewisxhe, 2024-08-13:
[image]
`AT+CNMP=13 ` Only 2G connections are allowed

Sig2018, 2024-08-13:
I had selected 2 (automatic).
In other modems and phones if VoLTE is not available it will switch to 2G when an incoming call is received. But Usually it will connect to LTE for data.
I need to use the board normally in LTE but be able to switch to 2G automatically when a call is receive.
Is it possible?

lewisxhe, 2024-08-13:
As far as I know if Volte is not available then the demodulator will automatically switch modes

lewisxhe, 2024-08-15:
Do you mean that if the current module is in the LTE network, if a call comes in, the modem will not be able to receive the call notification? Cannot answer the call?

Sig2018, 2024-08-15:
If an LTE SIM with Not VoLTE service is plugged on a modem from any other brand or a phone the network instruct the modem to switch to 3G or 2G and then call is received.
In your board apparently there is some missing setting because when modem is on LTE nothing happen when call is on connection state and caller receive a message saying customer (lilygo A7670) is not available. However customer is available and connected to LTE data and able to receive SMS too.
The problem is Fallback to 2G is not happening:
Can you ask SIMCOM about this issue??

lewisxhe, 2024-08-15:
So if it is in the GSM network, I can answer and make calls, right?

Sig2018, 2024-08-15:
Well is not working as SIMCOM say.
The SIM do support 2G calls, when I force the modem to 2G permanent the calls are received.
But en Auto mode if Modem is on LTE call does not work. 
I've tried with 3 SIMs from  different operator so it is not a network or service problem but a Modem Problem.
Maybe you can share with me the SIMCOM contact and I can do some test with their guidance.
Or you can told them and transfer what they say to me.

lewisxhe, 2024-08-16:
SIMCOM needs your device's running log to further analyze the problem.
1. Please download the tools needed to run the capture log [here](https://drive.google.com/drive/folders/115qZ_0FeM-abigEliNge3JbYbbcAuR2x?usp=sharing)
2. Please write ATDebug to the board, just turn on the modem
3. Prepare the preliminary work of capturing the log according to the steps in the document
4. When ready to capture
- Manually send AT+CFUN=0, then wait for a few seconds
- Send AT+CFUN=1 to exit the flight mode
- Wait for network registration, send AT+CPSI? Check whether the current network enters LTE
- Send ATD123456; Dial the phone, replace 123456 with the mobile phone number, wait for ten seconds and hang up
- Then use another mobile phone to call the SIM card of the modem
After waiting for thirty seconds, use the provided tool to export the log file, and then send it to me, I will forward it to SIMCOM for analysis

Tips, when capturing logs, the USB-C of the board is powered
MicroUSB is the capture port of the modem, if there is no driver, you can find it here
https://onedrive.live.com/?authkey=%21ALGH2%2DBvnBNS6BQ&id=5D4ECE5EA53AE966%216449&cid=5D4ECE5EA53AE966&parId=root&parQt=sharedby&o=OneUp

Sig2018, 2024-08-19:
Hi @lewisxhe 
I think I have the requested log(There are a couple of options on tool and the instructions were not really detailed)

However there is an error when I try to place call(please see below) so not sure if we will get something.
Where should I upload the log? I wouldn't like to upload here since I think there si some sensible UE and SIM information on log.

15:07:55.795 -> 
15:07:55.795 -> +CGEV: NW PDN DEACT 1
15:07:56.238 -> 
15:07:56.280 -> OK
15:08:03.715 -> AT+CFUN=1

15:08:06.370 -> OK
15:08:06.719 -> 
15:08:06.719 -> +CPIN: READY
15:08:07.672 -> 
15:08:07.672 -> SMS DONE
15:08:15.251 -> 
15:08:15.251 -> +CGEV: EPS PDN ACT 1
15:08:16.268 -> 
15:08:16.268 -> +CGEV: ME PDN DEACT 8
15:08:17.187 -> 
15:08:17.224 -> PB DONE
15:08:24.699 -> AT+CPSI?

15:08:24.731 -> +CPSI: LTE,Online,730-01,0x333B,3092229,271,EUTRAN-BAND28,9485,4,0,7,53,58,-3
15:08:24.804 -> 
15:08:24.804 -> OK
15:08:33.826 -> ATD99XXXXXXX

15:08:33.867 -> ERROR

Sig2018, 2024-08-20:
OK, so it is not working but command is accepted now.
Where should I upload the log file?

21:33:50.275 -> atd97xxxxxxxx;
21:33:50.485 -> +CLCC: 1,0,2,0,0,"97xxxxxxxx",129,"",,0
21:33:50.531 -> 
21:33:50.577 -> OK
21:34:26.917 -> 
21:34:26.917 -> +CLCC: 1,0,6,0,0,"97xxxxxxx",129,"",,0
21:34:26.949 -> 
21:34:26.981 -> NO CARRIER

lewisxhe, 2024-08-20:
Have you connected the microphone and lama? If not, you cannot make calls.

Sig2018, 2024-08-20:
I didn't yet but anyway I will need to receive calls and it is not working.
It is a modem problem because on 2G calls are received, didn't try sound but I can get ring message and also caller ID.
Does Not happen on LTE.
Where should I upload the log file?

lewisxhe, 2024-08-20:
According to the method mentioned above, after recording the log, paste the exported compressed package on gaithub.

Sig2018, 2024-08-20:
OK here you have the log: https://github.com/Sig2018/SIM7670SA
Please confirm you are able to download.

lewisxhe, 2024-08-21:
OK, I will send it to SIMCOM for them to check. I will reply here if there is any news.

lewisxhe, 2024-08-21:
You need to provide me with the version information of the modem you are using and send me the QR code on the label of the camera module.

Sig2018, 2024-08-21:
AT+CGMM
A7670SA-LASE
AT+CGMR
+CGMR: A011B19A7670M7

What do you mean by camera module?

lewisxhe, 2024-08-21:
Use at+simcomati to query the results

Sig2018, 2024-08-21:
at+simcomati

Manufacturer: INCORPORATED
Model: A7670SA-LASE
Revision: A011B19A7670M7
A7670M7_B19V01_240126
QCN:
IMEI: 863829074xxxxxx
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

lewisxhe, 2024-08-21:
OK

lewisxhe, 2024-08-22:
Please provide me the PN number on the label on the modem. SIMCOM replied that the logs you provided cannot be opened. Did you follow the manual steps to capture the logs?

lewisxhe, 2024-08-22:
SIMCOM cannot extract the log you sent. Please upgrade the latest firmware and test again. If it still doesn't work, please send `AT+MEDCR=0,161,16`, then send AT+CFUN=0, wait a few seconds and then send AT+CFUN=1, and try again. If it still doesn't work, please use this firmware to capture the log. Please refer to the upgrade document and video here https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

Sig2018, 2024-08-22:
Could you explain how to check the version of firmware I have and the latest one, changes, etc?
On the page you pointed there is not a clear reference of firmware version just a link to download.
I won't just download a file and make an upgrade not knowing what it is on it, with a high risk to brick the device.
The documentation is poor and needs to be improved.

Sig2018, 2024-08-22:
As you can see log is being captured.
There are a lot of red lines related with some come COM warning, maybe that is causing troubles to SIMCOM.
[image]

Sig2018, 2024-08-22:
Just Uploaded a new log. https://github.com/Sig2018/SIM7670SA
To check if they can see now.

lewisxhe, 2024-08-23:
Here is the upgrade video, have you watched it? Send AT+SIMCOMATI to check the current version https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md#record-video

lewisxhe, 2024-08-23:
SIMCOM responded:

Please send 
1. `AT+MEDCR=0,161,16`
2. `AT+CFUN=0`, wait a few seconds 
3. `AT+CFUN=1`
4. Wait for network registration to complete
5. After connecting to the LTE network, make a phone call
6. Then call A7670 from other mobile phones
7. Wait 30 seconds to export the log file and send it to me again

Sig2018, 2024-08-27:
Hi @lewisxhe 
Did you check the second version of the log I've uploaded?

lewisxhe, 2024-08-29:
I sent Log 2024-08-22 12;58;37.168.zip to SIMCOM and got the following reply:

Log 2024-08-22 12;58;37.168.zip

Sig2018, 2024-08-29:
I think you forgot to copy the reply

lewisxhe, 2024-08-30:
> SIMCOM responded:
>
> Please send
>
> 1. `AT+MEDCR=0,161,16`
> 2. `AT+CFUN=0`, wait a few seconds
> 3. `AT+CFUN=1`
> 4. Wait for network registration to complete
> 5. After connecting to the LTE network, make a [...]

This is the reply

Sig2018, 2024-08-30:
Log Upladed https://github.com/Sig2018/SIM7670SA/blob/main/Log%202024-08-30%2013%3B41%3B53.392.zip

Sig2018, 2024-09-05:
@lewisxhe Any answer with log?

lewisxhe, 2024-09-06:
@Sig2018 SIMCom reports that you need to upgrade to the latest firmware before capturing logs. Maybe the new firmware has fixed something.

Please upgrade the firmware before capturing logs. I have mentioned how to upgrade above.

Sig2018, 2024-09-18:
After having some time to update the firmware the one I've downloaded is exactly the Same the modem already had.
But now it can't register on network.
So I not only have one problem but 2.
Can you help?

Sig2018, 2024-09-18:
[image] before update](https://github.com/user-attachments/assets/834f3340-85bf-4029-986e-2758572078ca)

lewisxhe, 2024-09-19:
Do you mean after upgrading the firmware, the problem still exists, but log capture is not available?

Sig2018, 2024-09-19:
No i mean:
I already sent two logs following the procedure.
SIMCOM requested to send it again after upgrading firmware but firmware was already the latest available so :
Why SIMCOM cannot tell me what the problem is with the logs I already sent if the firmware is already the last one?
It looks there some disconnection between what they ask and what I sent. Are they really looking at my logs?

lewisxhe, 2024-09-20:
Please send `AT+BT`  to see the return value and tell me what it is

Tsury, 2024-11-20:
@Sig2018 DId you end up solving this issue?

Phoenix3232, 2024-11-27:
> No i mean: I already sent two logs following the procedure. SIMCOM requested to send it again after upgrading firmware but firmware was already the latest available so : Why SIMCOM cannot tell me [...]

Did you manage to solve the problem? I have the same problem with calls

Phoenix3232, 2024-11-27:
@Sig2018 ?

lewisxhe, 2024-11-28:
@Phoenix3232 If you encounter the same problem, please follow the instructions above to submit the modem log to me, and I will send it to simcom for analysis.

Sig2018, 2024-11-28:
@Phoenix3232 No I couldn't.
@lewisxhe I sent the log twice with no answer.
The Liliygo support on this matter is really weak

Sig2018, 2024-11-28:
@Tsury No I couldn't.
I sent my log to @lewisxhe twice with no answer.

Sig2018, 2024-11-28:
Did you read my answer from Sep 19th?

> No i mean: I already sent two logs following the procedure. SIMCOM requested to send it again after upgrading firmware but firmware was already the latest available so : Why SIMCOM cannot tell me [...]

lewisxhe, 2024-11-28:
The version that needs to submit the log should be A011B21V01A7670M7
[image]

Sig2018, 2024-11-28:
New log Last firmware https://github.com/Sig2018/SIM7670SA/blob/main/Log%202024-11-28%2000%3B17%3B33.032.zip

Sig2018, 2024-11-28:
I've tried with a SIM capable of use VoLTE and it works. As I already stated here the problem is the capacity of the modem to  CSFB (Circuit Switch Fallback) to 2G. It doesn't work.

lewisxhe, 2024-11-29:
The log has been submitted to SIMCOM and now we need to wait for their response.

lewisxhe, 2024-11-29:
@Sig2018 Is the log captured based on version A011B21V01A7670M7?

Sig2018, 2024-11-29:
Yes

lewisxhe, 2024-11-29:
@Sig2018 This is the firmware that SIMCOM sent to you for testing regarding this issue. Please try it and see if the issue is resolved. [A7678-22969_A7670SA-LASE_csfb_test](https://drive.google.com/file/d/10NUjVUjX_uW7KIzA2EkpJY-7835asMYH/view?usp=sharing)

Sig2018, 2024-11-29:
I can't flash the firmware
 "message" : "<COM11> FAILED (remote: erase partition \"bootloader\" failed.)\n"
These is one of the several messages 
Not sure if the problem was last firmware or what
Complete log here
https://github.com/Sig2018/SIM7670SA/blob/main/Fail%20flash.txt
Modem is working with no CSFB

lewisxhe, 2024-11-30:
Press and hold the modem boot button when powering on, and hold it until the modem turns on, then release it. If that doesn't work, I'll try it on Monday.

Sig2018, 2024-11-30:
> Press and hold the modem boot button when powering on, and hold it until the modem turns on, then release it. If that doesn't work, I'll try it on Monday.

Yes I'm doing that on every upgrade but this time it gave the error I sent to you

[21 more replies]

closed 2025-01-31 by github-actions[bot]