Xinyuan-LilyGO/LilyGo-Modem-Series #117: HTTPS requests not working, error 715

kgmuzungu, 2024-08-12:
Hi, 
the provided example HttpsBuiltlnGet.ino, connecting to https://httpbin.org/get, only seems to work BUT it does NOT. https://httpbin.org/get falls back to HTTP if HTTPS cannot be established. 

Can you provide a working example with HTTPS? E.g. connecting to https://vsh.pp.ua/TinyGSM/logo.txt

That would be great!
Many thanks in advance.

kgmuzungu, 2024-08-12:
My modem:
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1

In this [post](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/32#issuecomment-1426870449) someone states that with another Revision there is no issue connecting with HTTPS. 

Does updating the firmware change the "Revision"? And if so, how to do?

kgmuzungu, 2024-08-12:
It seems that Revision: A7670M7_V1.11.1 does not support all AT commands mentioned in the manuals in this github repository. `AT+CSSLCFG?` lists 11 response fields but my modem only shows 10 fields. `AT+CSSLCFG="ignorecertCN",1,1` is apparently not implemented. That makes it more complected to test SSL connections.

lewisxhe, 2024-08-13:
Take a photo of the label on the modem and send it to me. I want to check what version you have.

Sig2018, 2024-08-13:
Same experience wit A7670SA. The example is not really working for urls that doesn't accept fall to http.

kgmuzungu, 2024-08-13:
`curl -v https://vsh.pp.ua/TinyGSM/logo.txt
*   Trying 77.90.0.38:443...
* Connected to vsh.pp.ua (77.90.0.38) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Finished (20):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.2 (OUT), TLS header, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=vsh.pp.ua
*  start date: Jun 29 10:16:35 2024 GMT
*  expire date: Sep 27 10:16:34 2024 GMT
*  subjectAltName: host "vsh.pp.ua" matched cert's "vsh.pp.ua"
*  issuer: C=US; O=Let's Encrypt; CN=E5
*  SSL certificate verify ok.
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> GET /TinyGSM/logo.txt HTTP/1.1
> Host: vsh.pp.ua
> User-Agent: curl/7.81.0
> Accept: */*
> 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx/1.22.1
< Date: Tue, 13 Aug 2024 15:56:16 GMT
< Content-Type: text/plain; charset=UTF-8
< Content-Length: 121
< Connection: keep-alive
< X-Powered-By: Express
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Headers: X-Requested-With
< Cross-Origin-Resource-Policy: cross-origin
< Accept-Ranges: bytes
< Cache-Control: public, max-age=86400
< Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
< ETag: W/"79-15ec2936080"
< 

  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

* Connection #0 to host vsh.pp.ua left intact`

kgmuzungu, 2024-08-13:
`openssl s_client -showcerts -servername vsh.pp.ua -connect vsh.pp.ua:443`
```
CONNECTED(00000003)
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = E5
verify return:1
depth=0 CN = vsh.pp.ua
verify return:1
---
Certificate chain
 0 s:CN = vsh.pp.ua
   i:C = US, O = Let's Encrypt, CN = E5
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: ecdsa-with-SHA384
   v:NotBefore: Jun 29 10:16:35 2024 GMT; NotAfter: Sep 27 10:16:34 2024 GMT
-----BEGIN CERTIFICATE-----
MIIDrDCCAzKgAwIBAgISA0E1OqsFacxPlesosguHQchAMAoGCCqGSM49BAMDMDIx
CzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQDEwJF
NTAeFw0yNDA2MjkxMDE2MzVaFw0yNDA5MjcxMDE2MzRaMBQxEjAQBgNVBAMTCXZz
aC5wcC51YTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABFQS4s3BahgzL0kMSlcn
sd6YacXTH1mAu+bemjYQCUaML9ae5y1Bmqop0703GipllP9YwMXhckUeWD4lXg4q
Q1ajggJEMIICQDAOBgNVHQ8BAf8EBAMCB4AwHQYDVR0lBBYwFAYIKwYBBQUHAwEG
CCsGAQUFBwMCMAwGA1UdEwEB/wQCMAAwHQYDVR0OBBYEFITZnaeHoqIsWozAVrkr
+uoDPP74MB8GA1UdIwQYMBaAFJ8rX888IU+dBLftKyzExnCL0tcNMFUGCCsGAQUF
BwEBBEkwRzAhBggrBgEFBQcwAYYVaHR0cDovL2U1Lm8ubGVuY3Iub3JnMCIGCCsG
AQUFBzAChhZodHRwOi8vZTUuaS5sZW5jci5vcmcvME4GA1UdEQRHMEWCDmtodGMu
dnNoLnBwLnVhggtsLnZzaC5wcC51YYIMcXIudnNoLnBwLnVhggl2c2gucHAudWGC
DXd3dy52c2gucHAudWEwEwYDVR0gBAwwCjAIBgZngQwBAgEwggEDBgorBgEEAdZ5
AgQCBIH0BIHxAO8AdQAZmBBxCfDWUi4wgNKeP2S7g24ozPkPUo7u385KPxa0ygAA
AZBjthNHAAAEAwBGMEQCIBybCTv2ElV1Lh6Gl/Js+ITHFPV1Z8SK3HkoKhOF9W0r
AiA6Go1u2kAV6i5XEs6TdR9Seu698LXueFHkxfhMC5py7gB2AO7N0GTV2xrOxVy3
nbTNE6Iyh0Z8vOzew1FIWUZxH7WbAAABkGO2E1IAAAQDAEcwRQIhAJAVX/K0o7MT
t8tfe3IavYKdIdMeaSV+/nunKmNRSzjTAiBczdAoqnqLv+Btb5HFHZc6IscMuX7T
bapXofm45XytYjAKBggqhkjOPQQDAwNoADBlAjBZxJdkF6POxGiY0rUbcIJypOzI
EepAgEsyrYnT0td6jO3bdHGH+EfsEN1HnLpnLT8CMQCz0W1E+HwS56FSQiqPMQR6
cYj1x3rD00xq+BpQIHPeCpXesIp4sHlAMmN3HK5DZac=
-----END CERTIFICATE-----
 1 s:C = US, O = Let's Encrypt, CN = E5
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   a:PKEY: id-ecPublicKey, 384 (bit); sigalg: RSA-SHA256
   v:NotBefore: Mar 13 00:00:00 2024 GMT; NotAfter: Mar 12 23:59:59 2027 GMT
-----BEGIN CERTIFICATE-----
MIIEVzCCAj+gAwIBAgIRAIOPbGPOsTmMYgZigxXJ/d4wDQYJKoZIhvcNAQELBQAw
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMjQwMzEzMDAwMDAw
WhcNMjcwMzEyMjM1OTU5WjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3Mg
RW5jcnlwdDELMAkGA1UEAxMCRTUwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQNCzqK
a2GOtu/cX1jnxkJFVKtj9mZhSAouWXW0gQI3ULc/FnncmOyhKJdyIBwsz9V8UiBO
VHhbhBRrwJCuhezAUUE8Wod/Bk3U/mDR+mwt4X2VEIiiCFQPmRpM5uoKrNijgfgw
gfUwDgYDVR0PAQH/BAQDAgGGMB0GA1UdJQQWMBQGCCsGAQUFBwMCBggrBgEFBQcD
ATASBgNVHRMBAf8ECDAGAQH/AgEAMB0GA1UdDgQWBBSfK1/PPCFPnQS37SssxMZw
i9LXDTAfBgNVHSMEGDAWgBR5tFnme7bl5AFzgAiIyBpY9umbbjAyBggrBgEFBQcB
AQQmMCQwIgYIKwYBBQUHMAKGFmh0dHA6Ly94MS5pLmxlbmNyLm9yZy8wEwYDVR0g
BAwwCjAIBgZngQwBAgEwJwYDVR0fBCAwHjAcoBqgGIYWaHR0cDovL3gxLmMubGVu
Y3Iub3JnLzANBgkqhkiG9w0BAQsFAAOCAgEAH3KdNEVCQdqk0LKyuNImTKdRJY1C
2uw2SJajuhqkyGPY8C+zzsufZ+mgnhnq1A2KVQOSykOEnUbx1cy637rBAihx97r+
bcwbZM6sTDIaEriR/PLk6LKs9Be0uoVxgOKDcpG9svD33J+G9Lcfv1K9luDmSTgG
6XNFIN5vfI5gs/lMPyojEMdIzK9blcl2/1vKxO8WGCcjvsQ1nJ/Pwt8LQZBfOFyV
XP8ubAp/au3dc4EKWG9MO5zcx1qT9+NXRGdVWxGvmBFRAajciMfXME1ZuGmk3/GO
koAM7ZkjZmleyokP1LGzmfJcUd9s7eeu1/9/eg5XlXd/55GtYjAM+C4DG5i7eaNq
cm2F+yxYIPt6cbbtYVNJCGfHWqHEQ4FYStUyFnv8sjyqU8ypgZaNJ9aVcWSICLOI
E1/Qv/7oKsnZCWJ926wU6RqG1OYPGOi1zuABhLw61cuPVDT28nQS/e6z95cJXq0e
K1BcaJ6fJZsmbjRgD5p3mvEf5vdQM7MCEvU0tHbsx2I5mHHJoABHb8KVBgWp/lcX
GWiWaeOyB7RP+OfDtvi2OsapxXiV7vNVs7fMlrRjY1joKaqmmycnBvAq14AEbtyL
sVfOS66B8apkeFX2NY4XPEYV4ZSCe8VHPrdrERk2wILG3T/EGmSIkCYVUMSnjmJd
VQD9F6Na/+zmXCc=
-----END CERTIFICATE-----
---
Server certificate
subject=CN = vsh.pp.ua
issuer=C = US, O = Let's Encrypt, CN = E5
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 2440 bytes and written 391 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_AES_256_GCM_SHA384
    Session-ID: D79FDE1F317C39114C8BC3F4E57AC66DE83D7A5DC9B79C2C7A6419D0442A9407
    Session-ID-ctx: 
    Resumption PSK: A78BB0B194E45348462980BAF1E2FB86C8279B78FFE7048503F00B15A8EBD9C86F6A1A485461BDBED24E192F2205321C
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 86400 (seconds)
    TLS session ticket:
    0000 - 29 61 dc dd cf e8 da 6a-ca 6d 0b e3 47 cc ab 14   )a.....j.m..G...
    0010 - 0f 33 d7 cc b7 25 c5 64-b5 59 20 08 57 76 91 3f   .3...%.d.Y .Wv.?

    Start Time: 1723565227
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_AES_256_GCM_SHA384
    Session-ID: 27C6146A2C3FF1998DADF1CAA707C187812F123C47F3312FF508AFFFED8F0143
    Session-ID-ctx: 
    Resumption PSK: 1302A60DC421FEF614E209879353EFF3DE22760E0D2B96A9D102223DA87F07C55AFCC49D152063E57C6CEF49A7975E3E
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 86400 (seconds)
    TLS session ticket:
    0000 - 25 9d 46 e8 7b ae 39 39-25 e9 0e 2b 20 a3 01 f1   %.F.{.99%..+ ...
    0010 - f1 be 26 59 9c 5c 95 4e-43 4b db d0 23 e3 5d df   ..&Y.\.NCK..#.].

    Start Time: 1723565227
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
```

kgmuzungu, 2024-08-13:
What I have tried in many variants but without success. 
```

#define TINY_GSM_RX_BUFFER          1024 // Set RX buffer to 1Kb

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

//#include "cert_tinygsm_test.h"
#include "cert_tinygsm_test_whole_chain.h"
#include "utilities.h"
#include <TinyGsmClient.h>

#define DUMP_AT_COMMANDS 
#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

// It depends on the operator whether to set up an APN. If some operators do not set up an APN, 
// they will be rejected when registering for the network. You need to ask the local operator for the specific APN.
// APNs from other operators are welcome to submit PRs for filling.
// #define NETWORK_APN     "CHN-CT"             //CHN-CT: China Telecom

//const char server_url[] =  "https://httpbin.org/get";
const char server_url[] =  "https://vsh.pp.ua/TinyGSM/logo.txt";
// the IP of vsh.pp.ua is 77.90.0.38
//const char server_url[] =  "https://orf.at";
//const char server_url[] =  "https://eu-central-1-1.aws.cloud2.influxdata.com";

int8_t input_on = 0;

void setup()
{
    Serial.begin(115200); // Set console baud rate

    Serial.println("[SETUP] Start Sketch");

    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);

#ifdef BOARD_POWERON_PIN
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

    // Set modem reset pin ,reset modem
    pinMode(MODEM_RESET_PIN, OUTPUT);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL); delay(100);
    digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL); delay(2600);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);

    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);

    // Check if the modem is online
    Serial.println("[SETUP] Start modem...");

    // teatAT does: 
    int retry = 0;
    while (!modem.testAT(1000)) {
        Serial.println(".");
        if (retry++ > 10) {
            digitalWrite(BOARD_PWRKEY_PIN, LOW);
[... 157 lines cut ...]
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMjQwMzEzMDAwMDAw
WhcNMjcwMzEyMjM1OTU5WjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3Mg
RW5jcnlwdDELMAkGA1UEAxMCRTUwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQNCzqK
a2GOtu/cX1jnxkJFVKtj9mZhSAouWXW0gQI3ULc/FnncmOyhKJdyIBwsz9V8UiBO
VHhbhBRrwJCuhezAUUE8Wod/Bk3U/mDR+mwt4X2VEIiiCFQPmRpM5uoKrNijgfgw
gfUwDgYDVR0PAQH/BAQDAgGGMB0GA1UdJQQWMBQGCCsGAQUFBwMCBggrBgEFBQcD
ATASBgNVHRMBAf8ECDAGAQH/AgEAMB0GA1UdDgQWBBSfK1/PPCFPnQS37SssxMZw
i9LXDTAfBgNVHSMEGDAWgBR5tFnme7bl5AFzgAiIyBpY9umbbjAyBggrBgEFBQcB
AQQmMCQwIgYIKwYBBQUHMAKGFmh0dHA6Ly94MS5pLmxlbmNyLm9yZy8wEwYDVR0g
BAwwCjAIBgZngQwBAgEwJwYDVR0fBCAwHjAcoBqgGIYWaHR0cDovL3gxLmMubGVu
Y3Iub3JnLzANBgkqhkiG9w0BAQsFAAOCAgEAH3KdNEVCQdqk0LKyuNImTKdRJY1C
2uw2SJajuhqkyGPY8C+zzsufZ+mgnhnq1A2KVQOSykOEnUbx1cy637rBAihx97r+
bcwbZM6sTDIaEriR/PLk6LKs9Be0uoVxgOKDcpG9svD33J+G9Lcfv1K9luDmSTgG
6XNFIN5vfI5gs/lMPyojEMdIzK9blcl2/1vKxO8WGCcjvsQ1nJ/Pwt8LQZBfOFyV
XP8ubAp/au3dc4EKWG9MO5zcx1qT9+NXRGdVWxGvmBFRAajciMfXME1ZuGmk3/GO
koAM7ZkjZmleyokP1LGzmfJcUd9s7eeu1/9/eg5XlXd/55GtYjAM+C4DG5i7eaNq
cm2F+yxYIPt6cbbtYVNJCGfHWqHEQ4FYStUyFnv8sjyqU8ypgZaNJ9aVcWSICLOI
E1/Qv/7oKsnZCWJ926wU6RqG1OYPGOi1zuABhLw61cuPVDT28nQS/e6z95cJXq0e
K1BcaJ6fJZsmbjRgD5p3mvEf5vdQM7MCEvU0tHbsx2I5mHHJoABHb8KVBgWp/lcX
GWiWaeOyB7RP+OfDtvi2OsapxXiV7vNVs7fMlrRjY1joKaqmmycnBvAq14AEbtyL
sVfOS66B8apkeFX2NY4XPEYV4ZSCe8VHPrdrERk2wILG3T/EGmSIkCYVUMSnjmJd
VQD9F6Na/+zmXCc=
-----END CERTIFICATE-----)EOF";
```

kgmuzungu, 2024-08-13:
what is wrong here? @lewisxhe

lewisxhe, 2024-08-14:
Can you tell me the product number on the box? I need an identical one for testing, and I can't tell you why now because I don't have this product.
Although they are all A7670, there are dozens of versions of this demodulator.

kgmuzungu, 2024-08-14:
I mentioned that above: 
The QR code on my A7670E translates to: p/N:S2-10AAW-Z319S; SN:MP06233284C8E02; IMEI:860470067524920; BTMAC:5C46B08A2B6C; SW:A011B04A7670M7_F
I bought it in the Netherlands via tinytronics.nl
Do you need anything else?
[image]

lewisxhe, 2024-08-14:
No, there is a sticker on the box that contains the product's SKU, which should also be included on the purchase link.

kgmuzungu, 2024-08-14:
The SKU from the retailer is 005034. the semi transparent plastic box I only will have in a week again. I will post it then. I dont have it with me. 
I really need to get HTTPS POST command working. Otherwise this product is no option for me. We potentially would buy hundreds. 

Product number and SW number dont help you? Firmware update would be an option?
Thanks in advance.

lewisxhe, 2024-08-14:
Please send AT+SIMCOMATI to check the current running firmware version

kgmuzungu, 2024-08-14:
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 860470067524920
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

And Revision is:
Revision: A7670M7_V1.11.1

lewisxhe, 2024-08-14:
[image]

I feel like this is more like the operator's firewall blocking access to some IPs, because the other two websites are also https

lewisxhe, 2024-08-14:
I have contacted SIMCOM to seek their help and see if they can provide any clues. Now I just need to wait for their reply.

lewisxhe, 2024-08-19:
@kgmuzungu 
SIMCOM replied: Need to add
```
AT+CSSLCFG="enableSNI",0,1
```
Try it, I can't test it, I suspect there is something wrong with my network and I can't access these websites

[image]

kgmuzungu, 2024-08-23:
I will give it a try this weekend.

kgmuzungu, 2024-09-05:
I did try it and it did **N O T** work. It seems that native SSL is not working on this GSM modem or with my firmware version. If someone is interested I can post the code I used. 

It would be great @lewisxhe if you can get a working example from SIMCom plus the related firmware.

kgmuzungu, 2024-09-05:
I found a **workaround**. What works is that you open a TCP connection with the modem and then do the SSL part in software.
Here how I did it:
in platform.ini:
```
[env:T-A7670X]
extends = esp32dev_base
build_flags = ${esp32dev_base.build_flags}
	-DLILYGO_T_A7670
	-DTINY_GSM_MODEM_A7670
lib_deps =  coryjfowler/mcp_can@^1.5.1
                  SPI
	          TinyGSM
		  ArduinoHttpClient
		  StreamDebugger
	          https://github.com/govorox/SSLClient#95-release-130-fails-to-compile-on-arduino-esp32-v3 
```

And a code example:

lewisxhe, 2024-09-06:
OK. I will continue to communicate with SIMCOM to see if they can provide a solution. I will update the results here later.

lewisxhe, 2024-09-06:
SIMCOM has provided a new firmware, which I have tested and found to have fixed the problem. I have asked them to give me an official version of the firmware, and I will update it here after they give it to me.

kgmuzungu, 2024-09-06:
Regarding your latest commit (comment in code referenced just above) that the HTTPS connection problem is related to firewalls is simply not correct. It is rather related to the SSL implementation of the A7670. Prove me wrong, hence post here your tests that underpin your statement.

lewisxhe, 2024-09-06:
No, this is my note, I do have a firewall here, I need to make a note because these boards are also sold in my country

kgmuzungu, 2024-09-18:
@lewisxhe would it make sense to update the firmware on my SIMCom A7670E 

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 860470067524920
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

And Revision is:
Revision: A7670M7_V1.11.1

with 
* [A7670 LASE B14](https://drive.google.com/file/d/1ERblToPH4FoAo2dVYn3B--u_FzhIzeMQ/view?usp=sharing)
* or [A7670 FASE B14](https://drive.google.com/file/d/107PBlY_Pj4SOxmI-mp8ssxoW2SOMbbWT/view?usp=sharing)

This [thread](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/122) gives some hints how-to. 

My goal is to get *native HTTPS working*. Should I update the firmware?

lewisxhe, 2024-09-19:
The firmware that can be used now is not archived. SIMCOM will send it to me after it is archived. Then, I will publish it.

kgmuzungu, 2024-09-24:
My workaround with [gorovox SSLClient](https://github.com/govorox/SSLClient) was not satisfying. I've got frequent errors. 

I am using now [ESP_SSLClient](https://registry.platformio.org/libraries/mobizt/ESP_SSLClient) an SSLClient based on [BearSSL](https://bearssl.org/). Much better! 

But I am still looking forward to the SIMCom firware update! I hope we get it soon!

kgmuzungu, 2024-10-01:
@lewisxhe any new here?

lewisxhe, 2024-10-01:
I am currently on holiday, I will contact them after the holiday. I will add updates here.

chemmex, 2024-10-01:
I seem to have the same problem with A7608SA. Any chance to get a FW update for my module too?

lewisxhe, 2024-10-04:
@chemmex Please open a new issue and tag your modem to identify the hardware version.

kgmuzungu, 2024-10-08:
@lewisxhe pls get the latest firmware!

lewisxhe, 2024-10-08:
[A7670E-LASE A124B01](https://drive.google.com/file/d/1-14KABQxVgjcJjJVuofp1BkZd0VYZVWA/view?usp=sharing) SIMCom just sent me a new version of firmware, please flash it and test it

kgmuzungu, 2024-10-08:
🙏 I did the upgrade:
old:
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 860470067524920
MEID:

new:
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-LASE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 

I will test later how it behaves

kgmuzungu, 2024-10-08:
Good news! native HTTPS is working! hurray!

Bad news: 
AT+CGNSSPWR=?
AT+CGNSSPWR=?
ERROR
AT+CGNSSINFO
AT+CGNSSINFO
ERROR
AT+CGPSINFO
AT+CGPSINFO
ERROR
AT+CGNSSIPR=115200
AT+CGNSSIPR=115200
ERROR
AT+CGNSSIPR?
AT+CGNSSIPR?
ERROR
AT+CGPSCOLD
AT+CGPSCOLD
ERROR

no response to GPS commands!
@lewisxhe I have to fix this until this Thursday. pls help!

kgmuzungu, 2024-10-08:
I am now trying the other firmware, A011B14, that seems to be a bit older but I guess a bit newer than my original one.
Manufacturer: INCORPORATED
Model: A7670E-LASE
Revision: A011B14A7670M7
A7670M7_B14V04_221031
QCN:
IMEI: 860470067524920

kgmuzungu, 2024-10-08:
@lewisxhe  With the newest, A124B01, and the firmware before, A011B14, I dont get a response on GPS commands. What can I do? Or what did I wrong?

lewisxhe, 2024-10-09:
Sorry, it's a mistake. You should write [A7670E-FASE B07](https://drive.google.com/file/d/1-5A9w4MCXNz6F5ODhynne0rC9IUOeOdH/view?usp=sharing), please update again

kgmuzungu, 2024-10-09:
update worked. will test during the day

AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927

kgmuzungu, 2024-10-09:
**native HTTPS works and GPS responds to commands!** 🥇 

I havnt tested how stable it the whole thing is.

Mr. @lewisxhe that was a very difficult birth. 
My advice, keep getting firmware updates from SIMCOM and clean your adaptations to TinyGSM up.

JMartinezEco, 2024-10-16:
OK, but what about the other modules. I have the exact same failure with a SIM A7670G but I cannot find a fimrware update anywhere in the web. 

Some help here?

lewisxhe, 2024-11-20:
@kimble4 Please open a new issue and provide the log of the AT interaction.

closed 2025-01-05 by github-actions[bot]