Xinyuan-LilyGO/LilyGo-Modem-Series #118: LILYGO TTGO T-A7670E - cannot fix GPS tracking

katoury, 2024-08-12:
Hi, 
I have just bought the LILYGO TTGO T-A7670E to create a gps tracker. I have tried some examples (https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX) but they does not work. I use the T-A7670E outdoor connected to my computer with usbc  (no battery)  and no sim card because I wanted to try GPS first. 
Could you please help me to find the good example code to go on with my project.
Any advice would be appreciated.
Thanks

lewisxhe, 2024-08-13:
See [FAQ](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX#6%EF%B8%8F%E2%83%A3-faq)  Second point

katoury, 2024-08-13:
Hi,
Thanks for answering

I have bought the GPS version with the physical GPS antenna that I have plugged in.
I have tried the « GPS_NMEA_Output.ino » with the utilities.h attached.
I have selected #define LILYGO_T_A7670  
But when I compile before uploading on the ESP32 board I have got the following message : 

In file included from /private/var/folders/30/srdcy9nn0_b1mxmrt8jqwmc40000gr/T/.arduinoIDE-unsaved2024712-34922-q911f0.0fmrn/sketch_aug12i/sketch_aug12i.ino:17:
/Users/Daniel/Documents/Arduino/libraries/TinyGsmClient/TinyGsmClient.h:132:2: error: #error "Please define GSM modem model"
  132 | #error "Please define GSM modem model"
      |  ^~~~~
exit status 1
Compilation error: exit status 1

I have searched the web but cannot find a solution to fix it.

I would appreciated some help.

lewisxhe, 2024-08-13:
The README contains several notes and common problems. Please read them carefully before using. The error you mentioned is explained in the first line of the README.

katoury, 2024-08-13:
Sorry to bother you, but I have read the README and cannot find the solution. The files I use to compile in ArduinoIDE are those I have copied from the example folder. I have tried the lines into the .ino file to :
#define LILYGO_T_A7670
#include "utilities.h"__
instead of #define LILYGO_T_A7670 into the "utilities.h"
But it makes no difference.
I don't know what I am doing wrong, sorry

lewisxhe, 2024-08-14:
Only I know, I will tell you

katoury, 2024-08-14:
HI,
I want to go ahead with my project. But today it's raining and cloudly. I have tried the "GPS_BuiltIn" with no battery only usbC cable. Could you explain how the longitude and latitude can be found when it mentions "Visible Satellites:0	Used Satellites:0" ?
Thanks

katoury, 2024-08-14:
I think so as it is mentioned "AT+CGNSSPWR=1" ; see the serial monitor (I have had the 
11:13:49.717 -> AT+CGNSSINFO

11:13:49.717 -> +CGNSSINFO: ,,,,,,,,
11:13:49.717 -> 
11:13:49.717 -> OK
11:13:49.717 -> Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
11:14:04.722 -> Requesting current GPS/GNSS/GLONASS location
11:14:04.722 -> AT+CGNSSINFO
11:14:04.761 -> AT+CGNSSINFO

11:14:04.761 -> +CGNSSINFO: 3,13,,05,00,.....................
11:14:04.761 -> 
11:14:04.761 -> OK
11:14:07.759 -> FixMode:3
11:14:07.759 -> Latitude:.............	Longitude:................... (long. & lat. deleted)
11:14:07.759 -> Speed:0.14	Altitude:363.40
11:14:07.759 -> Visible Satellites:0	Used Satellites:0
11:14:07.759 -> Accuracy:1.98
11:14:07.759 -> Year:2024	Month:8	Day:14
11:14:07.759 -> Hour:9	Minute:14	Second:4
11:14:07.759 -> http://www.google.com/maps/place/.......................;
11:14:12.772 -> Retrieving GPS/GNSS/GLONASS location again as a string
11:14:12.772 -> AT+CGNSSINFO
11:14:12.772 -> AT+CGNSSINFO

11:14:12.772 -> +CGNSSINFO: 3,13,,05,00,........................ (deleted)
11:14:12.772 -> 
11:14:12.772 -> OK
11:14:12.772 -> GPS/GNSS Based Location String:3,13,,05,00,..................(deleted)
11:14:12.772 -> Disabling GPS
11:14:12.772 -> AT+CGNSSPWR=0
11:14:12.816 -> AT+CGNSSPWR=0

11:14:12.903 -> OK

lewisxhe, 2024-08-14:
Try this example https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/GPS_NMEA_Parse

katoury, 2024-08-14:
I have tried and it finds my location like the GPS_builtin one
So what else ?

katoury, 2024-08-14:
What am I suppose to do with the informations from LONDON ?

katoury, 2024-08-14:
OK, it seems to work now. How can I do to send SMS from LILYGO_T_A7670 as the library SoftwareSerial.h generate an issue (fatal error: avr/interrupt.h: No such file or directory)

lewisxhe, 2024-08-15:
GPS can find the location successfully, right? That means GPS is working properly.

katoury, 2024-08-15:
Yes it does

katoury, 2024-08-15:
I have mounted a SIM card to try to send SMS. As I could not send a SMS from the 7670 I had some others commands to verify what is wrong. But at every command like "mySerial.println("AT+CPIN?")" or "Serial.println("AT+CREG? " the firmware stops and never goes on. I have tried without the SIM card and the issue is the same. Could you tell me more ?

katoury, 2024-08-15:
I have tried  "https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/SendSMS/SendSMS.ino" with the APN changed for my SIM card but  I get "-> +CMS ERROR:Send sms message fail" and "->  Network timeout"

lewisxhe, 2024-08-16:
Provide full log

katoury, 2024-08-16:
11:19:08.443 -> T+CSQ
11:19:08.443 -> AT+CSQ

11:19:08.443 -> +CSQ: 99,99
11:19:08.443 -> 
11:19:08.443 -> OK
11:19:08.443 -> [37] Signal Quality:99
11:19:08.443 -> 
11:19:08.443 -> OK
11:19:08.443 -> API_FAST_FLASH_BOOT)
11:19:08.443 -> configsip: 0, SPIWP:0xee
11:19:08.443 -> clk_drv:0x00,q_drAT+CGREG?
11:19:08.773 -> AT+CGREG?

11:19:08.773 -> +CGREG: 0,0
11:19:08.773 -> 
11:19:08.773 -> OK
11:19:08.773 -> AT+CSQ
11:19:08.773 -> AT+CSQ

11:19:08.773 -> +CSQ: 99,99
11:19:08.773 -> 
11:19:08.773 -> OK
11:19:08.773 -> [38] Signal Quality:99
11:19:09.795 -> AT+CGREG?
11:19:09.795 -> AT+CGREG?

11:19:09.795 -> +CGREG: 0,0
11:19:09.795 -> 
11:19:09.795 -> OK
11:19:09.795 -> AT+CSQ
11:19:09.795 -> AT+CSQ

11:19:09.795 -> +CSQ: 99,99
11:19:09.795 -> 
11:19:09.795 -> OK
11:19:09.795 -> [39] Signal Quality:99
11:19:10.784 -> AT+CGREG?
11:19:10.817 -> AT+CGREG?

11:19:10.817 -> +CGREG: 0,0
11:19:10.817 -> 
11:19:10.817 -> OK
11:19:10.817 -> AT+CSQ
11:19:10.817 -> AT+CSQ

11:19:10.817 -> +CSQ: 99,99
11:19:10.817 -> 
11:19:10.817 -> OK
11:19:10.817 -> [40] Signal Quality:99
11:19:11.806 -> AT+CGREG?
11:19:11.806 -> AT+CGREG?

11:19:11.806 -> +CGREG: 0,0
11:19:11.806 -> 
11:19:11.806 -> OK
11:19:11.806 -> AT+CSQ
11:19:11.806 -> AT+CSQ

11:19:11.806 -> +CSQ: 99,99
11:19:11.806 -> 
11:19:11.806 -> OK
11:19:11.806 -> [41] Signal Quality:99
11:19:12.824 -> AT+CGREG?
11:19:12.824 -> AT+CGREG?

11:19:12.824 -> +CGREG: 0,0
11:19:12.824 -> 
11:19:12.824 -> OK
11:19:12.824 -> AT+CSQ
11:19:12.824 -> AT+CSQ

11:19:12.824 -> +CSQ: 99,99
11:19:12.824 -> 
11:19:12.824 -> OK
11:19:12.824 -> [42] Signal Quality:99
11:19:13.846 -> AT+CGREG?
11:19:13.846 -> AT+CGREG?

11:19:13.846 -> +CGREG: 0,0
11:19:13.846 -> 
11:19:13.846 -> OK
11:19:13.846 -> AT+CSQ
11:19:13.846 -> AT+CSQ

11:19:13.846 -> +CSQ: 99,99
11:19:13.846 -> 
11:19:13.846 -> OK
11:19:13.846 -> [43] Signal Quality:99
11:19:14.835 -> AT+CGREG?
11:19:14.835 -> AT+CGREG?

11:19:14.835 -> +CGREG: 0,0
11:19:14.835 -> 
11:19:14.835 -> OK
11:19:14.835 -> AT+CSQ
11:19:14.868 -> AT+CSQ

11:19:14.868 -> +CSQ: 99,99
11:19:14.868 -> 
11:19:14.868 -> OK
11:19:14.868 -> [44] Signal Quality:99
11:19:15.858 -> AT+CGREG?
11:19:15.858 -> AT+CGREG?

11:19:15.858 -> +CGREG: 0,0
11:19:15.858 -> 
11:19:15.858 -> OK
11:19:15.858 -> AT+CSQ
11:19:15.858 -> AT+CSQ

11:19:15.858 -> +CSQ: 99,99
11:19:15.858 -> 
11:19:15.858 -> OK
11:19:15.858 -> [45] Signal Quality:99
11:19:16.879 -> AT+CGREG?
11:19:16.879 -> AT+CGREG?

11:19:16.879 -> +CGREG: 0,0
11:19:16.879 -> 
11:19:16.879 -> OK
11:19:16.879 -> AT+CSQ
11:19:16.879 -> AT+CSQ

11:19:16.879 -> +CSQ: 99,99
11:19:16.879 -> 
11:19:16.879 -> OK
11:19:16.879 -> [46] Signal Quality:99
11:19:17.865 -> AT+CGREG?
11:19:17.898 -> AT+CGREG?

11:19:17.898 -> +CGREG: 0,0
11:19:17.898 -> 
11:19:17.898 -> OK
11:19:17.898 -> AT+CSQ
11:19:17.898 -> AT+CSQ

11:19:17.898 -> +CSQ: 99,99
11:19:17.898 -> 
11:19:17.898 -> OK
11:19:17.898 -> [47] Signal Quality:99
11:19:18.883 -> AT+CGREG?
11:19:18.916 -> AT+CGREG?

11:19:18.916 -> +CGREG: 0,11
11:19:18.916 -> 
11:19:18.916 -> OK
11:19:18.916 -> Registration Status:11
11:19:19.903 -> 
11:19:19.903 -> Registration Status:11
11:19:20.891 -> Init success, start to send message to  +336........
11:19:20.891 -> AT+GSN
11:19:20.891 -> AT+GSN

11:19:20.923 -> 862771076222671
11:19:20.923 -> 
11:19:20.923 -> OK
11:19:20.923 -> AT+CMGF=1
11:19:20.923 -> AT+CMGF=1

11:19:20.988 -> OK
11:19:20.988 -> AT+CSCS="GSM"
11:19:20.988 -> AT+CSCS="GSM"

11:19:21.086 -> OK
11:19:21.086 -> AT+CMGS="+336........"
11:19:21.118 -> AT+CMGS="+336........"

11:19:21.118 -> >Hello from 862771076222671 Hello from 862771076222671
11:19:21.118 -> +CMS ERROR:Send sms message fail
11:19:21.118 ->  Network timeout

katoury, 2024-08-16:
@lucioborges Thanks a lot. I'm going to read your code ASAP. 😉

katoury, 2024-08-19:
@lewisxhe Hi, did you read the log ?

lewisxhe, 2024-08-19:
Sorry, I ignored your reply. There are too many posts to reply to. Sorry.

Your problem lies in the network registration step. Are you sure that the SIM card you are using can make calls and send text messages normally?

From the log +CGREG: 11, it can be seen that it is currently only registered in the emergency channel

First, you need to make sure that the SIM card you are using is normal. Please check again

[image]

katoury, 2024-08-19:
Yes my sim card can send text messages I know that because it is the one I use with my secure system and I use to send and receive sms with it.

lewisxhe, 2024-08-20:
Well, you should try to test it with a normal SIM card

lewisxhe, 2024-08-20:
You can now send, but not receive?

katoury, 2024-08-20:
I have found how to receive the sms. So now I can send sms in 1 program and I can receive sms in another one. But I don't know how to mix all the commands (send and receive) in one program, my tries do not work. Is there an example for the whole commands (send an receive) ?

lewisxhe, 2024-08-21:
Please use AT debugging, do not use the program first, understand the sending sequence first and then write the program. For detailed commands, please see the AT command manual in the docs directory. I cannot provide you with examples at present.

katoury, 2024-08-21:
Hi, don’t worry I will find a way to fix it

katoury, 2024-08-21:
@lewisxhe By the way, in the program where I display the location informations, in the LOOP void, it starts with 0 satellite and do the loop 7 o8 times before starting detecting satellites and displaying the true informations. Do you know why ?

lewisxhe, 2024-08-21:
GPS is a cold start, and positioning must be restarted every time the device is turned on.

katoury, 2024-08-22:
@lewisxhe Hi, I see sending sms and sending gps informations are two different things. I have now 2 codes to do that so thanks for the advices you sent  to me. But now, when I try to do them in only one code that do not work. I even try to do this with your sendsms.ino modified to do it without success. I think I am lost.

lewisxhe, 2024-08-22:
Maybe you could post the code

katoury, 2024-08-22:
/**
 * @file      SendsmsOK_receive.ino
 * @author    Lewis He (lewishe@outlook.com)
 * @license   MIT
 * @copyright Copyright (c) 2023  Shenzhen Xin Yuan Electronic Technology Co., Ltd
 * @date      2023-11-16
 * @note      Not support T-SIM7672
 * *  Although the manual of SIM7672G states that it has the functions of making voice calls and sending text messages,
 * *  the current firmware does not support it.
 */

#include "utilities.h"
#include <HardwareSerial.h>
//============== Début Ajout Daniel 14/08/2024
// Ne pas inclure SoftwareSerial pour les cartes ESP32 ou ESP8266
// Créer une instance de HardwareSerial
HardwareSerial mySerial(1);   // Utiliser le port série 1
//============== Fin Ajout Daniel 14/08/2024

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#define SMS_TARGET  "+mynumber" //Change the number you want to send sms message

#include <TinyGsmClient.h>
#include <TinyGPSPlus.h>

#define NETWORK_APN "sl2sfr"  // Change this to your network's APN

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

// The TinyGPSPlus object
TinyGPSPlus gps;

uint32_t check_interval = 0;

//unsigned long startTime;
//unsigned long interval = 1000;  // Intervalle de 1 seconde entre les commandes

void setup()
{
    Serial.begin(115200);

    // Configure la communication série avec le module SIM7670
    mySerial.begin(9600, SERIAL_8N1, MODEM_TX_PIN, MODEM_RX_PIN); // Utilise les broches RX 16 et TX 17 de l'ESP32
    Serial.println("Initialisation du module SIM7670...");

    delay(3000);

    // Turn on DC boost to power on the modem
    #ifdef BOARD_POWERON_PIN
      pinMode(BOARD_POWERON_PIN, OUTPUT);
      digitalWrite(BOARD_POWERON_PIN, HIGH);
    #endif

    pinMode(MODEM_RESET_PIN, OUTPUT);
[... 287 lines cut ...]
                SerialAT.print("Demande prise en compte\r");
                delay(1000);
                SerialAT.write(26);  // ASCII 26 pour Ctrl+Z
                delay(1000);

            }
        }
    }

    delay(5000);  
Serial.println("FIN LOOP");
}

static void smartDelay(unsigned long ms)
{
    int ch = 0;
    unsigned long start = millis();
    do {
        while (SerialAT.available()) {
            ch = SerialAT.read();
            // Serial.write(ch);
            gps.encode(ch);
        }
    } while (millis() - start < ms);
}

katoury, 2024-08-22:
@lewisxhe if you find the way to make it work I would appreciate thanks

lewisxhe, 2024-08-23:
🫠How do you use the methods in TinyGPS? gps.xx.xxx() These are TinyGPS methods, you can't mix them, this is only suitable for external GPS module

katoury, 2024-08-23:
@lewisxhe So what do you suggest to fix it ? Which method ma I suppose to use

katoury, 2024-08-23:
@lewisxhe Because in the GPS_NMEA_Parse.ino you use "gps.xx.xxx, so I have tried to use the sendSMS.ino and the GPS_NMEA_Parse.ino commands in one code to get what I am expecting : to receive a sms and send back the location. Once again sorry I am a newbie in those.

lewisxhe, 2024-08-23:
I just added an example of sending SMS after positioning, please check it out https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS

katoury, 2024-08-23:
@lewisxhe Thanks a lot... I am going to try this ASAP. I will be back to tell you then.

katoury, 2024-08-24:
@lewisxhe Hi I have read the code. If I want to use this kind of code how can I get and send the course in degrees which is not in TinyGsmGPS.tpp arguments.

lewisxhe, 2024-08-25:
You can refer to the gps_nmea_ouput example. I'm on vacation and can't provide an example, but isn't that similar to this one? Right?

embedded-systems-lol, 2024-08-29:
> I just added an example of sending SMS after positioning, please check it out https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS

could you add example of sending http post after positioning, for example how to send gps location to telegram bot...

lewisxhe, 2024-08-30:
I modified the HttpsBuiltlnPost example myself. I can't use telegram because it is blocked by the firewall.

lewisxhe, 2024-08-30:
> > I just added an example of sending SMS after positioning, please check it out https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS
> 
> could you add example of sending http post after positioning, for example how to send gps location to telegram bot...

Please modify it according to the HttpsBuiltlnPost example. I cannot use Telegram because it is blocked by the firewall.

katoury, 2024-08-31:
> You can refer to the gps_nmea_ouput example. I'm on vacation and can't provide an example, but isn't that similar to this one? Right?  _(course data is not in TinyGsmGPS.tpp arguments.)_

@lewisxhe HI, I have tried your SendLocationFromSMS with modified data to send and have a look to gps_nmea_ouput and gps_nmea_parse examples, but I cannot find the way to add the course as you told. Any advice ? Thanks

lewisxhe, 2024-08-31:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS_Use_TinyGPS

katoury, 2024-09-02:
@lewisxhe Thanks, I now can go on with your explanation

EnziinSystem, 2024-09-05:
I have a related problem, when I compile any examples it errors.

```
In file included from C:\Users\xxx\Desktop\GPS\LilyGO-T-A76XX\examples\SendLocationFromSMS_Use_TinyGPS\SendLocationFromSMS_Use_TinyGPS.ino:23:
c:\Users\xxx\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:132:2: error: #error "Please define GSM modem model"
  132 | #error "Please define GSM modem model"
      |  ^~~~~
exit status 1
```
The board defined in utilities.h

```
// T-Call-A767X V1.0 and V1.1 have different Pinmaps and cannot be used universally.
#define LILYGO_T_CALL_A7670_V1_0
```

These are just basic examples, why don't you test them? Don't just sell boards.

EnziinSystem, 2024-09-05:
I also have problems with GPS on LILYGO_T_CALL_A7670_V1_0

Run the example

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS_Use_TinyGPS

The message

```
Start modem...
[10466] ### Unhandled: *ATREADY: 1
Wait for the modem to register with the network.[10] Signal Quality:99
[11] Signal Quality:99
[12] Signal Quality:28
Online registration successful

Registration Status:1
Enabling GPS/GNSS/GLONASS

GPS Enabled

UC6228CI lite G1L1E1 COM1
PN N/A
HWVer N/A
FWVer R3.4.21.0Build16211
Copyright (c), Unicore Communications Inc.
All rights reserved.
#CFGSYS,H101
#CFGMSG,0,0,1
#CFGMSG,0,1,1
#CFGMSG,0,2,1
#CFGMSG,0,3,1
#CFGMSG,0,4,1
#CFGMSG,0,5,1
#CFGMSG,0,6,0
#CFGMSG,0,7,0
#CFGNAV,1000,1000,3
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$GNRMC,094735.00,V,,,,,,,,,,N,V*15
$GNGGA,094735.00,,,,,0,00,99.99,,,,,,*74
$GNGLL,,,,,094735.00,V,N*58
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,01,26,,,36,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,094741.00,V,,,,,,,,,,N,V*16
$GNGGA,094741.00,,,,,0,00,99.99,,,,,,*77
$GNGLL,,,,,094741.00,V,N*5B
$GNRMC,094735.00,V,,,,,,,,,,N,V*15
$GNGGA,094735.00,,,,,0,00,99.99,,,,,,*74
$GNGLL,,,,,094735.00,V,N*58
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,01,26,,,36,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,094741.00,V,,,,,,,,,,N,V*16
$GNGGA,094741.00,,,,,0,00,99.99,,,,,,*77
$GNGLL,,,,,094741.00,V,N*5B
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,04,26,,,37,01,,,35,10,,,29,16,,,28,0*61
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,094747.00,V,,,,,,,050924,,,N,V*1A
$GNGGA,094747.00,,,,,0,00,99.99,,,,,,*71
$GNGLL,,,,,094747.00,V,N*5D
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,04,26,,,36,01,,,34,10,,,30,16,,,28,0*69
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,094753.00,V,,,,,,,050924,,,N,V*1F
$GNGGA,094753.00,,,,,0,00,99.99,,,,,,*74
$GNGLL,,,,,094753.00,V,N*58
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,04,26,,,36,01,,,33,10,,,29,16,,,27,0*69
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,094759.00,V,,,,,,,050924,,,N,V*15
$GNGGA,094759.00,,,,,0,00,99.99,,,,,,*7E
$GNGLL,,,,,094759.00,V,N*52
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,04,26,,,30,01,,,30,10,,,32,16,,,29,0*68
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
```

I don't think this is valid data.

katoury, 2024-09-05:
> https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS_Use_TinyGPS
@lewisxhe Hi, This on seems good to adapt but, I am stuck with the following because all I tried do not work : I cannot find the way to include read sms and send back only if it contains some word. Any advice on this ?

lewisxhe, 2024-09-06:
> I also have problems with GPS on LILYGO_T_CALL_A7670_V1_0
>
> Run the example
>
> https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS_Use_TinyGPS
>
> The message
>
> ``` [...]

This redirects the GPS messages to the AT port for TinyGPS to decode. If you enable AT debugging (DUMP_AT_COMMANDS) then the messages will be output to the serial port.

lewisxhe, 2024-09-06:
> > https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SendLocationFromSMS_Use_TinyGPS
> > @lewisxhe Hi, This one seems good to adapt but, I am stuck with the following because all I tried do not work while I did it well with the SendLocationFromSMS.ino example : but here I cannot find the way to include reading sms and sending back one only if it contains some words. Any advice on this ?

I don't understand what it means. Isn't it sending the location via SMS? To read SMS, see another example https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/ReadSMS

putthipongta, 2024-09-06:
Hi, I have a problem with GPS. I use LilyGo-T-A7670E with GPS antenna and 4G SIM card and run this example. 
[](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/GPS_NMEA_Output)

serial monitor
`Start modem...
Modem starting...AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT

+CPIN: READY
AT

*ATREADY: 1
AT

OK
Enabling GPS/GNSS/GLONASS
AT+CGNSSPWR=1
AT

OK
AT

OK
AT

OK
AT+CGNSSPWR=1

OK

SMS DONE

+CGEV: EPS PDN ACT 1

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200
AT+CGNSSIPR=115200

OK
AT+CGNSSMODE=3
AT+CGNSSMODE=3

OK
AT+CGNSSNMEA=1,1,1,1,1,1,0,0
AT+CGNSSNMEA=1,1,1,1,1,1,0,0

OK
AT+CGPSNMEARATE=1
AT+CGPSNMEARATE=1

OK
AT+CGNSSTST=1
AT+CGNSSTST=1

OK
AT+CGNSSPORTSWITCH=0,1
AT+CGNSSPORTSWITCH=0,1

OK
Next you should see NMEA sentences in the serial monitor
AT+CGNSSPWR?
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1

OK
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79

PB DONE
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79`

lewisxhe, 2024-09-06:
> Hi, I have a problem with GPS. I use LilyGo-T-A7670E with GPS antenna and 4G SIM card and run this example.
>
> serial monitor `Start modem... Modem starting...AT AT AT AT . AT AT AT AT . AT AT
>
> +CPIN: [...]

Please see the reply above

lewisxhe, 2024-09-06:
Reply again, after turning on GPS NMEA output, GPS NMEA data will be redirected to the AT port and decoded by TinyGPS. At this time, if DUMP_AT_COMMANDS is turned on, all data will be output to the serial port. Please do not copy the NMEA output sentence to the issue and say that it is invalid.

The example decodes the data through TinyGPS. If the positioning is successful, the GPS data redirection will be turned off by disableNMEA. Newbies, please use the https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/GPS_BuiltIn example. I only demonstrate the GPS NMEA example without explaining it. If you don’t understand, please see here to learn what NMEA sentences are. https://en.wikipedia.org/wiki/NMEA_0183

As long as the GPS has NMEA sentence output, it means that the GPS is in working condition. If you cannot lock the position, you should check whether the antenna is connected or whether it is outdoors. Please do not locate indoors.

lewisxhe, 2024-09-06:
Please do not perform GPS positioning indoors.

closed 2024-09-06 by lewisxhe