Xinyuan-LilyGO/LilyGo-Modem-Series #119: GNSS/GPS not working

SamJ89, 2024-08-13:
Hello,

I bought a LILYGO T-SIM7670G-S3 ESP32-S3 4G LTE Wireless Cellular IOT Devices Low-power Long-range Communication with GPS TTGO Development Board from the Lilygo amazon store. I have tried various example codes to get the GPS to work but I can't seem to get it working. the unit is placed outside with the antenna pointing to the sky.

Which boards have GPS capability built in, as this is not clear in the main readme page 
[image]

lewisxhe, 2024-08-14:
See [FAQ](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX#6%EF%B8%8F%E2%83%A3-faq)  Second point

SamJ89, 2024-08-14:
Explain how to perform "First, determine whether the purchased board model supports GPS function, regardless of whether the physical board contains a GPS antenna interface." 

If a board is installed with a GPS antenna mount, sold with an antenna, and your own shop lists it as having GPS on amazon, it would be reasonable to assume this is a function. explain how i confirm i verify the board has the capability before i upload the NMEA bin

lewisxhe, 2024-08-15:
If the GPS antenna is included, it usually has GPS function. You can perform the second step to test whether there is NMEA message output.