Xinyuan-LilyGO/LilyGo-Modem-Series #120: LILYGO T-A7670G doesn't work on battery

4egokavo, 2024-08-13:
The battery is fully charged and inserted into the board.
When the USB cable is disconnected, the voltage at the ends of the battery is 4.2V.
When turned on, there is a short activity - a blinking red LED.
Battery switch in the ON position.

I tried to test by sketch
```c
#define BOARD_PWRKEY_PIN   (4)
#define BOARD_ADC_PIN      (35)
#define BOARD_POWERON_PIN  (12)
#define BOARD_MISO_PIN     (2)
#define BOARD_MOSI_PIN     (15)
#define BOARD_SCK_PIN      (14)
#define BOARD_SD_CS_PIN    (13)
#define BOARD_BAT_ADC_PIN  (35)

#include <esp_adc_cal.h>
#include <Arduino.h>
#include <FS.h>
#include <SD.h>
#include <SPI.h>

void setup() {
  Serial.begin(115200);

#ifdef BOARD_POWERON_PIN
  pinMode(12, OUTPUT);
  digitalWrite(12, HIGH);
#endif

  SPI.begin(BOARD_SCK_PIN, BOARD_MISO_PIN, BOARD_MOSI_PIN);
  if (!SD.begin(BOARD_SD_CS_PIN)) {
      Serial.println("Card Mount Failed");
      return;
  }
  if (SD.cardType() == CARD_NONE) {
      Serial.println("No SD card attached");
      return;
  }
}

uint16_t batteryVoltage() {
  esp_adc_cal_characteristics_t adc_chars;
  esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);
  return esp_adc_cal_raw_to_voltage(analogRead(BOARD_BAT_ADC_PIN), &adc_chars) * 2;
}

uint16_t battery;
bool power;
unsigned long timeStamp;
void loop() {
  if (millis() - timeStamp > 1000) {
    timeStamp = millis();

    power = analogRead(BOARD_POWERON_PIN) > 0;
    battery = batteryVoltage();

    String message = String(millis()) + "|" + String(battery) + "|" + String(power) + "\n";
    Serial.print(message);

    File file = SD.open("/battery.txt", FILE_APPEND);
    file.print(message);
    file.close();
  }
}
```

The test file `/battery.txt` is filled with records when powered from USB only.

lewisxhe, 2024-08-14:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/508a27e9d92acfa2bfc7f6ece509b73fab7cb8f0/examples/ReadBattery/ReadBattery.ino#L110

4egokavo, 2024-08-14:
> When connected to the USB, the battery voltage data read is not the real battery voltage,

This is another question. I have an issue that the application does not start when the USB is disconnected.

lewisxhe, 2024-08-14:
You should remove the line 
```
power = analogRead(BOARD_POWERON_PIN) > 0; 
```
This sentence will change IO12 from output state to input state, causing the SD card to lose power.

4egokavo, 2024-08-14:
> This sentence will change IO12 from output state to input state, causing the SD card to lose power.

Great catch!, Thank you a lot, it helped.

Maybe you can explain how to check if there is power from the USB or not?

lewisxhe, 2024-08-14:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/commit/df1eb79590ab2af75635c7702cbf3a96812b6974

4egokavo, 2024-08-14:
Awesome! Thank you again.

Looks like `esp_adc_cal_raw_to_voltage` return mV.
I have to modify, like:
```c
uint16_t battery_voltage = esp_adc_cal_raw_to_voltage(analogRead(BOARD_BAT_ADC_PIN), &adc_chars) * 2;

if (battery_voltage > 4200) {
    Serial.println("USB is connected!");
}
```
instead of 
```c
if (battery_voltage > 4.2) {
```

4egokavo, 2024-08-14:
Looks like it depends on battery
I got without USB:
```
4206 mV
4208 mV
4206 mV
4206 mV
```
With USB:
```
4250 mV
4252 mV
```

lewisxhe, 2024-08-14:
This is an unreliable detection method. If the battery voltage is too low, the ADC detection voltage will be pulled down by the battery, resulting in failure to effectively detect whether the USB is connected. It is only suitable for situations where the battery voltage is similar to the USB voltage.

This board is different from the other one. This method cannot be used directly for detection. Currently, it is not possible to detect USB through existing hardware.

closed 2024-08-14 by 4egokavo