Xinyuan-LilyGO/LilyGo-Modem-Series #121: Powering TCall-V1.4 A7670E

MrFloppy70, 2024-08-15:
Hello.
Can someone confirm if the TCall-V1.4 A7670E board can be powered from the 3V3 pin or the 5V pin? I've searched the web for information and there's nothing from LiLy-go on this topic (and practically nothing at all).
Thanks.

lewisxhe, 2024-08-16:
V1.4 ??? Is there such a version? Do you have any photos?

MrFloppy70, 2024-08-16:
This is the version currently on sale:
https://www.lilygo.cc/products/t-call-v1-4?variant=43440642719925

I have not been able to find any picture or pinouts or schematics or any kind of information about this board. Even the "utilities.h" file in the examples is only up to version 1.1.

Thanks.

lewisxhe, 2024-08-16:
OK. There is no V1.4 version, only V1.1. The link name is V1.4, but the actual board is V1.1. You can see it from the silkscreen on the board.

This 5V Pin can be input, it is on the same line as the USB power supply. Note that if you input USB and 5VPin together, the two lines are connected, you can see it in the schematic

3.3V power supply cannot make the module work, only 5V

MrFloppy70, 2024-08-16:
Thank you very much for your reply.
If the board can only be powered from the USB input, it will not work for me. I will look for another one.

closed 2024-08-21 by lewisxhe