Xinyuan-LilyGO/LilyGo-Modem-Series #122: SIM7670G Firmware Update

Hydra-Gituser, 2024-08-16:
Hi,
I am trying to find the latest firmware for the SIM7670G module as used on the Lilygo T-SIM7670G V1.0.

I believe others have tried going direct to SimCom but were referred back to Lilygo.

I have queried the module and here are the results.

P/N:S2-10D1Y-Z327P
SN:MP06240383AEE03
SW:2360B01SIM767XM5A_M

Thanks in advance for your help.

lucioborges, 2024-08-16:
Use [HttpsOTAUpgrade] ?
Install Visual studio code & dependencies, inport https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX.
Platform IO .ini
Select board & Examples
;! SIM7672G and SIM7670G are exactly the same, except for the name change
 default_envs = T-SIM7672G
 src_dir = examples/HttpsOTAUpgrade
 
 Compiling and export to ESP32

Hydra-Gituser, 2024-08-17:
Thanks, I have already seen that sketch. Lines 21 to 35 refer to .bin files for testing only but I am looking for the server where the actual Simcom .bin files are located to see if there is a newer version than what I have. I'm not sure what those test files do and I don't want to risk bricking my board.

lewisxhe, 2024-08-17:
You will need the QR code from the label on your modem

Hydra-Gituser, 2024-08-18:
Hello Lewis.
I have the QR code and I tried to run the sketch but got this response.

SIM card online
Wait for the modem to register with the network. Online registration successful

Registration Status:1
Inquiring UE system information:LTE,Online,505-01,0x8011,148842283,42,EUTRAN-BAND1,100,5,5,-10,-102,-91,7
Network IP:AT+IPADDR+IPADDR: 10.135.32.78
Initialize HTTPS
Get firmware form HTTPS
HTTP get failed ! error code = 706

I believe this error is "Send/receive socket data failed".
Any suggestions about how to proceed?
Thanks.

lewisxhe, 2024-08-19:
HttpsOTAUpgrade only updates the application firmware, that is, the ESP, not the modem. If you want to update the SIM7670 firmware, please take a photo of the QR code on the modem label and send it to me. I need to verify the hardware model to check whether there is a new firmware.

Hydra-Gituser, 2024-08-19:
Thanks, photo sent direct to your email.

lewisxhe, 2024-08-20:
Did you send it? I didn't receive it. Which email address did you send it to?

Hydra-Gituser, 2024-08-20:
I sure did, about this time yesterday, to lewishe@outlook.com and it didn't bounce.
The attachment was 2MB so would that be a problem?

lewisxhe, 2024-08-20:
lucio.xxxxx@gmail.com> Is that your email?

lewisxhe, 2024-08-20:
I only received the text but not the photo, can you send it again?

Hydra-Gituser, 2024-08-20:
No, my email is j...@hy.......cs.com.au
Do you still want me to send again?

lewisxhe, 2024-08-20:
Oh! I found yours in the trash.

Hydra-Gituser, 2024-08-20:
OK, all good.

lewisxhe, 2024-08-20:
Sent to you via email

Hydra-Gituser, 2024-08-20:
Got it, thanks for your help. I will let you know how it goes.

AnlauTriCloud, 2024-08-20:
I have the exact same issue. My hardware model (according to the QA code) is:
P/N:S2-10D1Y-Z327P;SN:P1062350804AE70;IMEI:864643060013622;SW:2360B01SIM767XM5A_M

How do i get the latest firmware?

lewisxhe, 2024-08-20:
The following firmware is only suitable for SIM7670G P/N number S2-10D1Y-Z327P. Other modules cannot be flashed, otherwise it will become a brick.

[dirver](https://drive.google.com/drive/folders/1-7x2z00a5VO7GZS96C6mDupNTBXIh--H?usp=sharing)
[firmware & tools](https://drive.google.com/drive/folders/1T9GLM2fgUVj3BcnCpqmTxiVWUowx2icc?usp=sharing)
[doc]([LilyGO-T-A76XX/docs/update_fw.md at main · Xinyuan-LilyGO/LilyGO-T-A76XX (github.com)](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md))

@AnlauTriCloud

AnlauTriCloud, 2024-08-20:
Hi,

My device has the following serial number, according to the QA code on the chip:

P/N:S2-10D1Y-Z327P;SN:P1062350804AE70;IMEI:864643060013622;SW:2360B01SIM767XM5A_M

So, it seems that it is compatible.

Your colleague mentioned SIM7670xis. Just to understand, what does “xis” stand for? My device is a SIM7670G, where G stands for Global, right?

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

AnlauTriCloud, 2024-08-20:
It looks like it is the wrong guide. The guide shows how to update windows drivers, and open A76XX_A79XX_MADL V1.13 Only for Update.exe. This does not exist in the zip file.

You sent the “FlashTools_V4.0.7” and a zip file with 2 folders:
2374B02SIM767XM5A_M_SIM7670G-MNGV_V202240702
B03V01AG3352B

They both contain a lot of bin files. Which files do I need to flash to the SIM7670?

lewisxhe, 2024-08-20:
They are both similar. If you don't understand, I'll have to take a board and make a video to show you how to operate it.

AnlauTriCloud, 2024-08-20:
That would be great.

I need to know exactly the files that need to be flashed, and how to start the process (to avoid making my SIM7670 into a brick) 😊

lewisxhe, 2024-08-20:
Then you have to wait. I don't have a board right now. I can only do it tomorrow or the day after tomorrow.

AnlauTriCloud, 2024-08-20:
Fair enough.
But maybe you can guide me with just a few questions:

The “Browse” searches for .ini files. There are none in the zip files you linked to.
Under “Package”, it searches for .binpkg files. Only one exists, named qc_cat1.binpkg. However, selecting that lists no files below.

I can not add “bin” files to the list.

***@***.***

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

lewisxhe, 2024-08-20:
[image]
[image]

lewisxhe, 2024-08-20:
Here is a video introduction sent to me by simcom, you can watch it
[record.zip](https://github.com/user-attachments/files/16671926/record.zip)

AnlauTriCloud, 2024-08-20:
Thanks! I understand how to use it now.

I am using Windows 11, and I am not able to install the Win10 drivers. This means that USB serial devices are filtered out (external com is filtered).

Is there any way of forcing a COM port? Otherwise, I need to find a Windows 10 machine, and try on that instead.

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

lewisxhe, 2024-08-20:
No, win11 is also available.

AnlauTriCloud, 2024-08-20:
Did I get the wrong drivers?

***@***.***

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

lewisxhe, 2024-08-20:
You can't see any ports and devices in your Computer Manager?

AnlauTriCloud, 2024-08-20:
Yes, they are here:
COM5 – COM8.
COM4 is the ESP32.

***@***.***

Windows does not allow me to update the drivers. It says that the best drivers are already installed.

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

lewisxhe, 2024-08-20:
Can you take a screenshot?

AnlauTriCloud, 2024-08-20:
There was a screenshot in the last mail, but it was probably filtered out.

Here is a link instead:
https://anlauingestiontest.blob.core.windows.net/images/USBDevices.jpg?sp=r&st=2024-08-20T09:09:40Z&se=2025-08-20T17:09:40Z&spr=https&sv=2022-11-02&sr=b&sig=W72tN27h5XcuL6OCOLjoMGf7C8YlamhjddGxkg0XyaE%3D

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

lucioborges, 2024-08-20:
> [lucio.xxxxx@gmail.com](mailto:lucio.xxxxx@gmail.com)> Is that your email?

yes

lewisxhe, 2024-08-21:
[SIM7670G Firmware Upgrade YouTu](https://youtu.be/fAtrz_4DfVs)

lewisxhe, 2024-08-21:
> The following firmware is only suitable for SIM7670G P/N number S2-10D1Y-Z327P. Other modules cannot be flashed, otherwise it will become a brick.
> 
> Please note it is **SIM7670G**, **not A7670G**
> 
> * [dirver](https://drive.google.com/drive/folders/1-7x2z00a5VO7GZS96C6mDupNTBXIh--H?usp=sharing)
> * [firmware & tools](https://drive.google.com/drive/folders/1T9GLM2fgUVj3BcnCpqmTxiVWUowx2icc?usp=sharing)
> * [doc](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md)
> 
> @AnlauTriCloud

The driver directory has been updated. The ini file can be updated directly without installation.

AnlauTriCloud, 2024-08-21:
@lewisxhe Thanks, I succeeded in updating the firmware. The video was very helpful.
The trick was to hold the "sim boot" button when powering the board, to ensure that the flash tool could discover it.

0x657373656C6C, 2024-09-13:
I have a slightly different experience. 

Exact same hardware as per QR code: P/N:S2-10D1Y-Z327P;SN:MP06240383AED5C;IMEI:864643060021914;SW:2360B01SIM767XM5A_M

Correct firmware from Google Drive, watched and followed video and Flashtool successfully uploads SIM7670G firmware:

[image]

But, when it comes time to try ATDebug again, this is all I get. I cannot send any AT commands:

Start Sketch

QCRDY

That's it, nothing more.

Any insights?

closed 2024-10-28 by github-actions[bot]

majuss, 2025-01-24:
I have this on my QR Code: P/N:S2-10AAL-Z326G;SN:MP0624113952B5C;IMEI:862733060741991;SW:A50C4B11A7600M7
Is the most recent firmware installed?

learhy, 2026-09-13:
If anyone lands here trying to update the SIM7670G firmware without a Windows machine: there's now an open-source path that works from macOS and Linux.

I upgraded a SIM7670G-MNGV (B01V01 → 2374B08SIM767XM5A_M, the B08 GNSS-fix build) entirely from macOS: https://github.com/learhy/sim767x-mac-flasher

Key facts that took a while to figure out:

- The module's USB download protocol (EC618 platform) is open-source implemented in openLuat's `ectool` (https://github.com/openLuat/ectool2py) — pure Python, no Windows QDL tool needed.
- The gotcha that breaks *every* official approach on a re-flash: a previously-flashed module's boot ROM auto-runs the resident agentboot, so it REJECTS the standard DLBOOT handshake (`0x2b02d300` → replies `0x2b02d3ee`) even though the flasher agent is alive and the LPC handshake echoes perfectly. SIMCom's Windows QDL tool fails here too (that's why it looks like "Windows required"). The fix is to detect that state and go straight to the LPC burn phase.
- Enter download mode first (BOOT button held + power cycle; device shows as "QCom Boot" 05c6:9330, a single serial port).

Firmware is verified after flash via AT: `AT+CGMR` reports the new version. All three package images (bootloader/AP/CP) burn in ~90 s.