Xinyuan-LilyGO/LilyGo-Modem-Series #125: SIM7670G can not initialize MQTT and HTTP

AnlauTriCloud, 2024-08-20:
My issue is with the SIM7076G, where I am trying to connect to a MQTT broker.

The root cause seems to be, that I am not able to start the MQTT service with AT+CMQTTSTART. No matter what I do, it returns ERROR. I have the same issue with HTTP (AT+HTTPINIT).

Even though the PDP is up, time is set, and I have an IP address. It seems to me that the SSL connection succeeds also, but still the AT+CMQTTSTART command still fails, no matter when I call it. I have tried to move it to different places, before and after the PDP, before and after SSL. Nothing helps. I have read (and re-read) the application notes for the series. 

The PDP should be working since I get an IP address. 
AT+CPSI returns this:
+CPSI: LTE,Online,238-01,0xD48,5314856,394,EUTRAN-BAND7,2850,5,5,-11,-109,-97,14
AT+NETOPEN also succeeds.
Setting AT+CMEE=2 does not provide any information.

Any help on getting the MQTT and HTTP to work is greatly appreciated.

lewisxhe, 2024-08-20:
Provide all the logs so I can analyze the problem

AnlauTriCloud, 2024-08-20:
This is an example of the issue:

Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: V1.9.01
OK

ATI Manufacturer: SIMCOM INCORPORATED Model: SIM7670G-MNGV Revision: V1.9.01 IMEI: XXXXXXXXXXXXXXX
IMSI: AT+CIMI
AT+CIMI
XXXXXXXXXXXXXXX
OK

AT+CPIN?
+CPIN: READY
OK

AT+CREG?
+CREG: 0,5
OK

AT+CEREG?
+CEREG: 0,5
OK

AT+CPSI?
+CPSI: LTE,Online,238-01,0xD48,5314856,394,EUTRAN-BAND7,2850,5,5,-11,-109,-97,14
OK

AT+CGDCONT=1,"IP","TM"
OK

AT+CGACT=1,1
OK

AT+CGACT?
+CGACT: 1,1
OK

AT+CTZU=1
OK

AT+CNTP="time.nist.gov",32
OK

AT+CCLK?
+CCLK: "24/08/16,12:04:25+08"
OK

AT+CREG?
+CREG: 0,5
OK

Registration status:
Network registration successful, currently in roaming mode

AT+CGPADDR=1
+CGPADDR: 1,10.59.245.255
OK

AT+CDNSCFG="8.8.8.8","8.8.4.4"
OK

Modem initialized
System time set to: Fri Aug 16 12:04:30 2024

Connecting to Azure

Uploading DigiCert Global Root G2
AT+CCERTDOWN=cacert.pem,1295
>-----BEGIN CERTIFICATE-----
MIIDjjCCAnagAwIBAgIQAzrx5qcRqaC7KGSxHQn65TANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
...
8rGOmaFvE7FBcf6IKshPECBV1/MUReXgRPTqh5Uykw7+U0b6LJ3/iyK5S9kJRaTe
pLiaWN0bfVKfjllDiIGknibVb63dDcY3fe0Dkhvld1927jyNxF1WW6LZZm6zNTfl
MrY=
-----END CERTIFICATE-----
OK

AT+CTCPKA=1,2,5,1
OK

AT+CCHSTART
ERROR

AT+CCHADDR
+CCHADDR: 10.59.245.255
OK

AT+CSSLCFG="ignorelocaltime",0,1
OK

AT+CSSLCFG="negotiatetime",0,25
OK

AT+CSSLCFG="sslversion",0,4
OK

AT+CSSLCFG="authmode",0,1
OK

AT+CSSLCFG="cacert",0,"cacert.pem"
OK

AT+CCHSET=1,1
ERROR

AT+CCHSSLCFG=0,0
OK

AT+CCHOPEN=0,"IOTHUB.azure-devices.net",8883,2
+CCHOPEN: 0,1
OK

AT+CCHSET?
+CCHSET: 1,1
OK

AT+CMQTTSTART
ERROR

AT+CMQTTACCQ=0,"DEVICE",1
ERROR

AT+CMQTTSSLCFG=0,0
OK

AT+CMQTTCONNECT=0,"ssl://IOTHUB.azure-devices.net:8883",60,1,"IOTHUB.azure-devices.net/DEVICE/?api-version=2018-06-30","SharedAccessSignature sr=IOTHUB.azure-devices.net%2Fdevices%2FDEVICE&sig=rc3c2K%2BHF0l%2Bu16lv0DEi6lXqgSqIqouVgRwkr2tKKU%3D&se=1723810204"
+CMQTTCONNECT: 0,12
ERROR

lewisxhe, 2024-08-21:
Have you seen the test video I recorded before? https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/6a07a7df3f181b8e9fe50973eaf0c0ed1cb79e51/examples/MqttsBuiltlnAWS/MqttsBuiltlnAWS.ino#L15

The previous test was normal. From your log, I know that the error code is 12. You can check the manual to know that the error means parameter error.

AnlauTriCloud, 2024-08-21:
@lewisxhe I have modified your example "MqttsBuiltInAWS.ino" to connect to Azure IoT Hub instead. There are some minor changes required to do so:
1. Certificates, of course, needs to be replaced. The ca_cert must be Digicert Global Root G2, and I have generated device key and certificate for my testdevice. I tested that the certificates work in MQTT Explorer, where I am able to connect.
2. The MQTT connection requires a username (and empty password), that I added to the example.
3. The MQTT client id must equal the device name.

Otherwise, the example seems to work with the exception of the last step CMQTTCONNECT. It does not return an error code, but just returns ERROR. If I change something in the configuration, I get an error code (for example 12 or 15).

My question, what version of the MQTT protocol is the SIM7670 implementing? 
The IoTHub requires MQTT version 3.11. Could that be the issue?

Am I able to set the MQTT protocol verion by configuration? The CMQTTCFG can return "version", but the documentation does not explain how to interpret it.

My log looks like this:
`
AT+CREG?
+CREG: 0,5

OK
Network registration successful, currently in roaming mode

AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B02SIM767XM5A_M
SIM767XM5_B02V01_240717
IMEI: 864643060013622

OK
Registration Status:5
AT+CPSI?
+CPSI: LTE,Online,238-01,0x1BBC,11013732,271,EUTRAN-BAND1,300,5,5,-10,-107,-96,9

OK
Inquiring UE system information:LTE,Online,238-01,0x1BBC,11013732,271,EUTRAN-BAND1,300,5,5,-10,-107,-96,9
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR
+IPADDR: 10.59.244.207

OK
+IPADDR: 10.59.244.207
AT+CMQTTSTART
OK

+CMQTTSTART: 0
Connecting to <myIoTHub>.azure-devices.net

AT+CCERTDOWN="ca_cert.pem",1294
>-----BEGIN CERTIFICATE-----
MIIDjjCCAnagAwIBAgIQAzrx5qcRqaC7KGSxHQn65TANBgkqhkiG9w0BAQsFADBh
...
pLiaWN0bfVKfjllDiIGknibVb63dDcY3fe0Dkhvld1927jyNxF1WW6LZZm6zNTfl
MrY=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CCERTDOWN="cert.pem",1244
>-----BEGIN CERTIFICATE-----
MIIDaTCCAlGgAwIBAgIRANcdJGUSFZeWAERbUrPwUSowDQYJKoZIhvcNAQELBQAw
...
9S1D6Dn/xpgbQksRV1qhWnR8P6FsZsHeU+G73UbWspGS3xJYHv/1HTdkAd6SW7+o
a5GxCxXkPfWc3pT+1Q==
-----END CERTIFICATE-----
OK
AT+CSSLCFG="clientcert",0,"cert.pem"
OK
AT+CCERTDOWN="key_cert.pem",1703
>-----BEGIN PRIVATE KEY-----
MIIEvQIBADANBgkqhkiG9w0BAQEFAASCBKcwggSjAgEAAoIBAQDKrkTIfg8rhCbH
...
Uv7s9njlwM0t7yqBEie5F2qOrud2km4F1+f++aXlm9XpIoYA5WlswdKlVPjop+0k
sc7WKTC0HE3FyM4XnISLu3w=
-----END PRIVATE KEY-----
OK
AT+CSSLCFG="clientkey",0,"key_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,2
OK
AT+CMQTTREL=0
+CMQTTREL: 0,20
ERROR

AT+CMQTTACCQ=0,"<myDevice>",1
OK
AT+CMQTTCONNECT=0,"tcp://<myIotHub>.azure-devices.net:8883",60,1,"<myIotHub>.azure-devices.net/<myDevice>/?api-version=2021-04-12",""
ERROR
Failed!
`

lewisxhe, 2024-08-21:
I just checked and it seems that AWS has upgraded to MQTT5. I need to investigate.
[image]

lewisxhe, 2024-08-21:
I just checked that the program works fine. Have you checked your certificates? Test with MQTTX first. Test whether you can connect with your certificates first.

[image]

AnlauTriCloud, 2024-08-21:
I have verified that I can connect using MQTT version 3.1.1.

However, version 5.0 does not succeed.

That is why I wonder which MQTT version the SIM7670 is implementing? And is there a way of choosing the version?

image.png (view on web)<https://github.com/user-attachments/assets/ad28f31e-d76b-4d5b-848b-9ef18da86b05>

AnlauTriCloud, 2024-08-21:
Luckily, the IoTHub requires MQTT version 3.1.1. As far as I can tell, this is the version the SIM7670 uses.

However, I am still not able to connect. It works with MQTTX.

Do you have any examples of connecting to Microsoft Azure IoT Hub or/and DPS?

Kind Regards / Med venlig hilsen
Anders Nordberg Lauritsen

7670 does not support MQTT5.0 version!

lewisxhe, 2024-08-21:
As far as I know, all simcom series currently sold by LilyGo do not support MQTT5.0

AnlauTriCloud, 2024-08-22:
I succeeded in making a connection to Azure IoT Hub.

The required changes to make the SIM7670G connect are:

        // Enable SNI
        thisModem().sendAT("+CSSLCFG=\"enableSNI\",0,1");
        thisModem().waitResponse();

        // Ensure MQTT uses MQTT 3.1.1
        thisModem().sendAT("+CMQTTCFG=\"version\",0,4");
        thisModem().waitResponse();

The default protocol used by the SIM7670 is MQTT 3.1, where IoTHub requires MQTT 3.1.1.

lewisxhe, 2024-08-22:
Great, could you please provide me a pull request for an example that connects to Azure IoT Hub? If possible.

AnlauTriCloud, 2024-08-22:
Sorry, I have done intensive modifications to the codebase, that does not apply to the examples. Also, I have unfinished code for subscribing / publishing data.
I am also removing some of the things I tried, that may or may not be part of the solution.

But if you send me your direct email, I can provide you with a copy of the code, once I am done.

AnlauTriCloud, 2024-08-22:
By the way, I found a bug in mqtt_subscribe in TinyGsmMqttA76xx.h.

The code should be replaced by this:

    bool mqtt_subscribe(uint8_t clientIndex, const char *topic, uint8_t qos = 0)
    {
        if (clientIndex > muxCount) {
            return false;
        }
        thisModem().sendAT("+CMQTTSUB=", clientIndex, ',', strlen(topic), ',', qos);
        if (thisModem().waitResponse(10000UL, ">") != 1) {
            return false;
        }
        thisModem().waitResponse('>');
        thisModem().stream.write(topic);
        thisModem().stream.println();

        // Wait return OK
        if (thisModem().waitResponse(10000UL) != 1) {
            return false;
        }

        thisModem().waitResponse("+CMQTTSUB: ");
        int id = thisModem().streamGetIntBefore(',');
        int status = thisModem().streamGetIntBefore('\n');
        thisModem().stream.flush();
        return id == clientIndex && status == 0;
    }

The previous implementation used CMQTTSUBTOPIC, but that always returned error. The implementation above works for me.

lewisxhe, 2024-08-23:
OK. I am going to start my vacation. I will check after the vacation.

closed 2024-10-07 by github-actions[bot]

cucilo, 2025-04-07:
Hi, when I take code from your git, I have error: class "TinyGsmA7670" has no member "enableNetwork". I used A7670E-FASE

lewisxhe, 2025-04-08:
Obviously, the TinyGSM in lib is not synchronized to your library folder.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/lib/TinyGSM

lewisxhe, 2025-04-08:
Please open a new issue and state the board model you are using and paste all AT interaction logs.