Xinyuan-LilyGO/LilyGo-Modem-Series #127: GPSShield Example Returns INVALID Location and Incorrect Date/Time

georgelviv, 2024-08-21:
Hello!

I recently purchased the A7670G R2 with GPS and tried running the **examples/GPSShield** from the provided examples. However, the output consistently returns an invalid location and an incorrect date/time. Below is the output I receive:

`Location: INVALID  Date/Time: 0/0/2000 00:00:00.00`

I can confirm I have external GPS module on board

[image]

lewisxhe, 2024-08-21:
I added a warning that will appear if GPS is not present.

If there is no warning, it means that GPS is already connected. Please place the board where the sky is visible for positioning.

georgelviv, 2024-08-21:
I've checked outdoor, works great, thanks!

closed 2024-08-21 by georgelviv