Xinyuan-LilyGO/LilyGo-Modem-Series #130: Voice calls

Jvf01, 2024-08-27:
I recently purchased the LILYGO® TTGO T-A7670G/E/SA R2 4G card and I need to know if I can connect a microphone and speaker to make voice calls.
thank you

lewisxhe, 2024-08-29:
See [FAQ -  How to use voice call example?](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX#6%EF%B8%8F%E2%83%A3-faq)

Jvf01, 2024-08-29:
Thanks.  Apparently this card supports voice calls.  Can you send me a schematic for soldering the microphone and speaker?
Thank you

lewisxhe, 2024-08-29:
[image]
There is no connection diagram for the LilyGo, so I can only simply tell you where the solder points are.

Jvf01, 2024-08-29:
Ok.

Thank you, I wil try

closed 2024-08-29 by Jvf01