Xinyuan-LilyGO/LilyGo-Modem-Series #132: GPS Not Working on LilyGO A7670G Board

embedded-systems-lol, 2024-08-27:
I am experiencing issues with the GPS functionality on my LilyGO A7670G board. Despite following the recommended steps to test the GPS, I am unable to get any NMEA sentences output or successful GPS initialization. Below are the details of my troubleshooting process and the output logs. What tha fuck could went wrong ?

**Board Details:**
Model: LilyGO A7670G
Module Label Image: 
Firmware: GPS_NMEA_Output
Platform: (e.g., PlatformIO)

**Steps Taken:**
Built and uploaded the GPS_NMEA_Output firmware to the board.
Monitored the serial output to check for NMEA sentences.
Attempted multiple times to initialize GPS with the AT+CGNSSPWR=1 command.

**Observed Behavior**
The serial monitor repeatedly returns the following error message:
`AT+CGNSSPWR=1
ERROR
GPS startup failed. Please check whether the board you ordered contains GPS function.
The log also suggests that the board might not support GPS, despite being labeled as a GPS-capable model.`

**log from GPS_NMEA_Output**
[GPS_NMEA_Output.log](https://github.com/user-attachments/files/16770193/GPS_NMEA_Output.log)

ordered from aliexpress this one
[https://www.aliexpress.com/item/1005003036514769.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=21595548853&aff_fcid=e42d3eefa5d8453ba3857f5543e1e580-1724797632039-02719-_mMMfnwo&tt=MG&aff_fsk=_mMMfnwo&aff_platform=default&sk=_mMMfnwo&aff_trace_key=e42d3eefa5d8453ba3857f5543e1e580-1724797632039-02719-_mMMfnwo&shareId=21595548853&businessType=ProductDetail&platform=AE&terminal_id=bbfedbc90fe1413797dcc889367fef32&afSmartRedirect=y](url)
[image]
[image]
[image]
[image]
[image]
[image]

lewisxhe, 2024-08-29:
The A7670G does not have a built-in GPS modem, regardless of whether the motherboard contains a GPS connector. 
If the order includes GPS, then it belongs to the external GPS module program.

embedded-systems-lol, 2024-08-29:
okay I flashed GPSShield example and I am getting this in serial monitor 
is that okay ???? why I aint getting location ?? :(

`This sketch is only suitable for boards without GPS function inside the modem, such as A7670G
Works only with externally mounted GPS modules. If the purchased board includes a GPS extension module it will work, otherwise this sketch will have no effect
If the purchased modem supports GPS functionality, please use examples/TinyGSM_Net_GNSS
Location: INVALID  Date/Time: INVALID 09:50:40.00
Location: INVALID  Date/Time: INVALID 09:50:40.00
Location: INVALID  Date/Time: 8/29/2024 09:50:40.00
Location: INVALID  Date/Time: 8/29/2024 09:50:40.00
Location: INVALID  Date/Time: 8/29/2024 09:50:40.00
Location: INVALID  Date/Time: 8/29/2024 09:50:40.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:41.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:42.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:43.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:44.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:45.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:46.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:47.00
Location: INVALID  Date/Time: 8/29/2024 09:50:48.00
Location: INVALID  Date/Time: 8/29/2024 09:50:48.00 `

lewisxhe, 2024-08-29:
GPS positioning must be in an area with visible sky, not indoors

embedded-systems-lol, 2024-08-29:
Great !! it works now, thanks man

closed 2024-09-02 by lewisxhe