Xinyuan-LilyGO/LilyGo-Modem-Series #133: SIM7670G GPS reporting wrong hemisphere

Hydra-Gituser, 2024-09-02:
I have a T-SIM7670G and my sketch reads some voltages and GPS position and sends it to Adafruit IO by MQTT. Everything works but my location on the map is wrong. The longitude is correct but it puts me at +34 latitude instead of -34. I am using TinyGSMClient (forked version and updated) and I have defined the correct board in utilities. When I look at the response from AT+CGNSSINFO in serial monitor, I can see my correct position shown as 34 degrees south but that is not being taken into account when the position is extracted for publishing. If I bypass the code and just publish my actual location, it appears correctly in the dashboard map.

Is this something I have done wrong or is there a problem with the library?

Here is the relevant code:

`void sendReadings()
{
  while (lat2 <= 0 || lon2 <= 0)
  {
    Serial.println("Enabling GPS");
    while (!modem.enableGPS(MODEM_GPS_ENABLE_GPIO)) {
      Serial.print(".");
  }
  Serial.println();
  Serial.println("GPS Enabled");
  // Set GPS Baud to 115200
  modem.setGPSBaud(115200);

  Serial.println("Requesting current GPS location");

  if (modem.getGPS(&fixMode, &lat2, &lon2, &speed2, &alt2, &vsat2, &usat2, &accuracy2,
                 &year2, &month2, &day2, &hour2, &min2, &sec2)) {

  Serial.println("Latitude: " + String(lat2, 8) + "\tLongitude: " + String(lon2, 8));
  } else {
  Serial.println("Couldn't get GPS location, retrying in 20s.");
  delay(20000L);
  }
char *p = sendbuffer;
// add speed value
dtostrf(speed2, 2, 6, p);
p += strlen(p);
p[0] = ','; p++;
 
//concat latitude
dtostrf(lat2, 2, 6, p);
p += strlen(p);
p[0] = ','; p++;
 
//concat longitude
dtostrf(lon2, 3, 6, p);
p += strlen(p);
p[0] = ','; p++;

// concat altitude
dtostrf(alt2, 2, 6, p);
p += strlen(p);
 
// null terminate
p[0] = 0;
 
Serial.print("Sending: "); Serial.println(sendbuffer); // (Speed,Latitude,Longitude,Altitude)
modem.mqtt_publish(0, gps_topic, sendbuffer);
 `
Here is the response (edited to hide exact location):

Enabling GPS
AT+CGDRT=4,1

+CMQTTPUB: 0,0
AT+CGDRT=4,1
OK
AT+CGSETV=4,1
AT+CGSETV=4,1
OK
AT+CGNSSPWR=1
AT+CGNSSPWR=1
OK

GPS Enabled
AT+CGNSSIPR=115200
AT+CGNSSIPR=115200
OK
Requesting current GPS location
AT+CGNSSINFO
AT+CGNSSINFO
+CGNSSINFO: 3,10,00,,,34.xxxxxx,S,138.xxxxxx,E,020924,041428.000,25.6,0.08,249.97,2.32,2.16,0.85,5

OK
Latitude: 34.xxxxxxxx	Longitude: 138.xxxxxxxx
Sending: 0.080000,34.xxxxxx,138.xxxxxx,25.600000
AT+CMQTTTOPIC=0,25

HDC67, 2024-09-02:
GPS is correct. The S afterwards means South i.e. -34. Similarly the E means East.

Hydra-Gituser, 2024-09-02:
Yes, I looked at the file for my module (SIM7672) and it has the same bit of code. I can see where the N/S field and the E/W field is read but I don't see where that information is used to insert the - sign for southern hemisphere locations. I assume I would get the same error for locations to the west of the meridian?
I have not done any specific coding as such (due to my lack of experience), I am just using and adapting examples that appear to work for others. Maybe I need to allow for an extra character when I add the latitude reading to the buffer?
It would be interesting to know if anyone else has this issue.

HDC67, 2024-09-02:
Look at some of the other models' files - the code is much the same for most of them. You'll see some look for the N or the E properly and multiply by 1 or by -1 to get the right sign.

Seems fundamentally broken as is.

lewisxhe, 2024-09-02:
Yes, that's right. I didn't handle the southern and northern hemispheres. This was ignored during the previous test because I was in the northern hemisphere.

If the user is in the southern hemisphere, the coordinates will be wrong. I will fix it later.

lewisxhe, 2024-09-02:
The patch has been submitted, please update and then test

Hydra-Gituser, 2024-09-03:
Great, thanks for your quick work, all good now!
[image]

lewisxhe, 2024-09-04:
Thanks for testing, now you may be the first user in the southern hemisphere to encounter this bug, as no one has reported this bug to me before

Hydra-Gituser, 2024-09-04:
Interesting, I think the examples I saw were from India and Germany so they would not have this issue. I assume users to the west of 0 degrees would have had a similar problem?
I want to ask about visible/used satellites, should I open a new issue?

lewisxhe, 2024-09-04:
It can be seen that the satellite did not export it in the original TinyGSM, so I don't have this one either. What I am doing now is just to make it compatible, and I don't have extra time to add new functions, or I can make it stronger in extra time.

Hydra-Gituser, 2024-09-04:
Thanks, that's OK, it is not a big deal for me as I can see the numbers in the raw data and I don't need that info for my application.

closed 2024-09-10 by lewisxhe