Xinyuan-LilyGO/LilyGo-Modem-Series #135: MQTT not working

EnziinSystem, 2024-09-06:
My device:

```
#define LILYGO_T_CALL_A7670_V1_0
```

Step 1:  Copy all folders in [lib folder](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/lib) to Arduino library folder (e.g. C:\Users\my\Documents\Arduino\libraries)
And NOT update libs when I start the Arduino IDE.

Step 2:  Run the example [MqttsBuiltlnAuth](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/MqttsBuiltlnAuth)

The configure:

```
// MQTT details 
const char *broker = "mqtt-s1.gowest.cloud";
const uint16_t broker_port = 1883;
const char *broker_username = "dev";
const char *broker_password = "*********";
const char *clien_id = "mqttx_47919b19";

const char *subscribe_topic = "GsmMqttTest/subscribe";
const char *publish_topic = "GsmMqttTest/publish";

// Current connection index, range 0~1
const uint8_t mqtt_client_id = 0;
uint32_t check_connect_millis = 0;
```
Step 3: Check the connection with [MQTTX](https://mqttx.app/downloads) successfully

[image]

Step 4: Run MqttsBuiltlnAuth example

The logs:

```
12:29:07.016 -> Start Sketch
12:29:09.958 -> Start modem...
12:29:09.958 -> AT
12:29:10.259 -> AT
12:29:10.516 -> AT
12:29:10.816 -> AT
12:29:11.116 -> .
...
12:29:25.964 -> PB DONE
12:29:25.964 -> AT+IPADDR

12:29:25.964 -> +IPADDR: 21.137.125.43
12:29:25.964 -> 
12:29:25.964 -> OK
12:29:25.964 -> Network IP:21.137.125.43
12:29:25.964 -> AT+CMQTTSTART
12:29:25.964 -> AT+CMQTTSTART

12:29:25.964 -> OK
12:29:25.964 -> 
12:29:25.964 -> +CMQTTSTART: 0
12:29:26.980 -> Connecting to mqtt-s1.gowest.cloudAT+CSSLCFG="authmode",0,0
12:29:26.980 -> AT+CSSLCFG="authmode",0,0

12:29:26.980 -> OK
12:29:26.980 -> AT+CMQTTREL=0
12:29:26.980 -> AT+CMQTTREL=0

12:29:26.980 -> +CMQTTREL: 0,20
12:29:26.980 -> 
12:29:26.980 -> ERROR
12:29:26.980 -> AT+CMQTTACCQ=0,"mqttx_47919b19",1
12:29:27.015 -> AT+CMQTTACCQ=0,"mqttx_47919b19",1

12:29:27.015 -> OK
12:29:27.015 -> AT+CMQTTCFG="version",0,4
12:29:27.015 -> AT+CMQTTCFG="version",0,4

12:29:27.015 -> OK
12:29:27.015 -> AT+CMQTTCONNECT=0,"tcp://mqtt-s1.gowest.cloud:1883",60,1,"dev","pass"
12:29:27.015 -> AT+CMQTTCONNECT=0,"tcp://mqtt-s1.gowest.cloud:1883",60,1,"dev","pass"

12:29:28.183 -> +CMQTTCONNECT: 0,32
12:29:28.183 -> 
12:29:28.183 -> ERROR
12:29:28.183 -> Failed!
```

I have not yet successfully run any of the MQTT related examples in this repository.

EnziinSystem, 2024-09-06:
Even the simplest MQTT example has a compile error. Did you not click the compile button before git push?

[MqttClient](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/MqttClient)

The logs:

```
C:\Users\my\Desktop\GPS\LilyGO-T-A76XX\examples\MqttClient\MqttClient.ino:105:15: error: cannot declare variable 'client' to be of abstract type 'TinyGsmA7670::GsmClientA7670'
  105 | TinyGsmClient client(modem);
      |               ^~~~~~
In file included from c:\Users\my\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:66,
                 from C:\Users\my\Desktop\GPS\LilyGO-T-A76XX\examples\MqttClient\MqttClient.ino:80:
c:\Users\my\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientA7670.h:91:9: note:   because the following virtual functions are pure within 'TinyGsmA7670::GsmClientA7670':
   91 |   class GsmClientA7670 : public GsmClient {
      |         ^~~~~~~~~~~~~~
In file included from C:\Users\my\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/Arduino.h:197,
                 from C:\Users\my\AppData\Local\Temp\arduino\sketches\0450E4EFCA56BC5BF72AF8FB6E33D0DD\sketch\MqttClient.ino.cpp:1:
C:\Users\my\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:\Users\my\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~

exit status 1

Compilation error: cannot declare variable 'client' to be of abstract type 'TinyGsmA7670::GsmClientA7670'
```

lewisxhe, 2024-09-06:
When you find an error, check your configuration and the reason for the error instead of complaining about it here with a questioning attitude.
[image]

All examples have passed the compilation check of platformio. Please report your usage environment

EnziinSystem, 2024-09-06:
I use the Ardruino IDE 2.3.2 and I flow guide in the Readme.

lewisxhe, 2024-09-06:
[image]

lewisxhe, 2024-09-06:
MqttsBuiltlnAuth is running normally here. I can only use the test.mosquitto.org public server for testing. I can't help with other private servers. Please debug it yourself based on the returned error reason.

chemmex, 2024-09-08:
I am getting +CMQTTCONNECT: 0,15 on a simple no SSL test using just AT commands. DNS works, AT+CPING works, AT+CIPOPEN opens the socket. The command sequence is exactly as in the MQTT(S) Appnote. What can be missing or wrong?

EnziinSystem, 2024-09-09:
> MqttsBuiltlnAuth is running normally here. I can only use the test.mosquitto.org public server for testing. I can't help with other private servers. Please debug it yourself based on the returned error reason.

The problem is still not solved, I used a public server as you suggested:

```
// MQTT details 
const char *broker = "test.mosquitto.org";
const uint16_t broker_port = 1883;
const char *broker_username = "";
const char *broker_password = "";
const char *clien_id = "mqttx_7bc3c8dc";

const char *subscribe_topic = "GsmMqttTest/subscribe";
const char *publish_topic = "GsmMqttTest/publish";
```

The logs:

```
14:35:32.003 -> ets Jul 29 2019 12:21:46
14:35:32.003 -> 
14:35:32.003 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:35:32.003 -> configsip: 0, SPIWP:0xee
14:35:32.003 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:35:32.003 -> mode:DIO, clock div:1
14:35:32.003 -> load:0x3fff0030,len:4916
14:35:32.003 -> load:0x40078000,len:16436
14:35:32.068 -> load:0x40080400,len:4
14:35:32.068 -> ho 8 tail 4 room 4
14:35:32.068 -> load:0x40080404,len:3524
14:35:32.068 -> entry 0x400805b8
14:35:32.623 -> Start Sketch
14:35:35.486 -> Start modem...
14:35:35.486 -> AT
14:35:35.809 -> AT
14:35:36.126 -> AT
14:35:36.385 -> AT
14:35:36.685 -> .
14:35:36.685 -> AT
14:35:36.985 -> AT
14:35:37.285 -> AT
14:35:37.585 -> AT
14:35:37.885 -> .
14:35:37.885 -> AT
14:35:52.118 -> PB DONE
14:35:52.118 -> AT+IPADDR
14:35:52.118 -> +IPADDR: 26.63.27.240
14:35:52.118 -> 
14:35:52.118 -> OK
14:35:52.118 -> Network IP:26.63.27.240
14:35:52.118 -> AT+CMQTTSTART
14:35:52.118 -> AT+CMQTTSTART
14:35:52.118 -> OK
14:35:52.118 -> 
14:35:52.118 -> +CMQTTSTART: 0
14:35:53.097 -> Connecting to test.mosquitto.orgAT+CSSLCFG="authmode",0,0
14:35:53.131 -> AT+CSSLCFG="authmode",0,0
14:35:53.131 -> OK
14:35:53.131 -> AT+CMQTTREL=0
14:35:53.131 -> AT+CMQTTREL=0
14:35:53.131 -> +CMQTTREL: 0,20
14:35:53.131 -> 
14:35:53.131 -> ERROR
14:35:53.131 -> AT+CMQTTACCQ=0,"mqttx_7bc3c8dc",1
14:35:53.131 -> AT+CMQTTACCQ=0,"mqttx_7bc3c8dc",1
14:35:53.131 -> OK
14:35:53.131 -> AT+CMQTTCFG="version",0,4
14:35:53.131 -> AT+CMQTTCFG="version",0,4
14:35:53.131 -> OK
14:35:53.131 -> AT+CMQTTCONNECT=0,"tcp://test.mosquitto.org:1883",60,1,"",""
14:35:53.166 -> AT+CMQTTCONNECT=0,"tcp://test.mosquitto.org:1883",60,1,"",""
14:35:53.166 -> ERROR
14:35:53.166 -> Failed!
```

EnziinSystem, 2024-09-10:
It was a matter of using SSL or not.

Problem solved.

Thanks @lewisxhe

closed 2024-09-10 by EnziinSystem