Xinyuan-LilyGO/LilyGo-Modem-Series #137: A7670E GPS not working

putthipongta, 2024-09-09:
A7670E with GPS antenna and SIM card and outdoor test
example : GPS_BuiltIn

```
Start modem...
AT
AT
AT
AT
.
AT
AT
+CPIN: READY
[14212] ### Unhandled: +CPIN: READY
AT

*ATREADY: 1
AT

OK

Enabling GPS/GNSS/GLONASS
AT+CGNSSPWR=1
AT

OK
AT+CGNSSPWR=1

OK

SMS DONE

+CGEV: EPS PDN ACT 1

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

PB DONE
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
```

lewisxhe, 2024-09-10:
How to determine whether the modem GPS chip is working properly?

1. First, determine whether the purchased board model supports GPS function, regardless of whether the physical board contains a GPS antenna interface. A7670G does not have built-in GPS function. If the order includes GPS function, it is an external GPS module program. Please run example/GPSShield
2. Upload [GPS_NMEA_Output](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/GPS_NMEA_Output/GPS_NMEA_Output.ino) firmware, then open the serial monitor, adjust the baud rate to 115200, and check whether NMEA sentences are output
3. If there is no NMEA sentence output, or the GPS command fails, please open an issue in github, report the model you are using and upload the module label image (used to determine the modem hardware version, not the model), and provide the hardware log output message
4. If the NMEA sentence output is normal, place the device outdoors or outside the window for positioning test
5. If positioning cannot be performed outdoors, try to replace the antenna or change the position to reposition
6. If the above operations do not work, check the center pin of the GPS antenna Is there a 2.8~3.3V voltage? If there is voltage and data output, it can be judged that the GPS is completely working properly, and there may be a problem with the antenna.

The sixth point is particularly important. Data is output and there is voltage on the center pin of the antenna. This is a manifestation of GPS working. At this time, reporting an issue is powerless. The only thing you can do is to place the GPS where the sky is visible, or replace it with a better antenna for positioning.

putthipongta, 2024-09-10:
1. Board model is A7670E and support GPS function.
2. I tried to use [GPS_NMEA_Output](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/GPS_NMEA_Output/GPS_NMEA_Output.ino) and this is message
```
Modem starting...AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT

+CPIN: READY
AT

*ATREADY: 1
AT

OK

ATE0
OK
ATI
AT

OK
Unable to obtain module information normally, try again
ATE0
AT

OK
ATI
AT

OK
Unable to obtain module information normally, try again
OK
ATI
AT

OK
Unable to obtain module information normally, try again
ATE0
AT

OK
ATI
AT

OK
Unable to obtain module information normally, try again
ATE0

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862205059146268
+GCAP: +CGSM,+FCLASS,+DS

OK
ATI

SMS DONE

+CGEV: EPS PDN ACT 1

OK
Unable to obtain module information normally, try again
ATE0

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862205059146268
+GCAP: +CGSM,+FCLASS9146268
+GCAP: +CGSM,+FCLASS,+DS

OK
ATI

OK
Unable to obtain module information normally, try again
```

lewisxhe, 2024-09-11:
1. Upload the ATDebug sketch, wait for it to fully boot up, and then send AT+SIMCOMATI to report the module firmware version returned to me. Since I also use A7670E-FASE, I can get the module model normally.
```
« Start modem...
Modem starting...AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT

*ATREADY: 1
AT
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: FAETEST_V1.0.0
IMEI: 860470067519045

OK
Model Name:A7670E-FASE
Enabling GPS/GNSS/GLONASS
AT+CGNSSPWR=1

OK

+CPIN: SIM REMOVED

+CPIN: SIM REMOVED

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
AT+CGNSSMODE=3

OK
AT+CGNSSNMEA=1,1,1,1,1,1,0,0

OK
AT+CGPSNMEARATE=1

OK
AT+CGNSSTST=1

OK
AT+CGNSSPORTSWITCH=0,1

OK
Next you should see NMEA sentences in the serial monitor
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1

OK

UC6228CI lite G1L1E1 COM1
PN N/A
HWVer N/A
FWVer R3.4.21.0Build16211
Copyright (c), Unicore Communications Inc.
All rights reserved.
#CFGSYS,H101
#CFGMSG,0,0,1
#CFGMSG,0,1,1
#CFGMSG,0,2,1
#CFGMSG,0,3,1
#CFGMSG,0,4,1
#CFGMSG,0,5,1
#CFGMSG,0,6,0
#CFGMSG,0,7,0
#CFGNAV,1000,1000,3
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$OK*04
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGLL,,,,,,V,N*7A
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGLL,,,,,,V,N*7A
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGLL,,,,,,V,N*7A
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNVTG,,,,,,,,,N*2E

```

3. Delete the code in the figure below, and then upload it again to check

[image]

putthipongta, 2024-09-12:
1. This is my ATDebug message.
```
Trying baud rate 115200
Modem responded at rate:115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
 DISCLAIMER: Entering AT commands without knowing what they do
 can have undesired consiquinces...
***********************************************************

+CGEV: EPS PDN ACT 1

PB DONE
AT

OK
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 862205059146268
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK

``` 

2. I tried to upload NMEA_Output again and hard reset about 4-5 times until I can get the module model normally.
```
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862205059146268
+GCAP: +CGSM,+FCLASS,+DS

OK

OK

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
AT+CGNSSMODE=3

OK
AT+CGNSSNMEA=1,1,1,1,1,1,0,0

OK
AT+CGPSNMEARATE=1

OK
AT+CGNSSTST=1

OK
AT+CGNSSPORTSWITCH=0,1

OK
Next you should see NMEA sentences in the serial monitor
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1

OK
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
$GNRMC,,V,,,,,,,,,,N,V*37
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,1*33
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,3*31
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99,2*30
$GPGSV,1,1,00,0*65
$GAGSV,1,1,00,0*74
$GLGSV,1,1,00,0*79
```  

*At this time I tried to use GPS_Acceleration and hard reset about 4-5 times until I can get the module model and GPS normally. Thank you for support.
```
Start modem...
.
.
.
.
.
[13912] ### Unhandled: +CPIN: READY
.
[14212] ### Unhandled: *ATREADY: 1

Unable to obtain module information normally, try again
Unable to obtain module information normally, try again
Unable to obtain module information normally, try again
[38530] ### Modem: A7670E-FASE
Model Name:A7670E-FASE
SIM card online
Current network mode : AUTO
Wait for the modem to register with the network.Online registration successful

Registration Status:1
Inquiring UE system information:LTE,Online,520-04,0x3FC,41889539,9,EUTRAN-BAND1,250,3,0,10,46,49,0
Network IP:10.104.163.88
Enabling GPS/GNSS/GLONASS

GPS Enabled
GPS acceleration is enabled
 success!!!
Requesting current GPS/GNSS/GLONASS location
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
FixMode:3
Latitude:13.738743	Longitude:100.703163
Speed:27.64	Altitude:-8.10
Visible Satellites:0	Used Satellites:0
Accuracy:3.75
Year:2024	Month:9	Day:12
Hour:2	Minute:4	Second:34
Retrieving GPS/GNSS/GLONASS location again as a string
GPS/GNSS Based Location String:3,16,,00,00,13.7382612,N,100.7032013,E,120924,020438.00,-11.4,28.012,175.92,6.51,3.76,
Disabling GPS
```

closed 2024-10-26 by github-actions[bot]

jawwadcreation, 2025-04-20:
I recently purchased Liliygo A7670G R2 with external GPS
but can't connect with GPS I try many things but not working,

please give some code example for GPS start working, also I try many AT commands but all of those commands return Error.

see this code I have to try at last:

#include <TinyGPS++.h>
#include <HardwareSerial.h>

TinyGPSPlus gps;
HardwareSerial gpsSerial(1); // UART1 for GPS

void setup() {
Serial.begin(115200);
delay(1000);
Serial.println("🛰️ GPS Direct UART Reading Test...");

// 🔁 One line at a time active rakho, baaki comment karo

gpsSerial.begin(9600, SERIAL_8N1, 21, -1); // Try GPIO21
// gpsSerial.begin(9600, SERIAL_8N1, 16, -1); // Try GPIO16
// gpsSerial.begin(9600, SERIAL_8N1, 33, -1); // Try GPIO33
// gpsSerial.begin(9600, SERIAL_8N1, 36, -1); // Try GPIO36 (input only)
}

void loop() {
while (gpsSerial.available()) {
gps.encode(gpsSerial.read());
}

if (gps.location.isUpdated()) {
Serial.println("📍 GPS FIX RECEIVED!");
Serial.print("Latitude: ");
Serial.println(gps.location.lat(), 6);
Serial.print("Longitude: ");
Serial.println(gps.location.lng(), 6);
Serial.print("Satellites: ");
Serial.println(gps.satellites.value());
Serial.println("--------------------");
}

delay(1000);
}