Xinyuan-LilyGO/LilyGo-Modem-Series #138: LILLYGO TTGO A7670E R2 4G network registration problem

Jvf01, 2024-09-10:
Good morning,
I'm having some difficulties installing the LILLYGO TTGO A7670E R2 4G with gps card.
I'm using the Arduino IDE and the SendSMS code example.
I upload the code and get the following 2 situations: 
1 - Apparently the modem cannot register with the network.
2 - StreamDebugger gives an error.

I wilI appreciate your help.

Logs:
situation 1

Start modem...
AT
AT
AT
*ATREADY: 1
AT
OK

Wait SMS Done.

+CPIN: READY

SMS DONESet network apn : umts
AT+CGDCONT=1,"IP","umts"

AT+CGDCONT=1,"IP","umts"

OK
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99
OK
[11] Signal Quality:99
AT+CGREG?
AT+CGREG?
+CGREG: 0,0

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99
OK
[12] Signal Quality:99
AT+CGREG?
AT+CGREG?
+CGREG: 0,0

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99
OK
[13] Signal Quality:99
AT+CGREG?
AT+CGREG?
+CGREG: 0,0

StreamDebugger  error:

Waiting for gdb server to start...[2024-09-10T13:26:02.395Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\joaqu\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.12.0-esp32-20240318/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\joaqu\\Documents\\Arduino\\Bibliotecas\\LilyGO-T-A76XX\\examples\\SendSMS" -f "C:/Users/joaqu/AppData/Local/Programs/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
C:/Users/joaqu/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240318/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "C:/Users/joaqu/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240318/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
[2024-09-10T13:26:02.476Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

lewisxhe, 2024-09-11:
Use this [Network](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/Network) sketch to monitor serial output messages and then report back to me

Jvf01, 2024-09-11:
Start Sketch
Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
*ATREADY: 1
AT

OK

AT+CPIN?
AT+CPIN?
+CME ERROR: SIM failure

+CPIN: READY

SMS DONE
SIM card online
AT+SIMCOMATI
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
IMEI: 862771075501943

OK
AT+CNMP=2
AT+CNMP=2

OK
AT+CNMP?
AT+CNMP?

+CNMP: 2

OK
Current network mode : AUTO
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CSQ
AT+CSQ

+CSQ: 99,99

OK
[13] Signal Quality:99
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CSQ
AT+CSQ

+CSQ: 99,99

OK
[14] Signal Quality:99
AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CSQ
AT+CSQ

+CSQ: 99,99
OK
[16] Signal Quality:99
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?

Second problem:

Waiting for gdb server to start...[2024-09-11T09:32:15.362Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\joaqu\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.12.0-esp32-20240318/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "d:\\Arduino\\Network" -f "C:/Users/joaqu/AppData/Local/Programs/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
C:/Users/joaqu/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240318/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "C:/Users/joaqu/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240318/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
[2024-09-11T09:32:15.536Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

Jvf01, 2024-09-11:
I ididn,t putthe network apn..... i wil test again and send you the results.
forget the previous comment

lewisxhe, 2024-09-13:
Have you connected the LTE antenna? It looks like there is no signal at all

Jvf01, 2024-09-13:
No, i didn,t connect the LTE antenna. Yes my carda support 4G an 5G.
I will try with the LTE antenna connected and guve you feed back

Jvf01, 2024-09-13:
with the LTE antenna connected i have a new situation:

AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1
AT+CGREG?
Registration Status:-1

lewisxhe, 2024-09-13:
Provide full logs, which should include CSQ signal quality

Jvf01, 2024-09-13:
After a few attempts the situation is this:
1 - the "Registration Status:-1" error did not occur again;
2 - It turned on once.... it never happened again;
3 - Now it gives the error:

AT
AT
AT
AT

*ATREADY: 1
[10922] ### Unhandled: *ATREADY: 1
AT
AT

OK
Wait SMS Done.

+CPIN: READY

SMS DONESet network apn : umts
AT+CGDCONT=1,"IP","umts"

AT+CGDCONT=1,"IP","umts"

OK
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,0

OK
AT+CSQ
AT+CSQ

+CSQ: 99,99

OK
[12] Signal Quality:99
AT+CGREG?
AT+CGREG?

+CGREG: 0,3

OK
Network registration was rejected, please check if the APN is correct

PB DONE

lewisxhe, 2024-09-13:
From the log, it seems that your SIM card was rejected by the operator when you registered it. Is umts the correct APN of the SIM card operator you are using?

Jvf01, 2024-09-13:
yes, it is. "umts" is correct.
For once the modem registered well with the sketch "network"

lewisxhe, 2024-09-13:
If you still cannot register, you need to contact the operator to ask the reason.

Jvf01, 2024-09-13:
thank you

 What is the meaning of  "+CME ERROR: 13" ?

Start Sketch
Start modem...
AT

*ATREADY: 1
AT

OK

AT+CPIN?
AT+CPIN?
+CME ERROR: 13

+CPIN: READY

SMS DONE
SIM card online
AT+SIMCOMATI
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
IMEI: 862771075501943

OK
AT+CNMP=2
AT+CNMP=2

OK
AT+CNMP?
AT+CNMP?

lewisxhe, 2024-09-13:
This error can be ignored. It is reported when testing whether there is a SIM card. It is because it is not detected at the beginning.

Jvf01, 2024-09-13:
ok, thank you

Jvf01, 2024-09-13:
Sim card solved and It' working.
Only remain the debugger problem but it's not so important.

lewisxhe, 2024-09-14:
A7670G ESP32 does not have built-in USB JTAG and cannot be debugged

Jvf01, 2024-09-14:
The board that i have is the A7670E.., not the A7670G.
With the sketch READSMS how can i know that a SMS is arrived.
There are a way to be alerted when a new message arrived?

Jvf01, 2024-09-14:
The board that i have is the A7670E.., not the A7670G.
With the sketch READSMS how can i know that a SMS is arrived.
There are a way to be alerted when a new message arrived?

I saw the  message in serial monitor "+CMTI: "SM",1" each time I send a message to the board.

closed 2024-09-17 by Jvf01