Xinyuan-LilyGO/LilyGo-Modem-Series #140: SIM7670G n it doesn't communicate with the simon cumuncia

PIERLUIGI1983, 2024-09-12:
Hi, I've been trying for several days but without success, I can't get the 4 cards I bought, model SIM 7670G, to communicate with the TIM operator (Italy). I attach error from serial port 

[115061] ### TinyGSM Version: 0.12.0
[115061] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+CGMI

SIMCOM INCORPORATED

OK
AT+CGMM

SIM7670G-MNGV

OK
[115084] ### Modem: SIMCOM INCORPORATED SIM7670G-MNGV
[115084] ### Modem: SIMCOM INCORPORATED SIM7670G-MNGV
AT+CLTS=1

ERROR
Failed to restart modem, attempting to continue without restarting
Initializing modem...
Connecting to: ibox.tim.it
AT+CIPSHUT

ERROR
AT+SAPBR=3,1,"Contype","GPRS"

ERROR
AT+SAPBR=3,1,"APN","ibox.tim.it"

ERROR
AT+CGDCONT=1,"IP","ibox.tim.it"

OK
AT+CGATT=1

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

ERROR
AT+SAPBR=2,1

ERROR
Connetto a TIM

While it works well with the Sim7000G card, how can I make the SIM 7670G module work too?

Thank you

lewisxhe, 2024-09-13:
The command you are running does not exist in SIM7672. Please run the example in this repository to connect to the Internet.
SIM7000G is not the same as SIM7672G. Some commands are not available. Please check the [manual.](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/datasheet/SIM767X)

PIERLUIGI1983, 2024-09-14:
Hello,
sorry, but I've been trying for days without success, I tried this example but it doesn't work, can you help me, thanks

#include <SoftwareSerial.h>

SoftwareSerial sim7000(10, 11); // Pin 7 (RX), Pin 8 (TX)

void setup() {
  Serial.begin(115200);     ALSO TRIED WITH 9600
  sim7000.begin(115200);   ALSO TRIED WITH 9600

  Serial.println("Inizializzazione Modulo SIM7000...");

  delay(1000);

  sendATCommand("AT");            // Verifica la connessione
  sendATCommand("AT+CPIN?");       // Verifica lo stato del PIN della SIM
  sendATCommand("AT+CSQ");         // Controlla la qualità del segnale
  sendATCommand("AT+CREG?");       // Stato della registrazione di rete
}

void loop() {
  if (sim7000.available()) {
    Serial.write(sim7000.read());
  }
  if (Serial.available()) {
    sim7000.write(Serial.read());
  }
}

void sendATCommand(String command) {
  sim7000.println(command);
  delay(500);
  while (sim7000.available()) {
    Serial.write(sim7000.read());
  }
  Serial.println();
}

FROM THE SERIAL PORT
Inizializzazione Modulo SIM7000...
⸮⸮
⸮     ⸮⸮
T+AP⸮HhE ERPNYM bahT⸮
⸮
⸮
CAN YOU HELP ME UNDERSTAND WHERE I'M WRONG?

lewisxhe, 2024-09-18:
Are you serious? There are so many examples in this repository, why don't you use them? ? Take a look at the description in the homepage README first, which can solve most problems. ATDebug is already here, read the homepage README before using it
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/ATdebug

closed 2024-11-03 by github-actions[bot]