Xinyuan-LilyGO/LilyGo-Modem-Series #141: A7670E not working on battery or external power

kgmuzungu, 2024-09-13:
It seems to be related to [issue 85](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/85)
I am actually on a A7670E modem and not A7608.

On my board it is written V1.2 2022-7-18 with silkscreen.

But it seems a similar problem. With USB connected it works ok. But on battery or external power both red LEDs of the modem come up solid for a second but then go off and stay off.

kgmuzungu, 2024-09-13:
[image]

lewisxhe, 2024-09-14:
If you do not set GPIO12 to HIGH after the ESP is started, that is, when the sketch is started, there is a chance that the power will be lost, causing the modem to shut down. The solution is to set GPIO12 to HIGH during startup.

kgmuzungu, 2024-09-14:
I do set GPIO12 to HIGH during powering up the modem. After setting it to HIGH I never change it. 

`BOARD_POWERON_PIN` is (12)

```
void power_up_modem(){
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
    // Set modem reset pin ,reset modem
    pinMode(MODEM_RESET_PIN, OUTPUT);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL); 
    delay(100);
    digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL); 
    delay(2600);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
}
```

in setup()
```
void setup(){   
    Serial.begin(115200); // Set console baud rate
    while(!Serial){delay(50);}
    Serial.println("[SETUP] Start modem");
    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);
    while(!SerialAT){delay(50);}

    power_up_modem();
    wait_until_modem_is_up();
```

kgmuzungu, 2024-09-14:
🤔 maybe I have to first `power_up_modem()` then connect the Serial and then `modem.testAT()`?

lewisxhe, 2024-09-14:
[image]
Where do you input voltage from?

kgmuzungu, 2024-09-14:
either from 5V const. at the soldered pins ☝️  or from a 18650 battery. I tried both.

lewisxhe, 2024-09-14:
The position in the picture can be used as a charging input, but not as a board power supply.

If the 18650 battery cannot keep turning on, then it is suspected that the board power supply is damaged or the battery power supply voltage is insufficient, the battery discharge current is insufficient, etc.

kgmuzungu, 2024-09-14:
a few questions
* but the charging voltage can be on 5V together with the battery in the board and the board switched on? I presume thats correct?
* it should work with no battery in but with 5V const at VBAT?
* for testing should it work with VBAT connected to VBUS with no USB connected?

kgmuzungu, 2024-09-14:
maybe more to check:
* the supply voltage for the ESP32 is 3,3V, right?
* the supply voltage for the SIMCom modem is 4.2V? 
* where on the PCB could best check if the supply voltage of the modem is steady?

lewisxhe, 2024-09-14:
Mainly check the power supply of the modem, which can generally work normally at around 4~4.2V.
It can be measured at the capacitor position in the figure below
[image]

lewisxhe, 2024-09-14:
> maybe more to check:
> 
> * the supply voltage for the ESP32 is 3,3V, right?
> * the supply voltage for the SIMCom modem is 4.2V?
> * where on the PCB could best check if the supply voltage of the modem is steady?

1. Yes
2. Without battery, the voltage is reduced to 4.2V via USB
3. VBUS and USB are on the same line

kgmuzungu, 2024-09-14:
I have to confirm still but indeed it seems that the battery was too weak. I dont have an oscilloscope at hand right now to see if it dips when something like 2A are drawn. 

My voltage at the SIMCom modem is actually about 3.7V when the board is powered via USB. 3.7 is perfectly fine for the modem. But it is not the 4.2V as stated in the schema. 

My earlier comment/question about can you connect VBAT to V_BUS is obviously wrong. V_BAT is lower than 4.2V and has to be boosted up. Whereas V_BUS is at 5V and has to be reduced to 4.2V. 

I am not sure that it is clear to everyone that you cannot use the "solar_in" without battery to power the board. --> maybe mention it in the readme.md

closed 2024-09-14 by kgmuzungu