Xinyuan-LilyGO/LilyGo-Modem-Series #142: How to Upload Firmware from Server to LilyGO-T-A7670G?

JSS2000, 2024-09-19:
Hi everyone,

I’m currently working on a project using the LilyGO T-A7670G, and I need help with uploading firmware from my server to the board. I’ve managed to set up the server and can serve the firmware file, but I’m running into issues when trying to download and apply the firmware over the air (OTA).

Here are some details:

Server Setup: I'm using Flask to serve the firmware file, and I can access it via HTTP.
Modem Connection: The board connects to the internet using a modem, and I’ve tested the connection successfully.
Current Code: I have a function checkForOTAUpdate() that checks for the firmware file, but it fails during the download process. I’ve attached the relevant parts of my code for reference.
Questions:

What could be the differences in handling OTA updates between the ESP32-S2 and the LilyGO T-A7670G?
Are there specific libraries or configurations I should consider for the T-A7670G to make this work?
Any suggestions on debugging or modifying my current approach would be greatly appreciated!
Thanks in advance for your help!
[void checkForOTAUpdate.txt](https://github.com/user-attachments/files/17061555/void.checkForOTAUpdate.txt)

lewisxhe, 2024-09-20:
Here is an example https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsOTAUpgrade

JSS2000, 2024-09-20:
> Here is an example https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsOTAUpgrade

Thank you very much , your attention, but I have tried just that sketch and no building . The first error is error: 'MODEM_NETWORK_AUTO' was not declared in this scope and  error: 'TinyGsm' {aka 'class TinyGsmSim7672'} has no member named 'getSystemInformation and so. Or even, the problem is TinyGsm is not able to work with LILYGO_T_A7670. Using platformio, and with that routine , I can comunicate with server, but not do upload using thar routine attached.

lewisxhe, 2024-09-20:
Please read the first item in README. Many people make the mistake of not reading the introduction and using it directly.

JSS2000, 2024-09-20:
> Please read the first item in README. Many people make the mistake of not reading the introduction and using it directly.

Thank you, so much!!

closed 2024-11-04 by github-actions[bot]