Xinyuan-LilyGO/LilyGo-Modem-Series #143: I get error backtrace from A7670E board from my code

kod2548, 2024-09-21:
Hello everyone, I'm not good at writing code,
Now my code has a problem from backtrace. My code takes the code of the [https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/Network/Network.ino](url) and develops it to receive the value of the PMS7003 sensor Pin tx 32, rx 33 and SHT31 and displays the value on the LCD1602 screen.
The symptom after getting the IP of the SIM is backtrace.
that is [My code.txt](https://github.com/user-attachments/files/17083756/My.code.txt)
Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT

+CPIN: READY

*ATREADY: 1
AT

OK

AT+CPIN?
AT

OK
AT

OK
AT

OK
AT

OK
AT

OK
AT

OK
AT

OK
AT+CPIN?
+CPIN: READY

OK
SIM card online
AT+CNMP=2
AT+CNMP=2

SMS DONE

OK
AT+CNMP?

+CGEV: EPS PDN ACT 1
AT+CNMP?

+CNMP: 2

OK
Current network mode : AUTO
AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
Online registration successful

AT+CPSI?
AT+CPSI?

+CPSI: LTE,Online,520-03,0x3208,40744562,69,EUTRAN-BAND1,350,3,0,23,69,66,4

OK
Inquiring UE system information:LTE,Online,520-03,0x3208,40744562,69,EUTRAN-BAND1,350,3,0,23,69,66,4
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR

+IPADDR: 10.1x.xx.235

OK
Netwo�-JAi10.1x.xx.235

AT+CGREG?

+CGREG: 0,1

OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK

assert failed: tcpip_send_msg_wait_sem /IDF/components/lwip/lwip/src/api/tcpip.c:449 (Invalid mbox)

Backtrace: 0x40082981:0x3ffb1c80 0x4008b671:0x3ffb1ca0 0x4009095a:0x3ffb1cc0 0x400ec47e:0x3ffb1df0 0x400f8f79:0x3ffb1e20 0x400f8fd9:0x3ffb1e40 0x400ec1be:0x3ffb1e90 0x400d73ee:0x3ffb1eb0 0x400d6c3d:0x3ffb1f10 0x400d4e27:0x3ffb1f70 0x400d68d9:0x3ffb1fa0 0x400d6b53:0x3ffb2010 0x400d6b75:0x3ffb2030 0x400d2f83:0x3ffb2050 0x400d39cc:0x3ffb2210 0x400dd784:0x3ffb2270 0x4008dba2:0x3ffb2290

ELF file SHA256: 95567dbf8e342751

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc

Did I write something wrong?
Thank You

lewisxhe, 2024-09-23:
You should know that you are calling the HTTPClient of esp32, not the HTTPClient of modem.

I changed it to the HTTPClient inside the modem, you can test it.

```
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#include "utilities.h"
#include <TinyGsmClient.h>
#include "Arduino.h"
#include <Wire.h>
#include <Adafruit_SHT31.h>
#include <LiquidCrystal_I2C.h>
#include <HTTPClient.h>

Adafruit_SHT31 sht31 = Adafruit_SHT31();
LiquidCrystal_I2C lcd(0x27, 16, 2); // I2C address for the LCD (0x27), and a 16x2 display

// PMS7003 Config
#define PMS_RX 33
#define PMS_TX 32
HardwareSerial pmsSerial(2);

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

void setup()
{
    Serial.begin(115200);
    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);

    lcd.begin();
    lcd.backlight();
    lcd.clear();

    // Initialize PMS7003
    pmsSerial.begin(9600, SERIAL_8N1, PMS_RX, PMS_TX);

    // Initialize SHT31 sensor
    if (!sht31.begin(0x44)) {
        Serial.println("Couldn't find SHT31");
        while (1) delay(1);
    }

#ifdef BOARD_POWERON_PIN
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

    // Set modem reset pin, reset modem
    pinMode(MODEM_RESET_PIN, OUTPUT);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL); delay(100);
    digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL); delay(2600);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);

    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);

[... 200 lines cut ...]
        }

        modem.https_add_header("Content-Type", "application/x-www-form-urlencoded");

        int httpCode = modem.https_post(postData);
        if (httpCode != 200) {
            Serial.print("HTTP post failed ! error code = ");
            Serial.println(httpCode);
            return;
        }

        // Get HTTPS header information
        String header = modem.https_header();
        Serial.print("HTTP Header : ");
        Serial.println(header);

        // Get HTTPS response
        String body = modem.https_body();
        Serial.print("HTTP body : ");
        Serial.println(body);
#endif

}

```

kod2548, 2024-09-23:
Thank you, so much, this code works.

closed 2024-09-23 by kod2548