Xinyuan-LilyGO/LilyGo-Modem-Series #146: Dubt about schematic on USB-C Connector

alexlopez2612, 2024-09-30:
Hello
I have a ESP32 with A7640G board and I'm studiying the schematic to learn how the board works. By the way, the board works really good.
So, I want to learn how the USB-C connector is connected to the ESP32 uart. And when I see the part of the U11 (Typec) I see that pin A5 is on network CC1, and also I see that B8 is on CC1 network too. Why is this?? Is that correct or is a error on the schematic?
As I see on the connector documentation, SBU2 (B8 pin) it's for low speed signals and I don's understand this connection.
This is the image of what I'm seeing
[image]

Another questions about this U11 are:
- pin B5, In the shematic text says VCONN, but I see that B5 is CC2 on the USB-C documentation. Why is that?
- SUB1 pin, really means "SBU1", right?

Maybe I'm using a old version of the schematic? I'm studing the T-7670X v1.4 and v1.1 and is the same.

Thank you!

Best regards

lewisxhe, 2024-10-01:
Looks like this is wrong, I'll check with the guy who made this board after the holidays

alexlopez2612, 2024-10-01:
> Looks like this is wrong, I'll check with the guy who made this board after the holidays

Hello lewisxhe

Oh! I thought that It was me that I'm starting to study this kind of boards, that are very advance for me.
I'm studiying all this schematic. If I see something wrong, it's ok ask this things over here?

Thanks and best regards!

lewisxhe, 2024-10-04:
Of course, if I see it I'll answer you.

closed 2024-11-17 by github-actions[bot]