Xinyuan-LilyGO/LilyGo-Modem-Series #147: AT+CBC SIMCom modem supply voltage

kgmuzungu, 2024-10-01:
@lewisxhe on my board it is resulting in 0.013V .... can you check? all supply pins of the modem's really connected?

lewisxhe, 2024-10-01:
Please see the schematic diagram, this modem ADC is not connected to any circuit and has no function.

kgmuzungu, 2024-10-01:
There are commands AT+CADC and AT+CADC2 in chapter 11 to read the ADCs. AT+CBC should read the supply voltage.

lewisxhe, 2024-10-04:
This is a software definition. There is no actual hardware circuit connected to the modem's ADC pin, so it is invalid. You can only use the https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/ReadBattery sketch to read the battery voltage.

kgmuzungu, 2024-10-04:
The A7670E has one ADC pin. It can be read with AT+CADC. It is not connected as you wrote. 

I am referring to the command AT+CBC in this issue. It reads the VBAT voltage, the chips supply voltage. This is N O T the ADC pin! On my board it reads 0.013V. Something is not right. Check on your boards.

lewisxhe, 2024-10-04:
OK. I didn't see you said AT+CBC, I only saw AT+CADC in the front. Report the version you use, I will test it and report to SIMCOM

kgmuzungu, 2024-10-04:
pls find all data about my firmware and model [here, issue 117](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117)

lewisxhe, 2024-10-07:
[image]
[image]
It has been confirmed that the conditions obtained by AT+CBC are not met, so 0 is returned.

kgmuzungu, 2024-10-08:
The SIMCom documantations are so poorly written 😢 It might be as you say. But there is no reference that AT+CBC reads the ADC pin. 

Can you provide the source document you are citing?

BTW What does 'perfom partial external pressure' mean?

closed 2024-11-22 by github-actions[bot]