Xinyuan-LilyGO/LilyGo-Modem-Series #148: modem.enableGPS() might get stuck due to misinterpreting the OK response

NeariX67, 2024-10-01:
After starting up the A7670 and enabling the GPS functionality, the library sometimes fails to interpret the "OK" response. It does not happen on every boot but happens fairly often.

I'm using a SoftwareSerial connection with 38400 baud rate on the Waveshare board.
A corresponding failure timeout is missing too.

```
[...]
[5341] ### Modem: A7670E-FASE
[5341] ### Modem: A7670E-FASE
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
Modem initialized
Enabling GPS...
AT+CGNSSPWR=1

OK
[15472] ### Unhandled: OK
AT+CGNSSPWR=1

OK
[26977] ### Unhandled: OK
AT+CGNSSPWR=1

OK
[38482] ### Unhandled: OK
AT+CGNSSPWR=1

OK
[49987] ### Unhandled: OK
AT+CGNSSPWR=1

OK
[61492] ### Unhandled: OK
AT+CGNSSPWR=1

OK
[...]
```

Sometimes it works, but the response is different:
```
+CPIN: READY

OK
Modem initialized
Enabling GPS...
AT+CGNSSPWR=1

OK

SMS DONE

+CGEV: EPS PDN ACT 1

+CGNSSPWR: READY!
GPS enabled
Setup finished
```

lewisxhe, 2024-10-04:
After using AT+CGNSSPWR=1, you must wait for `+CGNSSPWR: READY!`, otherwise GPS is not turned on.
[image]

NeariX67, 2024-10-08:
I think the problem has to do with a restart of the esp32. The Modem is still turned on (as well as GNSS) and thus responds differently when turning it on again.

closed 2024-11-22 by github-actions[bot]