Xinyuan-LilyGO/LilyGo-Modem-Series #149: Problem for the module A7670E to register to the network

jpcasta, 2024-10-02:
Hello,
I bought the LILYGO TTGO T-A7670E development board and the module A7670E was able to register to the network once and after switching off the board alimentation and switching on again, the module did not want to register again!

Here is an excerpt of the log:
20:40:03.469 -> 
20:40:03.469 -> OK
20:40:03.469 -> AT+CSQ
20:40:03.469 -> 
20:40:03.469 -> +CSQ: 22,99
20:40:03.469 -> 
20:40:03.469 -> OK
20:40:03.469 -> [2207] Signal Quality:22
20:40:04.504 -> AT+CGREG?
20:40:04.504 -> 
20:40:04.504 -> +CGREG: 0,0
20:40:04.504 -> 
20:40:04.504 -> OK
20:40:04.504 -> AT+CSQ
20:40:04.504 -> 
20:40:04.504 -> +CSQ: 22,99
20:40:04.504 -> 
20:40:04.504 -> OK
20:40:04.504 -> [2208] Signal Quality:22

Here is the module information:
Manufacturer: INCORPORATED
Model: A7670E-LASE
Revision: A7670M7_V1.11.1
IMEI: 860470062753011
+GCAP: +CGSM,+FCLASS,+DS

Currently the Firmware version is A7670M7_V1.11.1.
On the github web site there is a firmware file that is provided: A011B14A7670M7_A7670E-LASE_V101210903.rar
It is not clear for me if this file contains a more recent version or not.
Is it a more recent version and will it be able to resolve the network registration issue?

Please could you tell me how this issue can be resolved?

Thanks for your help
Regards

lewisxhe, 2024-10-04:
Run https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/Network and send all logs to me

jpcasta, 2024-10-04:
Hello,
Please find attached the log file of the running program.
[A7670_Network.log](https://github.com/user-attachments/files/17254741/A7670_Network.log)

As you can see once the program ended it took 2minutes and half to connect to the network, but it was able to register successfully to the network.
Do you think that is normal it took so many time?

However yesterday, doing some tests I was able to register to the network after submitting the following AT command:
AT+COPS=?
+COPS: (2,"Free","Free","20815",7),(3,"Free","Free","20801",0),(3,"F SFR","SFR","20810",0),(3,"F-Bouygues Telecom","BYTEL","20820",0),(3,"F SFR","SFR","20810",7),(1,"Free","Free","20816",7),(3,"F-Bouygues Telecom","BYTEL","20820",7),(3,"Free","Free","20801",7),,(0,1,2,3,4),(0,1,2) 

Is there an AT command to request the module to force to restart the registration?

Thanks for your help
Regards

lewisxhe, 2024-10-04:
Which country are you in? Long registration time is usually caused by a wide search range. If you can determine the band where you are, you can directly lock the band to quickly access the network.
From your log, it can be seen that your network is registered to the emergency channel.

jpcasta, 2024-10-04:
I am located in France close to Paris.
Yes that is correct, I have seen it registers first to the emergency channel.

AT+CNBP? command returns:
+CNBP: 0X0000000000000180,0X00000000000800D5

and AT+CPSI? returns:
+CPSI: LTE,Online,208-15,0x238C,1894738,370,EUTRAN-BAND3,1675,4,0,7,30,32,-3

Do I need to lock the band using the following AT command?
AT+CNBP=0X0000000000000180,0X00000000000800D5

Thanks

lewisxhe, 2024-10-05:
Can you find out the frequency band used by your operator? Please tell me, I need to ask SIMCOM for a specific command, AT+CNBP. Please do not set it first. If you set it wrong, you will not be able to access the network.

jpcasta, 2024-10-05:
Hello,
Thanks for your reply.
Once the frequency is set, will I be able to unset it?
If not since it is able now to register I can put a loop in the program to wait for the registration.

However to answer to your question please find attached the LTE frequencies of the operator.
It is in French, sorry….(Fréquences —> Frequencies)
[image]

Thanks
Regards

lewisxhe, 2024-10-05:
It is a Chinese holiday now. I will ask SIMCOM in a few days. Please wait.

jpcasta, 2024-10-05:
Ok no problem 😉Thanks

closed 2024-11-23 by github-actions[bot]