Xinyuan-LilyGO/LilyGo-Modem-Series #151: Possible errors on schematic T-A7670X-V1.4 ??

alexlopez2612, 2024-10-03:
Hello
I bought the board that has an ESP32 and a A7670G, to test it and because has an schematic to learn how everything works. The board, it's really good, but I have some dubts about the schematic:

- C35 and  C36 are the same? says both are 22uF/10V, but in the board are not of same size and it's weird that 2 capacitors are the same.
- C55 and  C56 are the same? says both are 22uF/10V, but I think maybe one of them is a 100nF?
- C54 and  C80 are the same? The same thing as the previous 
- C63 and C64 are polarized? because in the board, don't seem to be like this.
- I just see the "BAT" net, only next to Q8, but I cant find this network on other place. Maybe it's a "V_BAT" net?.

For now, it's all the weird things that I see. But this things plus my previous post:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/146
I'm afraid to learn things badly.

Maybe I'm using a wrong version of the schematic? 

I'll be waiting for your answer.

Thank you very much

Best Regards

lewisxhe, 2024-10-05:
Hi @alexlopez2612 

The questions you asked are all hardware related. I am not a hardware engineer for this board, but from the actual object, what you said is basically correct. There are differences between many board schematics and the actual BOM materials in the later production.

C35 and C36 can be seen, one is 22uF and the other should be 100nF

C55 and C56 are the same as above

C63 and C64 are ceramic capacitors on the actual PCB, and there is no polarity

Q8 BAT network is connected to the 55th pin of A7670

L6 model is wrong, the inductance value is correct, but the model is wrong because it is not 0603

Sorry, I can only help you so much, the engineers of this board are very bad, I can't do anything, I can only provide better software service as much as possible

alexlopez2612, 2024-10-07:
Hello lewisxhe
Thank you for your answers!!!! Now I can continue my tests.
Now I understand that this issues are for software issues, I did't knew that. I wrote those issues here because the schematic is on this Git.
So, where I can report this issues? I'm still founding things that seems to be wrong. For example, on the schematic, on the a7670:
- pin 26 says GPIO2, but in the datasheet says GPIO4.
- pin 48 says GPIO3, but in the datasheet says GPIO9.
- pin 53 says GPIO4, but in the datasheet says GPIO10.

So, I'm very frustrated with this... because I bought this board because was a development board with all this documentation that could be usefull for my to develop my own proyect with the things that I could learn from this. But, as you see, is not a very good way to learn with this amount of errors on the schematic.
There's a lot of things that I didn't knew, like all the battery and USB circuits, but now I can't trust in this information and I'm worry to learn some things wrong that lead me to errors on the future.

I don't know if you could put me in contact with the schematic people. Or maybe you consider that this is enough documentation for your costumers. In any case, please, can you tell to your company this? because you should say that the schematics are not ok. I'm thinking for people with non that much experience like me, and this docs can be very dangerous.

Sorry for the long speech, I know this is not your fault, but I don't know another channel to say this this things.

I have to say, I love the board, works really well. There's one thing that don't work, but I have to do more test, is that a variable  RTC_DATA_ATTR losese is value on wake up. But I have to do more tests.

I'll be waiting for your answers.

Again, thank you very much for your time and sorry if you feel offended by any of my words. This is not my intention.
Regards

lewisxhe, 2024-10-08:
You may be reading the wrong manual, you should be reading the [A7670X_R2](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/datasheet/A76XX/A7670C_R2_%E7%A1%AC%E4%BB%B6%E8%AE%BE%E8%AE%A1%E6%89%8B%E5%86%8C_V1.06.pdf) manual.

alexlopez2612, 2024-10-08:
Thank you very much.
I have 2 questions
1- This is valid for A7670G?
2- Is normal that the pin numbers are not correlative? I mean, after pin 17, I see the 89, And after the 91, I see 18. Is this ok?
I show you this in yellow:
[image]

Thank you!!!
Regards

lewisxhe, 2024-10-09:
Please refer to the SIMCOM manual.

jimbobshocks, 2024-10-19:
@alexlopez2612 For losing RTC memory, search for "brownout" in the closed tickets. Its a shame.

alexlopez2612, 2024-10-21:
Hello @jimbobshocks 
I' didn't notice that I have this comment, thank you very much for this!!!!! I'm designing a board based on this one and thanks to you, I will not "import" this error. As I am f
I saw that the info that you told me and undestand that is like the board power off and power on really fast because IO12. But I don't understand why... By the way, for me, this is a really big bug. I never thought that was a hardware problem! I thought that I'm not a good programmer. That's the reason that I did not dig more.
Now, I will continue triying to undestand why is happening and see a way to fix this. But, I ton't know that is too much to ask, but you know what should I change to fix this?
Or, @lewisxhe , as this is a design issue (if I undestand correctly) can you tell me how to solve this? Now, I'm routing my board, so, I can change everything that is necessary to make it work fine, because I need to save data in the RTC memory for my project.
Thanks very much for the help
Regards

closed 2024-12-05 by github-actions[bot]