Xinyuan-LilyGO/LilyGo-Modem-Series #152: T-A7670E got failed in GPS acceleration example

dong100001, 2024-10-04:
Got this in the console

[10936] ### Modem: A7670E-FASE
Model Name:A7670E-FASE
[11957] ### Unhandled: +CME ERROR: 13

*ISIMAID: "A0000000871004FFFFFFFF89"
SIM card online
Current network mode : AUTO
Set network apn : v-internet
Wait for the modem to register with the network.[14] Signal Quality:31
Online registration successful

Registration Status:1
Inquiring UE system information:LTE,Online,452-04,0x2CF8,38587906,201,EUTRAN-BAND3,1700,5,0,12,63,68,1
Network IP:26.224.40.41
Enabling GPS/GNSS/GLONASS

GPS Enabled
GPS acceleration is enabled
 failed !!!
Requesting current GPS/GNSS/GLONASS location
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location

lewisxhe, 2024-10-05:
Turn on DUMP_AT_COMMANDS and re-run the program to see the cause of the error.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/e90f156f7b1ec4f3f0d18c165eaaec192f8bf59a/examples/GPS_Acceleration/GPS_Acceleration.ino#L23

dong100001, 2024-10-07:
Ev

closed 2024-10-07 by dong100001