Xinyuan-LilyGO/LilyGo-Modem-Series #153: T-A7608-S3: HTTPS requests not working, error 715. Firmware update request

chemmex, 2024-10-05:
Model: A7608SA-H
Revision: A50C4B11A7600M7
A7600M7_B11V05_231108

Symptoms similar to #117 , can't establish HTTPS connection, response is 
`+HTTPACTION: 0,715,0`
Connecting in http mode is fine.

Requesting updated firmware from SimCom

lewisxhe, 2024-10-07:
Please provide the modem tag, required to confirm hardware version

chemmex, 2024-10-07:
[image]

lewisxhe, 2024-10-08:
Feedback has been given. It will take some time to get results. I will update here when there are results.

lewisxhe, 2024-10-10:
Please follow the documentation and video to upgrade the modem to the latest firmware, and then use the capture log to provide the log to SIMCOM for analysis.

[A7608SA-H firmware ](https://drive.google.com/file/d/1-LsgDug-Zz_0kU96HlCGtfks3ruvjcQb/view?usp=sharing)

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/capture_log.md

chemmex, 2024-10-14:
Sorry for late reply, it took time to collect the logs. And thanks for the new FW, updated it without problem.

Now it can successfully connect to some of the web domains like below 
```
at+httpinit
OK
at+httppara="URL","https://domain1.cc/"
OK
at+httpaction=0
OK

+HTTPACTION: 0,200,57324
```
[Log 2024-10-14 13;26;48.552.zip](https://github.com/user-attachments/files/17362749/Log.2024-10-14.13.26.48.552.zip)
but still I am getting 715 error while connecting to another domain 
```
at+httpinit
OK
at+httppara="URL","https://domain2.cc/files/OTA/dev/A7608OTA.bin"
OK
at+httpaction=0
OK

+HTTPACTION: 0,715,0
```
[Log 2024-10-14 13;30;01.171.zip](https://github.com/user-attachments/files/17362761/Log.2024-10-14.13.30.01.171.zip)

Trying to connect to github, I get err #706:
```
at+httpinit
OK
at+httppara="URL","https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/153"
OK
at+httpaction=0
OK

+HTTPACTION: 0,706,0
```

As before, http links are opened without problem

lewisxhe, 2024-10-15:
Please follow the link above to grab the modem operation log and send it to me. I need to send it to simcom for analysis.

chemmex, 2024-10-15:
The logs are attached to my previous message

lewisxhe, 2024-10-15:
OK. I will send it to SIMCOM for analysis.

lewisxhe, 2024-10-15:
I can't access this

https://domain2.cc/files/OTA/dev/A7608OTA.bin

This requires special means to access, and China cannot use the GSM network to access it smoothly
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/153

Can you change other access links?

For example:
https://vsh.pp.ua/TinyGSM/logo.txt
Or other websites?

I will send the logo to SIMCOM first and see how they respond

chemmex, 2024-10-15:
Here they are:

https://vsh.pp.ua/TinyGSM/logo.txt - replied with 200 code
[Log 2024-10-15 12;23;45.077.zip](https://github.com/user-attachments/files/17375917/Log.2024-10-15.12.23.45.077.zip)

https://map.airge.ge/ - replied with 715 error
[Log 2024-10-15 12;27;00.433.zip](https://github.com/user-attachments/files/17375924/Log.2024-10-15.12.27.00.433.zip)

lewisxhe, 2024-10-16:
OK, I have sent it to SIMCOM for analysis. I will update you with the results.

lewisxhe, 2024-10-17:
simcom replied, please set AT+CSSLCFG="enableSNI",0,1 before the

at+httppara command, and then try to see if it works

chemmex, 2024-10-17:
It works!
Thanks a lot, closing the issue

closed 2024-10-17 by chemmex