Xinyuan-LilyGO/LilyGo-Modem-Series #154: A7670E GPS powers down randomly

kgmuzungu, 2024-10-06:
At random moments GPS powers down (AT+CGNSSPWR? results in AT+CGNSSPWR=0). When it goes down it also looses the location and takes some seconds to come up again. This is highly annoying. 

This event is accompanied by the following sequence. This sequence occurs **long** (40sec or more) after initialization. 
*ATREADY: 1

+CPIN: READY

SMS DONE

+CGEV: EPS PDN ACT 1

Any clue @lewisxhe whats the cause or how to fix it?

kgmuzungu, 2024-10-06:
Actually, I only recognize it now, this sequence, see below, is occasionally interrupting TCP connections! Causing them to fail.

*ATREADY: 1

+CPIN: READY

SMS DONE

+CGEV: EPS PDN ACT 1

PB DONE

any clue?

lewisxhe, 2024-10-07:
There is no good way. Currently, you can only check whether GPS is turned on during operation. If it is not turned on, turn it on again, just like the example

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/e90f156f7b1ec4f3f0d18c165eaaec192f8bf59a/examples/GPS_NMEA_Output/GPS_NMEA_Output.ino#L131

kgmuzungu, 2024-10-08:
what an advice 🤦‍♂️  . 
or the device could send a postcard with its location, right?

closed 2024-11-22 by github-actions[bot]