Xinyuan-LilyGO/LilyGo-Modem-Series #156: sim7670 command to warm/hot start?

125007007, 2024-10-07:
Im wondering how to actually do a warm/hot start with some saved gps data? the AT command manual only has this: ```AT+CGNSSCMD="$PCAS02,1000*2E"```
So i assume the command to send saved data is something like that? Have googled but seems like those command are proprietary or something as can't find anything. Help would be appreciated.

lewisxhe, 2024-10-08:
Currently, all boards are not connected to the backup power supply, so the premise of hot start is that the modem power supply cannot be turned off, otherwise the hot start will be invalid.
The commands can be found in the AT command manual
[image]

125007007, 2024-10-08:
Ok so if I understand correctly hot start can only happen if power is not interrupted to the board? 
So I couldn't save necessary info to SD card before shutdown and on power up use for hot start.

lewisxhe, 2024-10-09:
Yes

closed 2024-10-12 by 125007007