Xinyuan-LilyGO/LilyGo-Modem-Series #158: the SIM internet and GPS can be used simultaneously on A7670G?

paul218, 2024-10-11:
Hello, I want to buy some modules for car tracking, 
I saw some models that could not connect at the same time to the 2G network and GPS positioning !

Does the module LILYGO® TTGO T-A7670G connect both interfaces at the same time? 
If the reply is not, could you suggest some module, please?

Thanks

lewisxhe, 2024-10-12:
A7670G does not have built-in GPS and needs to rely on an external GPS module.
Other models have built-in GPS

paul218, 2024-10-23:
I mean, the Module ESP32 could manage both interface (the 4G network and the GPS ) at the same time?
Do I need to disconnect one interface before connecting another?

lewisxhe, 2024-10-24:
Both GPS and 4G are accessed through the same uart, they are both inside the A7670, GPS and 4G can be used at the same time

lewisxhe, 2024-10-24:
A7670G? Ok.

A7670G is special without GPS function.

If you buy the A7670G + GPS Shield version, they are on two ports respectively, I think this has been explained in my example and noted in the README

paul218, 2024-10-24:
Ok, Now I am very clear. 
Thank you so much !

closed 2024-12-08 by github-actions[bot]