Xinyuan-LilyGO/LilyGo-Modem-Series #159: A7670E gets a dry spell in connecting to GPRS/LTE

djseban, 2024-10-11:
Hey there, recently I started a little project. I started out with connecting with a SIM to the network. Thing is sometimes the device gets a streak where it connects to the network  as soon as it boots, but then randomly it won't connect at all and continue to do so. During troubleshooting I discovered that switching to a SIM-card that was recently in phone fixes a problem - I turn off the LilyGO, switch the SIM-card and magically it works, but when I restart it won't connect again. What might be a culprit?

lewisxhe, 2024-10-12:
1. How to restart?
2. Is the LTE antenna connected to the correct position?
3. What is returned when sending **AT+CGREG?** and **AT+CSQ** queries when unable to connect to the network?

kgmuzungu, 2024-10-17:
My experience is to repeat the above commands until it eventually connects. My connects within 30s but mostly quicker. Also get the latest firmware, the old frimware versions are buggy

closed 2024-11-30 by github-actions[bot]