Xinyuan-LilyGO/LilyGo-Modem-Series #160: Conflict Between LoRa CS Pin and GSM Modem Reset Pin on LilyGo-TA7670 v1.2

kadrimanaj, 2024-10-13:
I am using a LilyGo-TA7670 v1.2 (2022-07-18), LILYGO 868/915MHz Model: LORA32 (20-3-6) v1.1. I've noticed that the LoRa CS pin is IO5, which also functions as the modem reset pin. This overlap raises some questions about usage:

Can I use each module (LoRa and GSM modem) independently without interference from one another? For example:

Using LoRa while keeping the GSM modem off.

Using both LoRa and the GSM modem simultaneously, such as receiving LoRa data and logging it to a server using the GSM modem.

If simultaneous use is not possible due to this pin conflict, is there any recommended workaround? Would another LoRa shield that uses a different CS pin be a better solution?

Any guidance or suggestions would be greatly appreciated. Thanks in advance!

lewisxhe, 2024-10-14:
If you are using the expansion board above, then removing the MODEM rst is the best option. The modem rst has no effect on this board because of the Power Ctrl.
If you want, you can remove the resistor pointed by the arrow to release ESP32 GPIO5

[image]

kadrimanaj, 2024-10-16:
Thanks, bro i will test and tell you the results 
My logic is to get data with Lora and save it to an SD card 
put sleep GSM and where I need to send data examples every 10 min 
wakeup send data and put GSM to sleep again.

closed 2025-01-01 by github-actions[bot]