Xinyuan-LilyGO/LilyGo-Modem-Series #161: Is T-SimHat compatible with T-A7608?

Teme-V, 2024-10-15:
Is T-SimHat (CAN+RS485) compatible with T-A7608 or just T-A7670 and T-SIM7000G? Documentation does not explain it

lewisxhe, 2024-10-17:
T-SimHat was designed to be compatible with SIM7000G, not A76XX. I checked the schematic and found that the interrupt of T-SimHat's accelerometer conflicts with the solar ADC detection pin of A7670 series.

Teme-V, 2024-10-17:
So there is no option to use T-SimHat with A-7608E-H S3? How about combatibility with T-SIM7670G S3?

lewisxhe, 2024-10-17:
Conflicts with SIM7670G, cannot be used

T-A7608-ESP32 version conflicts with solar ADC

T-A7608-ESP32S3 version can be used

T-A7670X-ESP32 version can be used

I personally recommend using A7670x, it requires less current.

Although A7608X is CAT-4, it is limited by the serial communication rate and has no advantages. It consumes more power and has higher power requirements

Teme-V, 2024-10-17:
Last question (so far) about compatibility. I dont have must need for the battery
is this T-SimHat & T-PCIE with A7608E-H suitable combination?

lewisxhe, 2024-10-17:
PCIE?
It's not compatible.

closed 2024-12-01 by github-actions[bot]