Xinyuan-LilyGO/LilyGo-Modem-Series #162: HTTPS Post Issue With A7670G . Error 715

JMartinezEco, 2024-10-16:
#### Description:
When trying to perform an HTTP POST request using the SIMCOM A7670G module, I encounter the error `HTTP post failed ! error code = 715`. The module successfully connects to the network and retrieves an IP address, but fails when sending the HTTP POST request using the `AT+HTTPACTION=1` command.

[image]

#### Actual Behavior:
I'm using the base example HttpsBuiltInPost.ino. The module connects to the network and obtains an IP, but the request fails with error `715` when sending a POST. 

```cpp

#include <Arduino.h>
#include "config.h"
#include "modules.h"
#include <TinyGsmClient.h>

TinyGsm modem(SerialAT);

// Crear instancias de los módulos
SIM A7670G(&modem);

...

  const char *server_url = "https://xxxxxx.xxx/insertXXXXxxxxXXXxxx";

  // Initialize HTTPS
  modem.https_begin();

  // Set GET URT
  if (!modem.https_set_url(server_url))
  {
    Serial.println("Failed to set the URL. Please check the validity of the URL!");
    return;
  }

  //
  modem.https_set_content_type("application/json");

  String post_body = "This is post example!";

  int httpCode = modem.https_post(post_body);

  if (httpCode != 200)
  {
    Serial.print("HTTP post failed ! error code = ");
    Serial.println(httpCode);
    return;
  }

  // Get HTTPS header information
  String header = modem.https_header();
  Serial.print("HTTP Header : ");
  Serial.println(header);

  // Get HTTPS response
  String body = modem.https_body();
  Serial.print("HTTP body : ");
  Serial.println(body);

...

```

As in https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117 It is said that the AT commands for this type of requests are not suported by the A767X modules, is there any firmware update available for the module to make this work?

kgmuzungu, 2024-10-16:
You can get the firmware version with the command ATI and AT+SIMCOMATI commands. (`mode.sendAT("I")` and `modem.sendAT("+SIMCOMATI")`)

maybe try to connect to https://vsh.pp.ua/TinyGSM/logo.txt first (this is the test page from the TinyGSM guy). AWS or Google could services might be trickier.

lewisxhe, 2024-10-17:
Please update the latest baseband version firmware and test again, this problem should be fixed. 
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

tomfav, 2024-10-20:
hello,I just upgraded my simcom A7670E module:
Manufacturer: INCORPORATED
Model: A7670E-PHASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
MEID:
+GCAP: +CGSM,+FCLASS,+DS

but it still gives me when I run post requests to server like:  
https://posttestserver.dev/p/1z44p3vffg1cya27/post
it returns me 715

closed 2024-12-04 by github-actions[bot]