Xinyuan-LilyGO/LilyGo-Modem-Series #163: GPS acceleration error

Jvf01, 2024-10-16:
Good morning,
like the issue reported with #152 I also have the same error.
As indicated in issue #152 I dumped the AT commands.

Log file attached.

[gpslog.docx](https://github.com/user-attachments/files/17399728/gpslog.docx)

lewisxhe, 2024-10-17:
The log you provided has obtained the longitude and latitude.

Also, I found an extra command in your log, AT+CGNSSIPR=74880

Please remove this command and test again. Do not modify the example I provided, just change the correct APN.

Jvf01, 2024-10-17:
Yes, it's true I can obtain my coordinates.
I didn't make any changes to the example outline.   
I will repeat the test without the mentioned command.
Thanh you

Jvf01, 2024-10-17:
Good morning,
I did a new test as per your recommendation.
The gps acceleration failure result remains.
I didn't remove the AT+CGNSSIPR=74880 command because I think it matches "modem.setGPSBaud(74880)".
I changed it to the original 115200 and now in the log the only change is the change from 74880 to 115200.

lewisxhe, 2024-10-18:
Which modem do you use? You can try to update the latest firmware to test
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

Jvf01, 2024-10-18:
The information returned by the AT+SIMCOMATI command was:
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
IMEI: xxxxxxxxxxxxxxx
__________________________________
The modem's QRCode has the information:
P/N:S2-10DAX-Z32C4;SN:MP062422800E6CF;IMEI:862771075501943;BTMAC:5C46B0CB289E;SW:A110B01A7670M7_F
__________________________________
The firmware I downloaded is called:
A011B07A7670M7_F_A7670E-FASE_CD_V101230711

Is this the correct firmware version to upload?

lewisxhe, 2024-10-21:
Please try this one [A011B07V01A7670M7_F_A7670E-FASE_CD_V101230711
A011B07A7670M7_F_A7670E-FASE_CD_V101230711](https://drive.google.com/file/d/1-5A9w4MCXNz6F5ODhynne0rC9IUOeOdH/view)

lewisxhe, 2024-10-21:
This means that the touch coordinates are opposite to the actual display coordinates. You need to convert the coordinates manually to the correct coordinates.

Jvf01, 2024-10-21:
Ok  i will try

Jvf01, 2024-10-23:
Good morning,
I updated the firmware that you suggested and the error initially reported has been resolved.
However, two new situations emerged:
1 - An error that did not exist "Set network apn error !"
2 - When I uploaded the sketch I didn't need to do anything..... now I have to press the "boot" button to be able to upload it.
New firmware and new error information in
Log attached.

[errorlog.docx](https://github.com/user-attachments/files/17492153/errorlog.docx)

Jvf01, 2024-10-23:
One more thing.... you said : "This means that the touch coordinates are opposite to the actual display coordinates. You need to convert the coordinates manually to the correct coordinates."

In my tests the coordinates are exatly in the same way than before.

lewisxhe, 2024-10-24:
> One more thing.... you said : "This means that the touch coordinates are opposite to the actual display coordinates. You need to convert the coordinates manually to the correct coordinates."
> 
> In my tests the coordinates are exatly in the same way than before.

This is the wrong reply, not the one for this issue, sorry

lewisxhe, 2024-10-24:
Please turn on [DUMP_AT_COMMANDS](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/1f49a65db953a10d3bfc658852c0c18a763ce6cd/examples/GPS_Acceleration/GPS_Acceleration.ino#L23). I need the original information of the interaction to judge.

Jvf01, 2024-10-24:
Ok, here is the logfile with the "apn error"
[log.docx](https://github.com/user-attachments/files/17508539/log.docx)

lewisxhe, 2024-10-25:
Please send `at+simcomati` to check if the version number in the reply is the same as mine.
```
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
IMEI: xxxxxxxxxxxxxx
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:
```

Jvf01, 2024-10-25:
Yes, it's  the same:
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
IMEI: xxxxxxxxxxxxxxxxx
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

lewisxhe, 2024-10-25:
That's strange. I tested that the apn setting can return OK correctly. Please re-flash the firmware and test it again.

Jvf01, 2024-10-25:
The problem remains after reflash twice.

[log01.docx](https://github.com/user-attachments/files/17526903/log01.docx)

Jvf01, 2024-10-31:
Good morning,
the reported error still persists despite having reflashed it twice.
Do you have any suggestions to try to resolve this problem?
Can I use the firmware, A011B07A7670M7_F_A7670E-FASE_CD_V101230711, which is published on Github?

https://github.com/user-attachments/files/17526903/log01.docx

lewisxhe, 2024-11-01:
I suspect the firmware is incorrect. Please take a photo of the sticker on the modem and send it to me so I can confirm the version. Please mosaic the IMEI number.

Jvf01, 2024-11-02:
According to your request, I will send you a photo of the modem

Jvf01, 2024-11-13:
About two weeks ago I sent the photo and the IMEI of my card as you requested.
The objective was to check if the firmware version that was installed was the correct one.
Before proceeding with other tests and checks I would like to make sure I have the correct firmware version.
Due to the lack of an answer to this question, I will close this issue without having a solution to the reported problem.

lewisxhe, 2024-11-13:
@Jvf01 I didn't receive your email, could you please send it again?

Jvf01, 2024-11-13:
---------- Forwarded message ---------

#163)

closed 2024-11-13 by Jvf01