Xinyuan-LilyGO/LilyGo-Modem-Series #168: CMUX not working on LilyGo T-SIM7670G S3 / SIMcom A7670G

morgana2313, 2024-10-25:
The CMUX mode on the LilyGo T-SIM7670G S3 (SIMcom A7670G) doesn't seem to work.

I have some questions:

1. Can you confirm that this is a known problem?
2. Can this be solved with a modem firmware update?
3. The [2374_SIM767XM5A_M Firmware Release Note](https://en.simcom.com/service-2128.html) mentions a document called _"SIM767XX Series_CMUX_USER_GUIDE. docx"_, but I cannot find it anywhere on the internet or the SIMCOM website. Can you make this document available?
4. Is this related to the SIM7670G using the Qualcomm QCX216 ?

### Details: 

* Manufacturer: SIMCOM INCORPORATED
* Model: SIM7670G-MNGV
* Revision: V1.9.01

Using [esp_modem modem_console](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console) show that the A7670G doesn't respond to the CMUX DLCI0 SAMB frame `f9 03 3f 01 1c f9`.

```
esp> cmd ATE0
ATE0
OK

esp> set_mode CMUX
I (66156) modem_console: Switching to CMUX name...
I (66156) TX: 0x3fca6c2c   41 54 2b 43 4d 55 58 3d  30 0d                    |AT+CMUX=0.|
I (66166) RX: 0x3fca5520   0d 0a 4f 4b 0d 0a                                 |..OK..|
I (66666) TX: 0x3fca6bc0   f9 03 3f 01 1c f9                                 |..?...|
E (67686) modem_console: Failed to set the desired mode
Command returned non-zero error code: 0x1 (ERROR)
esp> 
```

Other boards like the [LilyGo T-A7670E/G/SA](https://www.lilygo.cc/products/t-sim-a7670) does respond to the CMUX DLCI0 SAMB frame (see below). My conclusion is that CMUX mode in the SIM7670G chip is broken. 

Under "News" heading in this repo thare are some cryptic remarks:

* A7672G is exactly the same as A7670G. A7672G is just another name.
* SIM7672G and SIM7670G are exactly the same, except for the name change , The readme file indicates these two models with SIM767X
* SIM7670G uses the Qualcomm platform, A7670x uses the Asrmicro platform.

```
<wait for the modem to boot = the red LED next to the modem chip to start blinking>

esp> cmd ATE0
I (35934) modem_console: Sending command ATE0 with timeout 1000
I (35934) TX: 0x3ffbe690   41 54 45 30 0d                                    |ATE0.|
I (35954) RX: 0x3ffbca28   41 54 45 30 0d 0d 0a 4f  4b 0d 0a                 |ATE0...OK..|
I (35964) modem_console: ATE0
OK

esp> set_mode CMUX
I (49364) modem_console: Switching to CMUX name...
I (49364) TX: 0x3ffbe59c   41 54 2b 43 4d 55 58 3d  30 0d                    |AT+CMUX=0.|
I (49374) RX: 0x3ffbca28   0d 0a 4f 4b 0d 0a                                 |..OK..|
I (49884) TX: 0x3ffbe530   f9 03 3f 01 1c f9                                 |..?...|
I (49884) RX: 0x3ffbca28   f9 03 73 01 d7 f9                                 |..s...|
I (49904) TX: 0x3ffbe530   f9 07 3f 01 de f9                                 |..?...|
I (49904) RX: 0x3ffbca28   f9 07 73 01 15 f9                                 |..s...|
I (49924) TX: 0x3ffbe530   f9 0b 3f 01 59 f9                                 |..?.Y.|
I (49924) RX: 0x3ffbca28   f9 0b 73 01 92 f9                                 |..s...|
I (49944) TX: 0x3ffbe334   f9 05 ef 0b                                       |....|
I (49944) TX: 0x3ffbe55c   41 54 45 30 0d                                    |ATE0.|
I (49954) TX: 0x3ffbe338   bb f9                                             |..|
I (49964) RX: 0x3ffbca28   f9 05 ef 0b 41 54 45 30  0d bb f9 f9 05 ef 0d 0d  |....ATE0........|
I (49964) RX: 0x3ffbca38   0a 4f 4b 0d 0a 5f f9                              |.OK.._.|
I (49984) TX: 0x3ffbe2a4   f9 05 ef 3b                                       |...;|
I (49984) TX: 0x3ffbfed4   41 54 2b 43 47 44 43 4f  4e 54 3d 31 2c 22 49 50  |AT+CGDCONT=1,"IP|
I (49994) TX: 0x3ffbfee4   22 2c 22 69 6e 74 65 72  6e 65 74 22 0d           |","internet".|
I (50004) TX: 0x3ffbe2a8   9f f9                                             |..|
I (50014) RX: 0x3ffbca28   f9 05 ef 0d 0d 0a 4f 4b  0d 0a 5f f9              |......OK.._.|
[... 148 lines cut ...]
I (61904) RX: 0x3ffbca58   4f 50 51 52 53 54 55 56  57 58 59 5a 5b 5c 5d 5e  |OPQRSTUVWXYZ[\]^|
I (61904) RX: 0x3ffbca68   5f 60 61 62 63 64 65 66  67 68 69 6a 6b 6c 6d 6e  |_`abcdefghijklmn|
I (61914) RX: 0x3ffbca78   6f 70 71 72 73 74 75 76  77 78 79 7a 7b 7c 7d 5d  |opqrstuvwxyz{|}]|
I (61934) RX: 0x3ffbca88   7d 5e 7f 80 63 9b 7e 22  f9                       |}^..c.~".|
I (61934) modem_console_ping: 64 bytes from 8.8.8.8 icmp_seq=5 ttl=58 time=95 ms

I (62764) modem_console_ping: 
--- 8.8.8.8 ping statistics ---
I (62764) modem_console_ping: 5 packets transmitted, 5 received, 0 packet loss, time 586ms

esp> cmd AT

I (415014) modem_console: Sending command AT with timeout 1000
I (415014) TX: 0x3ffbe524   f9 09 ef 07                                       |....|
I (415024) TX: 0x3ffbe690   41 54 0d                                          |AT.|
I (415034) TX: 0x3ffbe528   35 f9                                             |5.|
I (415044) RX: 0x3ffbca28   f9 09 ef 07 41 54 0d 35  f9 f9 09 ef 0d 0d 0a 4f  |....AT.5.......O|
I (415054) RX: 0x3ffbca38   4b 0d 0a d8 f9                                    |K....|
I (415064) modem_console: 
OK

esp> 
```

This problem was discussed in https://github.com/espressif/esp-protocols/issues/669.

lewisxhe, 2024-10-28:
Related software and manuals have been added. If you have any other questions, please continue to contact me.

morgana2313, 2024-10-28:
> Related software and manuals have been added. If you have any other questions, please continue to contact me.

Thank you very much, I will study these. I do have other questions:

1. Can you confirm that this is a known problem or should CMUX work on the SIM7670G?
2. Can this be solved with a modem firmware update?
3. Is this related to the SIM7670G using the Qualcomm QCX216 ?

The README in this repo in 7.3 contains links to the SIM767XG-Datasheet that are broken because the directory does not exist:

* [SIM7672 Hardware Design manual](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/datasheet/SIM7672G/SIM7672X_Series_Hardware_Design_V1.02.pdf)
* [SIM7672 AT Command](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/datasheet/SIM7672G/SIM767XX%20Series_AT_Command_Manual_V1.01.pdf)

morgana2313, 2024-10-31:
With the provided windows CMUX software doesn't work with the default `SIM767XM5A_B01V03_231207` firmware. To 'AT+CMUX=0' it just responds 'OK' but does absolutely nothing.

However after an [upgrade](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX?tab=readme-ov-file#6%EF%B8%8F%E2%83%A3-faq) to [SIM767XM5_B03V02_240914](https://drive.google.com/file/d/1bBrze2eDtrjEuJ_2yiufSo87eIk_mYkE/view) CMUX works with windows. 

I have two questions:

1. Does LilyGo provide modem firmware downloads?
2. http://www.community.lilygo.cc has disappeared. Is the statement below from [this archived article](https://web.archive.org/web/20240719085847/https://www.community.lilygo.cc/topic/974/t-sim7670g-s3-firmware-for-sim7670g) correct? If so, how can I order `SIM7672G-MNGV`? My recent purchase were `SIM7670G-MNGV`.

> Please note that the SIM7670G-MNGV are not supported in Europe and will not enjoy the certifications and for FW updates you need to contact the source from where the board was purchased.
> The correct version for outside China is the SIM7672G-MNGV. Basically the same module as the SIM7670G-MNGV.

lewisxhe, 2024-11-01:
CE certification and the latest B04 firmware have been added

morgana2313, 2024-11-01:
With modem firmware `SIM767XM5_B03V02_240914` or `SIM767XM5_B04V01_241010 CMUX` works with the SIM7670G. Thank you for your support!

lewisxhe, 2024-11-02:
```
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B04SIM767XM5A_M
SIM767XM5_B04V01_241010

OK
Enabling GPS/GNSS/GLONASS
AT+CGDRT=4,1
AT+CGDRT=4,1

OK
AT+CGSETV=4,1
AT+CGSETV=4,1

OK
AT+CGNSSPWR=1
AT+CGNSSPWR=1

OK

GPS Enabled
AT+CGNSSIPR=115200
AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO
AT+CGNSSINFO
```

morgana2313, 2024-11-02:
`+CGNSSINFO: 3,11,07,10,14,52.145793,N,4.497525,E,021124,102142.000,-2.3,0.04,239.76,1.31,1.02,0.82,14`
parsed by `num=sscanf(gps_txt, "+CGNSSINFO: %d,%d,%d,%d,%f,%c,%f,%c,%02d%02d%02d,%02d%02d%f,%f,%f,%f,%f,%f,%f",`

num=6 and not 16. why?

morgana2313, 2024-11-02:
Thank you, setting the i/o direction helped!