Xinyuan-LilyGO/LilyGo-Modem-Series #169: gps and gsm not working

toneshoo, 2024-10-27:
when i first pluged in the device the power led (blue) was on and there was 2 other led (red) i belive for gsm and gps 
once i uploaded the firmware from github and started working on it using platform , io the problem started 
the gps and the gsm dont even blink anymore and can be seen 
i tried to change the wire reset etc nothing worked 
also there is a problem with the baudrate i keep getting random raw data whenever i upload anycode 
it compiles well but then the terminal shows !!!!#($#J$ somthing like this
i would love any help i can get on these issues

lewisxhe, 2024-10-28:
Show this? Can you give me a full screenshot? I've never seen this error before, and I don't know which software this is from
[image]

toneshoo, 2024-10-28:
[image]
i tried to change the baudrate but samething 
also the 2 led are not blinking

lewisxhe, 2024-10-28:
Monitor baud rate is 9600, change sketch baud rate to 9600 or set monitor to 115200
[image]

toneshoo, 2024-10-28:
[image]
alright one problem out of the way 
but still the gps and gsm led are not on and i get this

lewisxhe, 2024-10-28:
You need SIM-HAT hardware support to run. The examples above all need SIM-HAT hardware support. Please run the minimum ATDebug to test and read the README first.

toneshoo, 2024-10-28:
this is kind of my first time here 
could you explain more ?

lewisxhe, 2024-10-28:
All examples with SIMhat in their names require the support of this hardware to run https://www.lilygo.cc/products/lilygo%C2%AE-t-simhat-can-rs485-relay-5v

README explains which examples can be used and which cannot

toneshoo, 2024-10-28:
i saw many videos on youtube that connected a wifi and gps to the same excat model i use which is a7670e esp32 wroover e 
if i cant use these libaraies how should i proced?

lewisxhe, 2024-10-28:
What do you mean? I don't understand what you mean?

toneshoo, 2024-10-28:
before i decided to buy this model which is A7670E  
i checked some youtube videos to make sure it can work without any extra hardware
but if the examples above require some extra hardware 
can you help me find examples for the board itself alone ?

lewisxhe, 2024-10-28:
[image]

toneshoo, 2024-10-28:
hmmmmmm wait so in order to be able to use gsm and gps i need to buy the extra hardware ??

lewisxhe, 2024-10-28:
Well, to use GSM and GPS you only need to buy A7670. SIM-HAT is a separate additional hardware, you can exclude it completely, I just want to tell you that SIMHAT*** example requires additional hardware support, understand? I think I made it clear.

toneshoo, 2024-10-28:
yeah i get it now
what tuturial do i follow to start the gps and gsm 
right now there leds are off

lewisxhe, 2024-10-28:
ATDebug

toneshoo, 2024-10-28:
done that already 
every time i check the serial moniter when uploading any code i always get Failed to find INA219 1 chip in

lewisxhe, 2024-10-28:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/a4aef6e8a0c768eef6ada97d62b5747b5a840c08/platformio.ini#L44

closed 2024-12-12 by github-actions[bot]