Xinyuan-LilyGO/LilyGo-Modem-Series #171: HTTPS requests not working, error 706, SIMCOM 7670G

Kylektt, 2024-11-05:
Hi, I am currently trying to implement the OTA update on the device esp32-SIMCOM7670G-s3. I am having a problem with https call which I has a returned error code with 706. In the screenshot I also attached the version code by adding AT+SIMCOMATI. The other code is the same with the ota update example code. Is anyone have any idea?
[image]

lewisxhe, 2024-11-05:
Update the SIM7670G internal firmware to the latest version and try again

chemmex, 2024-11-05:
I have the same problem even with plain http.

My board is upgraded to 2374B04SIM767XM5A_M_SIM7670G-MNGV

Nothing changed.

Kylektt, 2024-11-05:
> I have the same problem even with plain http.
> 
> My board is upgraded to 2374B04SIM767XM5A_M_SIM7670G-MNGV
> 
> Nothing changed.

I'm able to do http call through ArduinoHttpClient library

lewisxhe, 2024-11-06:
I tested it and it's the same problem. I will communicate with SIMCOM and I will update here when the problem is solved.

Kylektt, 2024-11-06:
> I tested it and it's the same problem. I will communicate with SIMCOM and I will update here when the problem is solved.

Thanks for your reply. You're legend.

I'm also investigating from my site. I have found this example in **SIM7672X & SIM7652X Series_HTTP(S)_Application Note_V1.00.**

[image]

Seems like it requires setting for ssl before http init.
[image]

In this case, I have tried to send AT command myself and test it:

[image]

[image]

And I received a new error code called 710, which is defined as **710 start a new ssl session failed** in the manual
Here are the logs:
[image]
[image]

I have used google.com for the testing.

chemmex, 2024-11-06:
'
I'm able to do http call through ArduinoHttpClient library
'
I mean I cannot get http connection to my OTA server with the latest firmware
Both http and https requests return the same 706 error

Kylektt, 2024-11-06:
> > I'm able to do http call through ArduinoHttpClient library
> 
> I mean I cannot get http connection to my OTA server with the latest firmware Both http and https requests return the same 706 error

What library you are using for http request? tinygsm?

chemmex, 2024-11-06:
> What library you are using for http request? tinygsm?

No, I am testing with AT commands in the terminal. The sequence is the same as in the TinyGSM

Kylektt, 2024-11-07:
> > What library you are using for http request? tinygsm?
> 
> No, I am testing with AT commands in the terminal. The sequence is the same as in the TinyGSM

understood, let see how Lewisx's progress with SIMCOM. Just the remind that I can raise http request with ArduinoHttpClient library, if you are urgent to communicate with your ota server, you could have a try of the ArduinoHttpClient library, pretty straightforward.

chemmex, 2024-11-07:
I had similar problem with A7608, it could not establish HTTPS connection with most of sites. Simcom has fixed that by lewisxhe's request, so we may expect the same with SIM7670.
As to the ArduinoHttpClient, I have OTA code working well with A7608 using native http functions, so I need to do the same with SIM7670 as both modules basically share the same AT command set.

lewisxhe, 2024-11-11:
@Kylektt 
The problem was found. The available user storage space of SIM7670G is only 288Kb. The update firmware on my website has reached 352Kb. I compiled a small version of ATDebug as the update firmware and now it works. For files larger than 288Kb, it is not possible to download directly.
[image]

lewisxhe, 2024-11-11:
@chemmex  A7608 What problem did you encounter? Please open a new issue

chemmex, 2024-11-11:
Wait, does that mean that SIM7670 cannot download files larger than 288Kb using native http client?

lewisxhe, 2024-11-11:
> Wait, does that mean that SIM7670 cannot download files larger than 288Kb using native http client?

Yes

chemmex, 2024-11-11:
So if my complied sketch exceeds 288kb, the only chance for OTA would be using TinyGSM via tcp socket instead?

lewisxhe, 2024-11-11:
Yes, or you need to manually split the upgrade firmware into multiple files less than 288KB, and then merge them through esp32 and then upgrade

chemmex, 2024-11-11:
I see, thanks for the clarification 
I will try your trick with smaller image, and if it fails, open a new issue. Because I was receiving 706 errors when opening other sites, too

Kylektt, 2024-11-12:
@lewisxhe Thanks for your reply, it is very helpful.

I actually haven't had the chance to go into the step of upgrading the file, but I completely understood what you mean in the reply.

After seeing your reply, I go try the HTTPS built-in Example, and have the same issue with 
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117#issuecomment-2287998078
which I got 715 error with https://vsh.pp.ua/TinyGSM/logo.txt and https://ipapi.co/timezone.

I haven't upgraded the firmware as I met some trouble with detecting COM Port by the flash tool as raised in another issue.
I will keep trying it.

However, in the meanwhile, could you please help me to test that whether the latest firmware of SIM7670G is able to make https call with 
https://vsh.pp.ua/TinyGSM/logo.txt and https://ipapi.co/timezone?
As for the file link you provided in OTA Upgrade example is too large for the firmware as you mentioned, and I don't have the smaller file link that you used in the testing.

[image]
[image]

Notice that I have sent enableSNI after the HTTPINIT.

Thank you very much @lewisxhe

lewisxhe, 2024-11-13:
https://ipapi.co/timezone is blocked by the firewall in my country (it cannot be accessed using a computer), so I cannot access it, the same is 715, but https://vsh.pp.ua/TinyGSM/logo.txt can be accessed normally. Here are my versions and logs
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x42011f02
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x27e0
entry 0x403c98bc
Start Sketch
Start modem...

SIM card online
Wait for the modem to register with the network.Online registration successful

Registration Status:1
Inquiring UE system information:LTE,Online,460-11,0x775C,117004595,399,EUTRAN-BAND3,1650,5,5,-9,-89,-80,5
Network IP:AT+IPADDR+IPADDR: 10.6.251.113
Request URL : https://httpbin.org/get
HTTP Header : HTTP/1.1 200 OK
Date: Wed, 13 Nov 2024 02:12:29 GMT
Content-Type: application/json
Content-Length: 254
Connection: keep-alive
Server: gunicorn/19.9.0
Access-Control-Allow-Origin: *
Access-Control-Allow-Credentials: true
HTTP body : {
  "args": {}, 
  "headers": {
    "Accept": "*/*", 
    "Cache-Control": "no-cache", 
    "Host": "httpbin.org", 
    "X-Amzn-Trace-Id": "Root=1-67340b0d-526171ce4dbbd1fd4f42c737"
  }, 
  "origin": "113.84.2.125", 
  "url": "https://httpbin.org/get"
}

-------------------------------------
Request URL : https://vsh.pp.ua/TinyGSM/logo.txt
HTTP Header : HTTP/1.1 200 OK
Server: nginx/1.22.1
Date: Wed, 13 Nov 2024 02:04:58 GMT
Content-Type: text/plain; charset=UTF-8
Content-Length: 121
Connection: keep-alive
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"
HTTP body : 
  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

-------------------------------------
Request URL : https://ipapi.co/timezone
HTTP get failed ! error code = -1
Request URL : https://ipapi.co/timezone
HTTP get failed ! error code = 0
Request URL : https://ipapi.co/timezone
HTTP get failed ! error code = 0
-------------------------------------
Request URL : http://ip-api.com/json/23.158.104.183
HTTP get failed ! error code = 0
Request URL : http://ip-api.com/json/23.158.104.183
HTTP get failed ! error code = 0
Request URL : http://ip-api.com/json/23.158.104.183
HTTP get failed ! error code = 0
-------------------------------------
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B04SIM767XM5A_M
SIM767XM5_B04V01_241010
IMEI: xxxxxxxxxxxxxxxxxx

OK

```

lewisxhe, 2024-12-06:
I don't have this example. This requires the combination of firmware server and application layer. I can't provide this example.

closed 2025-01-20 by github-actions[bot]

Zontex, 2025-02-25:
> I don't have this example. This requires the combination of firmware server and application layer. I can't provide this example.

Opening this again because I tried every possible way.. split.. no split.. download 100kb or even 1kb file, all interrupted at 85% download. can u provide the bin file you tried to OTA with? the small sample? i will give it a try, been spending months on this issue, and i upgraded the module to the latest firmware issue.