Xinyuan-LilyGO/LilyGo-Modem-Series #172: A7670 suddenly switches off

GiovanniDeRosa1, 2024-11-05:
Hi, 
I am finding some difficulties with the operation of the A7670E module.  I list the operations I perform;
I set I/O 25 to a high level to turn on the voltage regulator on the T-PCIE board, I wait a few seconds. 
I set I/O 4 to a high level for a minimum time of 500mS then i set it high, wait 12 seconds, then I initialise the uart and after 2 seconds I send an “AT” command to which I receive a reply.
If I try to power the T-PCIE board from a USB 3.0 to the “AT” command and the subsequent “ATDXXXXXXXXX” I get a response and the call starts, if I try with a different usb port it works sporadically. The improssion is that the board with A7670E shuts down.
I checked the voltages and they are stable at 4.2V. the PowerKey is high as is the Reset. The “STATUS” and “NETLIGHT” LEDs go off and the module no longer responds to commands via UART. What could be the problem ? 
I use esp- idf and c as programming language.
Thank you for your support.

lewisxhe, 2024-11-05:
This situation is usually caused by unstable power supply current. Plug in the battery as a backup power source and plug in the USB-C at the same time.

GiovanniDeRosa1, 2024-11-05:
This was our main doubt, we tried to externally power vbus pins with a 12V and 2A max. power supply, regulated to 5V and occasionally the same problem occurred

lewisxhe, 2024-11-05:
12V? Are you using a PCIE board? The maximum VBUS input is 5V. Try connecting a battery. A momentary power loss on VBUS will cause the PMU to shut down for protection.

GiovanniDeRosa1, 2024-11-05:
It is connected to the mains with 12V, but we have a regulator before we enter the T-PCIE, so we enter at 5V. can the back-up battery still be useful ? entering directly on the bipolar connector of the T-PCIE ?

lewisxhe, 2024-11-05:
Connect to the battery holder of the PCIE motherboard.

GiovanniDeRosa1, 2024-11-05:
What voltage should the battery have?

lewisxhe, 2024-11-06:
4.2V standard lithium-ion battery

closed 2024-12-22 by github-actions[bot]