Xinyuan-LilyGO/LilyGo-Modem-Series #177: GPS on A7670 get satellites always is 0

EnziinSystem, 2024-11-15:
Dear,

I run the example GPS_BuiltIn

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/GPS_BuiltIn

With the config file:

```
#define LILYGO_T_CALL_A7670_V1_0
```

However, I always get satellites are 0

```
OK
FixMode:3
Latitude:****	
Longitude:****
...
Visible Satellites:0	
Used Satellites:0
...

Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO
```
The example still works and can return the correct longitude and latitude.

I need the visible and usable satellite value.

What is the problem?

Thanks.

lewisxhe, 2024-11-16:
The number of visible GPS satellites has been fixed, but the number of visible satellites is not available. If you need more GPS messages you need to use the GPS_NMEA_Parse function.

EnziinSystem, 2024-11-16:
Hi,

Thanks for your reply.

I see the Visible Satellites number.

```
OK
FixMode:3
Latitude:**** Longitude:*****
........
Visible Satellites:9	Used Satellites:0
......
Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO
```

And the Used Satellites always is 0.

Is this due to hardware limitations or something else?

lewisxhe, 2024-11-16:
I think I have said above that the number of satellites used is not available. This is because the GNSSINFO command does not return this value.

closed 2024-11-16 by EnziinSystem