Xinyuan-LilyGO/LilyGo-Modem-Series #178: Connect to the network

GiovanniDeRosa1, 2024-11-15:
Hi, i am using an a7670E module and want to connect to the network and use mqtt to post/submit topics. however, i do not understand if i am actually connected to the network. The status led is blinking very fast 

`Risposta ricevuta: AT+CFUN=1
OK
Risposta ricevuta: AT+CPIN?
+CPIN: READY
OK
Risposta ricevuta: AT+CSQ
+CSQ: 14,0
OK
Risposta ricevuta: AT+CREG=1
OK
Risposta ricevuta: AT+COPS?
+COPS: 0,2,"22201",3
OK
Risposta ricevuta: AT+CREG?
+CREG: 1,1
OK
Risposta ricevuta: AT+CGATT?
+CGATT: 1
OK
Risposta ricevuta: AT+CPSI?
+CPSI: GSM,Online,222-01,0x46C2,43674,1021 PGSM 900,25,0,15-15
OK
Risposta ricevuta: AT+CGDCONT= 1,"IP","apn.fastweb.it","0.0.0.0",0,0
OK
Risposta ricevuta: AT+CGACT=1,1
+CME ERROR: requested service option not subscribed (#33)
+CGEV: ME PDN DEACT 1
Risposta ricevuta: AT+CGACT?
+CGACT: 3,0
OK
Risposta ricevuta: AT+CGPADDR=1
OK
Risposta ricevuta: AT+NETOPEN
OK
+NETOPEN: 0
Risposta ricevuta: AT+NETSTATE
ERROR
Risposta ricevuta: AT+CPING="www.google.com"
ERROR
`
best regards for the support

lewisxhe, 2024-11-16:
```
+CPSI: GSM,Online,222-01,0x46C2,43674,1021 PGSM 900,25,0,15-15
```

The above indicates that it has been properly registered in the GSM network.

```
AT+NETOPEN
OK
+NETOPEN: 0
```

This indicates that the network has been opened normally

AT+CPING="www.google.com" is incorrectly used. You must specify whether the address type is IPV4. The correct one is

```
AT+CPING="www.google.com" ,1
```

GiovanniDeRosa1, 2024-11-18:
The connection to the network seems ok, now I want to try connecting to the mqtt broker. it is a secure connection but does not require checking the certificate (at least on the esp-wifi side), the connection attempt returns ok, but with an error.
`
Rx<<<: AT+CFUN=1
OK
Rx<<<: AT+CPIN?
+CPIN: READY
OK
Rx<<<: AT+CSQ
+CSQ: 12,0
OK
Rx<<<: AT+CREG=1
OK
Rx<<<: AT+COPS?
+COPS: 0,2,"22201",3
OK
Rx<<<: AT+CREG?
+CREG: 1,1
OK
Rx<<<: AT+CGATT?
+CGATT: 1
OK
Rx<<<: AT+CPSI?
+CPSI: GSM,Online,222-01,0x46C2,438,9 PGSM 900,21,0,11-11
OK
Rx<<<: AT+CGDCONT= 1,"IPV4V6","apn.fastweb.it","0.0.0.0",0,0
OK
Rx<<<: AT+CGACT=1,1
OK
+CGEV: ME PDN ACT 1,0
AT+CGACT?
+CGACT: 1,1
+CGACT: 3,0
OK
Rx<<<: AT+CGPADDR=1
+CGPADDR: 1,10.132.200.26
OK
Rx<<<: AT+NETOPEN=1
OK
+NETOPEN: 0
Rx<<<: AT+CIPSRIP=1
OK
Rx<<<: AT+CMQTTSTART
OK
+CMQTTSTART: 0
Rx<<<: AT+CMQTTSSLCFG=0,1
OK
Rx<<<: AT+CMQTTCONNECT=1,"tcp://xxxxyyycloud.ddns.net:8883",120,0,"yyyyyyyyy","xxxxxxxxxxxxxxxxxxxxx"
+CMQTTCONNECT: 1,20

ERROR`
how i can fix it ? 
thanks for the support

GiovanniDeRosa1, 2024-11-18:
`
Rx<<<: AT+CFUN=1
OK
Rx<<<: AT+CPIN?
+CPIN: READY
OK
Rx<<<: AT+CSQ
+CSQ: 13,0
OK
Rx<<<: AT+CREG=1
OK
Rx<<<: AT+COPS?
+COPS: 0,2,"22201",3
OK
Rx<<<: AT+CREG?
+CREG: 1,1
OK
Rx<<<: AT+CGATT?
+CGATT: 1
OK
Rx<<<: AT+CPSI?
+CPSI: GSM,Online,222-01,0x46C2,43674,1021 PGSM 900,24,0,14-14
OK
Rx<<<: AT+CGDCONT= 1,"IPV4V6","apn.fastweb.it","0.0.0.0",0,0
OK
Rx<<<: AT+CGACT=1,1
OK
+CGEV: ME PDN ACT 1,0
AT+CGACT?
+CGACT: 1,1
+CGACT: 3,0
OK
Rx<<<: AT+CGPADDR=1
+CGPADDR: 1,10.81.78.158
OK
Rx<<<: AT+NETOPEN=1
OK
+NETOPEN: 0
Rx<<<: AT+CIPSRIP=1
OK
Rx<<<: AT+CMQTTSTART
OK
+CMQTTSTART: 0
Rx<<<: AT+CMQTTSSLCFG=1,1
OK
Rx<<<: AT+CSSLCFG="sslversion",2,3
OK
Rx<<<: AT+CSSLCFG="authmode",2,0
OK
Rx<<<: AT+CSSLCFG="enableSNI",2,1
OK
Rx<<<: AT+CDNSCFG="8.8.8.8","8.8.4.4"
OK
Rx<<<: AT+CMQTTACCQ=1,"10061C843E36",1
OK
Rx<<<: AT+CMQTTCONNECT=1,"tcp://xxxxx.ddns.net:8883",120,0,"xxxxx","xxxxxxxxxxxxx"
+CMQTTCONNECT: 1,32`

lewisxhe, 2024-11-20:
Port 8883 needs to be encrypted to use encrypted transmission, please check the provided example operation.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/MqttsBuiltlnEMQX/MqttsBuiltlnEMQX.ino

GiovanniDeRosa1, 2024-11-20:
Hi, I am currently using the Esp-Idf framework so without arduino and the example is not of much help to me. Is it possible to know the correct AT command sequence ?

lewisxhe, 2024-11-20:
You can open https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/01925ece37dad7d66b27c4b5c3224da6f67f5cd7/examples/MqttsBuiltlnEMQX/MqttsBuiltlnEMQX.ino#L15 to print all interaction logs

GiovanniDeRosa1, 2024-11-20:
ok, but from the example i don't see user and pass. for my broker ( mqtts) it is necessary to enable ssl ( the certificate is not necessary), but i have to report user and pass to access the command ‘ AT+CMQTTCONNECT=1, “tcp://xxxxx.ddns.net:8883”,120,0, “user”, “password”’ is correct ?

lewisxhe, 2024-11-20:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/MqttsBuiltlnAuth/MqttsBuiltlnAuth.ino
This example matches your description

GiovanniDeRosa1, 2024-11-22:
unfortunately i cannot use the example provided with arduino for my c application with esp.idf, is it possible to get support on the commands to use and the correct formatting ?

lewisxhe, 2024-11-22:
If you run the [example](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/MqttsBuiltlnAuth/MqttsBuiltlnAuth.ino) you get all the interactions, don't you want to or can't you? I can only run it again and submit it to you, hopefully you can figure it out.

```
Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT

OK

AT+CPIN?
AT+CPIN?

ERROR
AT+CPIN?

*ATREADY: 1
AT+CPIN?
+CME ERROR: SIM failure
AT+CPIN?
AT+CPIN?
+CME ERROR: SIM busy

+CPIN: READY
SIM card online
AT+CNMP=2
AT+CNMP=2

OK
AT+CNMP?

+CGEV: EPS PDN ACT 1
AT+CNMP?

+CNMP: 2

OK
Current network mode : AUTO
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?
AT+CPSI?

+CPSI: LTE,Online,460-11,0x7764,102547543,168,EUTRAN-BAND8,3740,3,0,20,68,67,14

OK
Inquiring UE system information:LTE,Online,460-11,0x7764,102547543,168,EUTRAN-BAND8,3740,3,0,20,68,67,14
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

SMS DONE
AT+IPADDR

+IPADDR: 10.7.30.107

OK
Network IP:10.7.30.107
AT+CMQTTSTART
AT+CMQTTSTART

OK

+CMQTTSTART: 0
Connecting to test.mosquitto.org
AT+CSSLCFG="authmode",0,0
AT+CSSLCFG="authmode",0,0

OK
AT+CMQTTREL=0
AT+CMQTTREL=0

+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1

OK
AT+CMQTTCFG="version",0,4
AT+CMQTTCFG="version",0,4

OK
AT+CMQTTCONNECT=0,"tcp://test.mosquitto.org:8885",60,1,"rw","readwrite"
AT+CMQTTCONNECT=0,"tcp://test.mosquitto.org:8885",60,1,"rw","readwrite"

OK

+CMQTTCONNECT: 0,0
successed.
AT+CMQTTDISC?

AT+CMQTTDISC?

+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
MQTT has connected!
AT+CMQTTSUBTOPIC=0,21,0
AT+CMQTTSUBTOPIC=0,21,0

>
GsmMqttTest/subscribe

OK
AT+CMQTTSUB=0
AT+CMQTTSUB=0

OK

+CMQTTSUB: 0AT+CMQTTTOPIC=0,19
,0
AT+CMQTTTOPIC=0,19

>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10

>RunTime:26

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60

OK

+CMQTTPUB: 0,0

```

GiovanniDeRosa1, 2024-11-22:
I used an example borker without user and pass, and actually managed to publish, 
however if i use my broker with user and pass i get ’
Rx<<: AT+CMQTTCONNECT=0, ‘tcp://mybroker.ddns.net:8883’,60,1, ‘user’, ‘pass’
+CMQTTCONNECT: 0,32’

lewisxhe, 2024-11-22:
Are you testing the same server?

GiovanniDeRosa1, 2024-11-22:
yes if I use the mqtt side esp32 configured in this way everything works correctly 

`
esp_mqtt_client_config_t mqtt5_cfg = {
        .broker.address.uri = BROKER_URL,
        .broker.address.port = BROKER_PORT,  
        .buffer.size = 1024,        
        .broker.verification.skip_cert_common_name_check = true,
        .session.protocol_ver = MQTT_PROTOCOL_V_5,
        .network.disable_auto_reconnect = true,
        .credentials.username = MQTT_USER,
        .credentials.authentication.password =MQTT_PASS,
        .session.last_will.retain = true,
        .credentials.authentication.certificate=NULL,
        .broker.verification.use_global_ca_store = false,      
        .credentials.client_id=ID_CODE,
    };
`

lewisxhe, 2024-11-22:
Before connecting `AT+CMQTTCONNECT`, send `AT+CSSLCFG="enableSNI",0,1`  to try

GiovanniDeRosa1, 2024-11-22:
still not working

`+CMQTTSTART: 0
Rx<<<: AT+CSSLCFG="sslversion",0,4
OK
Rx<<<: AT+CSSLCFG="authmode",0,0
OK
Rx<<<: AT+CSSLCFG="enableSNI",0,1
OK
Rx<<<: AT+CMQTTACCQ=0,"ID_CODE",1
OK
Rx<<<: AT+CMQTTCONNECT=0,"tcp://broker.ddns.net:8883",60,1,"user","pass"
+CMQTTCONNECT: 0,32`

lewisxhe, 2024-11-22:
1. Is the firmware version you are using the latest one? Send AT+SIMCOMATI to check
2. Can you provide me with a test account to test it?
3. If you are using the latest version and cannot provide me with a test account, then you need to provide the modem log, and I need to submit the log to SIMCOM for analysis. How to capture logs and upgrade please see here

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/a7670x_capture_log.md

GiovanniDeRosa1, 2024-11-22:
do you need broker,user,pass and code id ? 
I can send it to you privately in case (it's corporate and I don't want to keep it public)
```
Rx<<<: AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670E-LASE
Revision: A011B12A7670M7
A7670M7_B12V02_220613
QCN:
IMEI: 862771071742426
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
```

lewisxhe, 2024-11-22:
Please upgrade to the latest version of the firmware first, then try again. If it still doesn't work, send the mqtt test information to my email

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md#a7670e-lase-no-gps

GiovanniDeRosa1, 2024-11-22:
is there a firmware update guide ?

lewisxhe, 2024-11-22:
There is a video https://youtu.be/AZkm-Z7mKn8

lewisxhe, 2024-11-22:
You are using T-PCIE.？？？ 
The USB interface of T-PCIE is here. You need a USB cable to connect the solder joints.

[image]

GiovanniDeRosa1, 2024-11-22:
I have updated, with the new version 

```
x<<<: AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-LASE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI:
OK
```

however, some data were introduced that did not appear before, such as 
```
Rx<<<: 
+CGEV: NW PDN DEACT 1

+CIPEVENT: NETWORK CLOSED UNEXPECTEDLY,1

+CGEV: NW REATTACH

+CGEV: EPS PDN ACT 1

+CGEV: ME PDN ACT 8,1
```

lewisxhe, 2024-11-24:
Has the MQTT issue been resolved?

GiovanniDeRosa1, 2024-11-25:
however i have a doubt on the network side. now i am using my sim (whose apn i know). but i would like to make it so that i can use any sim. is there a way to set up the apn independently?

GiovanniDeRosa1, 2024-11-25:
I am carrying out tests by not selecting the apn 
`Rx<<: AT+CGDCONT= 1, ‘IPV4V6’,"’`
is this advisable ? i am seeing that 3/5 connects

lewisxhe, 2024-11-25:
Whether to set APN depends on the operator you use. Some operators require APN to be set before allowing network access. In China, it is not necessary because the operator issues the APN.

GiovanniDeRosa1, 2024-11-25:
ok so it would always be better to set it (not knowing the intended use)

lewisxhe, 2024-11-25:
Well, you can try it yourself

GiovanniDeRosa1, 2024-11-25:
As far as the pub is concerned, I have no problems. however, if I try to sub-register, I get an error. is the procedure wrong? 

+CMQTTPUB: 0,0
Rx<<<: AT+CMQTTSUBTOPIC=0,33,1
OK
+CMQTTSUBTOPIC: 0,0
Rx<<<: AT+CMQTTSUB=0,1,120
ERROR

lewisxhe, 2024-11-26:
If an error occurs, please read the manual first

[image]

GiovanniDeRosa1, 2024-11-26:
I also solved with the sub, thanks

GiovanniDeRosa1, 2024-11-27:
i use the same topic for convenience. the A7670E modules with gps have arrived, and i am trying to retrieve the position. i switch on the module and wait for the ‘ready’ but get empty values. the sequence seems correct to me 

```
+CGNSSPWR: READY!
Rx<<<: AT+CGNSSTST=1
OK
Rx<<<: AT+CGNSSPORTSWITCH=0,1
OK
Rx<<<: AT+CGPSCOLD
OK
Rx<<<: AT+CGPSNMEARATE=1
OK
Rx<<<: AT+CGNSSNMEA=1,0,1,1,1,0,0,0
OK
Rx<<<: AT+CGPSINFO
+CGPSINFO: ,,,,,,,,
```

lewisxhe, 2024-11-27:
Only when the positioning is fixed will there be valid data. There are many issues in the GPS issue that can be referred to.

GiovanniDeRosa1, 2024-11-27:
I spotted a problem, I have two A7670E boards
one upgraded to version 240927 and it doesn't connect (or at least I can't see) the gnss side prints, however there is one print that I can't figure out what it is 
```
Rx<<<: 
+MSTK: 11, "D0819281030113008202818305000607919373652636000B7C41000C9193736500000100046F027000006A0D000000004432310000000000000109082922808379799857020A989380000070799857F083088A2677012724420804088A2677012724710705143F01EDE2119C00078400001F4260000043000000060200D70702214308060001000003000B09082922808379799857"
```
however it works on the mqtt side.
another board i tested it stock as it came with version 240105 and the gnss works correctly, while the mqtt connection does not ( the strange print is not present).
the AT commands sent to both boards are identical, the t-pcie module was not flashed during the board swap

lewisxhe, 2024-11-28:
pcie,a7670 has a separate GPIO to control the GPS antenna switch, have you done it?
[image]

lewisxhe, 2024-11-28:
Please take a picture of the sticker on the modem

GiovanniDeRosa1, 2024-11-28:
this is the firmware I loaded on the one on the left 
[image]

lewisxhe, 2024-11-28:
I don't have a modem with this PN number now, I need some time

GiovanniDeRosa1, 2024-11-28:
> > hai bisogno di broker, utente, pass e codice id? Posso inviarteli privatamente nel caso (è aziendale e non voglio tenerlo pubblico)
> > ```
> > Rx<<<: AT+SIMCOMATI
> > Manufacturer: INCORPORATED
> > Model: A7670E-LASE
> > Revision: A011B12A7670M7
> > A7670M7_B12V02_220613
> > QCN:
> > IMEI: 8627xxxxxxxxxx
> > MEID:
> > +GCAP: +CGSM,+FCLASS,+DS
> > DeviceInfo:
> > 
> > OK
> > ```
> 
> Questa è la versione del firmware che avevi prima dell'aggiornamento?

no, this is the version of this board (without gps)  
[image]

lewisxhe, 2024-11-28:
Confirm again whether it is as shown in the figure below? The new version (A7670M7_B07V01_240927) GPS is invalid, and the old version (A7670M7_B01V01_240105) is valid?
[image]

lewisxhe, 2024-11-28:
Are you using the firmware and writing tools I provided?

GiovanniDeRosa1, 2024-11-28:
I used the tools/firmware on this page 
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

GiovanniDeRosa1, 2024-11-28:
in the meanwhile, can I get version A7670M7_B01V01_240105 so that I can upload it to the module and test whether it actually works on the gnss side ?

lewisxhe, 2024-11-28:
https://drive.google.com/file/d/10IcmnCdVcRooT8N2r52LK6sUne_WvIti/view?usp=sharing

lewisxhe, 2024-11-29:
So the GPS should work fine with the old version, right? I will test it when I get the board.

lewisxhe, 2024-11-29:
I used the same PN version to test, B01, B07 can be positioned normally

# A7670M7_B07V01_240927

```

*ATREADY: 1
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862771072422945
+GCAP: +CGSM,+FCLASS,+DS

OK
[20690] ### Modem: A7670E-FASE
Model Name:A7670E-FASE
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
IMEI: 862771072422945
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
Enabling GPS/GNSS/GLONASS
AT+CGDRT=4,1

OK
AT+CGSETV=4,0

OK
AT+CGNSSPWR=1

OK

+CPIN: READY

SMS DONE

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

PB DONE

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: 3,07,,03,00,22.6562366,N,114.0598297,E,291124,023114.00,118.4,0.679,,3.43,2.74,2.06,06

OK
FixMode:3
Latitude:22.656237	Longitude:114.059830
Speed:0.68	Altitude:118.40
Visible Satellites:7	Used Satellites:0
Accuracy:2.74
Year:2024	Month:11	Day:29
Hour:2	Minute:31	Second:14
Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO

+CGNSSINFO: 3,11,,03,00,22.6564522,N,114.0598373,E,291124,023117.00,120.1,2.647,,3.43,2.74,2.06,06

OK
GPS/GNSS Based Location String:3,11,,03,00,22.6564522,N,114.0598373,E,291124,023117.00,120.1,2.647,,3.43,2.74,2.06,06
Disabling GPS
AT+CGNSSPWR=0

OK

```
# A7670M7_B01V01_240105

```
*ATREADY: 1
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862771072422945

OK
[10675] ### Modem: A7670E-FASE
Model Name:A7670E-FASE
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
IMEI: 862771072422945

OK
Enabling GPS/GNSS/GLONASS
AT+CGDRT=4,1

OK
AT+CGSETV=4,0

OK
AT+CGNSSPWR=1

OK

+CPIN: READY

SMS DONE

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

PB DONE

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: 3,10,,03,00,22.6554260,N,114.0598373,E,291124,021942.00,183.6,0.821,,7.28,6.05,4.04,06

OK
FixMode:3
Latitude:22.655426	Longitude:114.059837
Speed:0.82	Altitude:183.60
Visible Satellites:10	Used Satellites:0
Accuracy:6.05
Year:2024	Month:11	Day:29
Hour:2	Minute:19	Second:42
Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO

+CGNSSINFO: 3,10,,03,00,22.6553040,N,114.0598526,E,291124,021944.00,170.3,0.561,,7.26,6.04,4.04,06

OK
GPS/GNSS Based Location String:3,10,,03,00,22.6553040,N,114.0598526,E,291124,021944.00,170.3,0.561,,7.26,6.04,4.04,06
Disabling GPS
AT+CGNSSPWR=0

OK

```

GiovanniDeRosa1, 2024-11-29:
> So the GPS should work fine with the old version, right? I will test it when I get the board.

yes, it immediately connects and works normally, I try reloading the new firmware and place the module in the same position

lewisxhe, 2024-11-29:
Did you re-flash the new version?

GiovanniDeRosa1, 2024-11-29:
> Did you re-flash the new version?

yes i do

lewisxhe, 2024-11-29:
Well, surprising

GiovanniDeRosa1, 2024-11-29:
and I am currently using your same commands AT
AT+CGDRT=4,1
AT+CGSETV=4,0
AT+CGNSSPWR=1
AT+CGNSSIPR=115200
AT+CGNSSINFO  (looping every x seconds if it doesn't read anything)

after 10 minutes he found the position

lewisxhe, 2024-11-29:
I was outside for about a minute and he was able to find a spot.

GiovanniDeRosa1, 2024-11-29:
these commands I am not currently using, are they necessary? 
AT+CGPSCOLD
AT+CGNSSTST=1
AT+CGNSSNMEA=1,1,1,1,1,1,1,1
AT+CGNSSPORTSWITCH=0,1

lewisxhe, 2024-11-29:
This is not necessary, it is used for debugging

closed 2025-02-03 by github-actions[bot]