Xinyuan-LilyGO/LilyGo-Modem-Series #179: Unable to establish connection to the 4g modem

JoakimRE, 2024-11-17:
I have a LilyGO A7670E - Everything seems to be working fine, except for one thing;

I've been reading up on TinyGSM documentation and verifyed that i'm referencing the correct variables.
I have also created a simple script for debugging 

# 
bool checkSimStatus() {
    modem.sendAT("+CPIN?"); 

    String response;
    if (modem.waitResponse(1000, response) > 0) { 
        Serial.println("Response: " + response); 
        if (response.indexOf("READY") != -1) {
            return true; 
        } else if (response.indexOf("SIM PIN") != -1) {
            Serial.println("SIM card is locked with a PIN.");
        } else {
            Serial.println("SIM status unknown.");
        }
    }
    return false; // SIM is not ready or there is an issue
}

#167 

#  if (checkSimStatus()) {
        Serial.println("SIM card is installed and working.");
    } else {
        Serial.println("SIM card is not installed or is not working.");
        return; // Exit if SIM is not working
    }

    // Connect to GPRS
    if (modem.gprsConnect(apn, user, pass)) {
        Serial.println("Connected to GPRS network.");
    } else {
        Serial.println("Failed to connect to GPRS network.");
    }
#

This returns "SIM card is not installed or is not working."

I also noticed the github repo included an ATdebug function, so i thought i might try that one to see if i could try some AT commands.

Compiling .pio\build\T-A7670X\src\ATdebug.ino.cpp.o
In file included from LilyGO-T-A76XX-main/examples/ATdebug/ATdebug.ino:9:
examples/ATdebug/utilities.h:18: warning: "LILYGO_T_A7670" redefined
 #define LILYGO_T_A7670
 
<command-line>: note: this is the location of the previous definition
Retrieving maximum program size .pio\build\T-A7670X\firmware.elf
Checking size .pio\build\T-A7670X\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.6% (used 21536 bytes from 327680 bytes)
Flash: [==        ]  21.6% (used 282697 bytes from 1310720 bytes)
================================================================================= [SUCCESS] Took 1.93 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
T-A7670X       SUCCESS   00:00:01.929
================================================================================= 1 succeeded in 00:00:01.929 =================================================================================
 *  Terminal will be reused by tasks, press any key to close it.

Script compiles succesfully, but i never get past the AutoBaud() function;

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start Sketch
Trying baud rate 115200
Trying baud rate 9600
Trying baud rate 57600
Trying baud rate 38400
Trying baud rate 19200
Trying baud rate 74400
Trying baud rate 74880
Trying baud rate 230400
Trying baud rate 460800
Trying baud rate 2400
Trying baud rate 4800
Trying baud rate 14400
Trying baud rate 28800
***********************************************************
 Failed to connect to the modem! Check the baud and try again.
***********************************************************

I'm starting to suspect physical issues, but i'm also open to the possibility that i might have missed something?

Any help would be appreciated

lewisxhe, 2024-11-18:
Determine if you are using T-A7670

https://lilygo.cc/products/t-sim-a7670e

Is it the same board as in the linked photo?

If it is the same board, is the red status light on the modem?

JoakimRE, 2024-11-18:
https://github.com/user-attachments/assets/cb2812f2-bf2b-4377-af26-8542a799249f

This is the led indicator status
 

Status light is lit constantly, net light is blinking

lewisxhe, 2024-11-20:
Can you measure whether there is voltage across these two terminals?
[image]

JoakimRE, 2024-11-20:
Should i measure directly across(terminal to terminal) or terminal to GND?

lewisxhe, 2024-11-20:
Terminal to GND

JoakimRE, 2024-11-20:
Alright, ive located U3, and i will measure whether there is voltage across these two terminals or not.

If the voltage is as it should be, what is the next step in the process?

lewisxhe, 2024-11-21:
If the voltage is normal, you can try to connect the two pins indicated by the line segment according to the method in the picture to test whether they can communicate normally.

[image]

JoakimRE, 2024-11-21:
Alright.
Connect them usin a pin, or how do you mean?

lewisxhe, 2024-11-22:
Connect them as shown in the picture and check if the communication is normal.

JoakimRE, 2024-11-26:
Alright.

I have checked voltage, and it works. 
So something else must be wrong? 

Why do you think it is related to the physical board?

JoakimRE, 2024-11-26:
And why is the net light blinking? What does this indicate?

JoakimRE, 2024-11-26:
Alright, so it turns out the NET light stop blinking when i discard the SIM card.
So when i boot up the ATdebug script WITH sim card, the NET led is blinking. When i boot up the ATdebug script WITHOUT sim card, the LED is on constantly.

However, when plugged into PC, neither status or net is lit.

lewisxhe, 2024-11-27:
The NET light is on, indicating that you are already in the network or searching for the network

The inability to use ATDebug to communicate with the modem has nothing to do with the SIM card. So you still can't communicate now?

JoakimRE, 2024-11-28:
Alright, so.

The ATdebug script will not give me access to the modem if the SIM is inserted for some reason. Bear in mind, this is a european sim. Here's an example:

WITH SIM INSERTED:
Start Sketch
Trying baud rate 115200
Trying baud rate 9600
Trying baud rate 57600
Trying baud rate 38400
ets Jul 29 2019 12:21:46

// I unhooked the sim-card and hit the restart button on the board here

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start Sketch
Trying baud rate 115200
Modem responded at rate:115200
***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
 DISCLAIMER: Entering AT commands without knowing what they do
 can have undesired consiquinces...
***********************************************************

The ATdebug script should not even consider the SIM-card, right.
So what's happening here?

JoakimRE, 2024-11-28:
Alright, i found the issue.

Turns out, if i unhook the SIM lock and manually press keep the SIM card in place by firnly pressing it down with my finger, the the modem successfully boots with the SIM card inserted.

The reson is probably that the SIM pins are bent a bit.. If you look at the below pictures you can see what i mean

[image]

[image]

[image]

lewisxhe, 2024-11-29:
OK. So your question is, if you insert a SIM card, you can't access the modem and use ATDebug? Can you use it if you remove the SIM card?

ATDebug does not check the existence of the SIM card and will not prevent the program from running, so your question is a bit strange.

JoakimRE, 2024-11-29:
No, my question is not strange, i provides you command line output, troubleshootong steps and results.. Clear as day.

Did you read my last post in it's entirety?

lewisxhe, 2024-11-29:
You mean you have to remove the SIM card for the modem to work properly, but you can't do it without it, right?

JoakimRE, 2024-11-29:
Correct.

I also provided a resson as to why this is the case - The SIM pins on the board are bent in different directions.
Now, why the ATdebug script would prevent me from accessing the modem when it only detects the SIM on, say, 4 or 5 out of all of the 6 PINS is beyond me, and by all accounts does not make sense, but it is what happens.

Basically i found the answer to my question.

Now my next question is;

Why are the SIM pins bent from the beginning? Did aliexpress send me a used board? 😊

lewisxhe, 2024-11-29:
So what I said above is not a problem,

- ATDebug does not work when SIM card is inserted
- ATDebug works when SIM card is not inserted

Although I think this is unlikely, I cannot get your board, so I cannot test it. Please contact the dealer for replacement.

JoakimRE, 2024-11-29:
If you think its unlikely, even with the debugging info sent you, i can send you a video demonstrating

lewisxhe, 2024-12-02:
No, I believe you, I just find it incredible. You can contact customer service and tell them that you have communicated with me here, and they will ask me.

lewisxhe, 2024-12-07:
@Sebastian-DK 800L is different from A7670, do you use ATDebug from this repository?

Sebastian-DK, 2024-12-08:
Yes i used ATDebug, and from what I can see the rx and tx pins are the same, so coms with the A7670 module should work the same.

lewisxhe, 2024-12-09:
I mean, did you use ATDebug from this repository? A7670x has the same RX, TX, but there are several more control pins to control A7670X

Sebastian-DK, 2024-12-09:
Yep, I used the ATDebug from the repository. I'm not sure I've done it right, but when I run the code it tests the different baud rates. When the first is tested(115200), the red LEDs don't light up, I'll just try with some more delay when I find time :)

lewisxhe, 2024-12-10:
The first condition for communication is that the red LED light of the modem must be on, otherwise communication will not be possible.

closed 2025-01-24 by github-actions[bot]