Xinyuan-LilyGO/LilyGo-Modem-Series #180: How to update firmware for T-Call-A7670X-V1-0?

Tsury, 2024-11-20:
Hello.

I have `T-Call-A7670X-V1-0`. I executed `AT+SIMCOMATI` and received:

```
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
```

From here I'm very confused how to continue - I've seen the video recording and read the guide, but it seems to all be related to the other T-Call, the one that has a MicroUSB connection for firmware update. Mine doesn't - mine only has a USB C connection.

Please guide me in the right direction so I'll be able to update the firmware, thanks.

lewisxhe, 2024-11-21:
T-Call requires a USB cable, which is then connected as shown below, and can then be plugged into a computer for upgrades. The rest of the board is the same as other boards.

[image]

Also, what do you mean by frozen? No response?

lewisxhe, 2024-11-25:
If there are no problems during use, do not update the firmware. There is no update instruction. SIMCOM does not provide this.

If the indicator light is off when freezing, it is likely caused by insufficient voltage. You need to test whether the input has reached 5V.

Tsury, 2024-11-25:
Got it, appreciate the help and fast response. I really love your hardware, thank you.

closed 2024-11-25 by Tsury

alexandrevandamme, 2026-11-10:
Hi, two years have passed and I would hope that a firmware update would solve some UARToUSB disconnections.

Are you able to contact SimCom to share their latest firmware update?