Xinyuan-LilyGO/LilyGo-Modem-Series #181: Error 715 doing HTTPS on A7670E-FASE

kimble4, 2024-11-20:
As requested in https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117#issuecomment-2487111863

I've upgraded the firmware, but it's still not working...

QR code reads:
```
P/N:S2-10DAX-Z32C4;SN:MP0624168B95541;IMEI:862771072419073;BTMAC:5C46B0BE1F89;SW:A110B01A7670M7_F
```

Box label reads:
```
XY240706
Q334  T-SIM-A7670E
CHB102F QFN24 [chinese characters] GPS
```

After firmware upgrade:

AT+SIMCOMATI:
```
Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
```

GPS works, but I'm still getting the 715 error with the HtppsBuiltinGet example:

```
Start Sketch
Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT

*ATREADY: 1
AT

OK

AT+CPIN?
AT+CPIN?
+CME ERROR: SIM failure
AT+CPIN?
AT+CPIN?
+CME ERROR: SIM busy

+CPIN: READY
SIM card online
AT+CNMP=2

SMS DONE
AT+CNMP=2

OK
AT+CNMP?

+CGEV: EPS PDN ACT 1
AT+CNMP?

+CNMP: 2

OK
Current network mode : AUTO
Wait for the modem to register with the network.AT+CGREG?
AT+CGREG?

+CGREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?

+CGEV: ME PDN ACT 8,1
AT+CPSI?

+CPSI: LTE,Online,234-20,0x7E5,3426311,398,EUTRAN-BAND20,6175,2,0,17,50,46,6

OK
Inquiring UE system information:LTE,Online,234-20,0x7E5,3426311,398,EUTRAN-BAND20,6175,2,0,17,50,46,6
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR

+IPADDR: 10.178.176.81

OK
Network IP:10.178.176.81
AT+HTTPTERM
AT+CSSLCFG="enableSNI",0,1

OK
AT+HTTPINIT
AT+HTTPTERM

ERROR
Request URL : https://httpbin.org/get
AT+HTTPPARA="URL","https://httpbin.org/get"
AT+HTTPINIT

OK
AT+HTTPACTION=0
AT+HTTPPARA="URL","https://httpbin.org/get"

OK
AT+HTTPACTION=0

OK

PB DONE

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://httpbin.org/get
AT+HTTPPARA="URL","https://httpbin.org/get"

AT+HTTPPARA="URL","https://httpbin.org/get"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://httpbin.org/get
AT+HTTPPARA="URL","https://httpbin.org/get"

AT+HTTPPARA="URL","https://httpbin.org/get"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
-------------------------------------
Request URL : https://vsh.pp.ua/TinyGSM/logo.txt
AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://vsh.pp.ua/TinyGSM/logo.txt
AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://vsh.pp.ua/TinyGSM/logo.txt
AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

AT+HTTPPARA="URL","https://vsh.pp.ua/TinyGSM/logo.txt"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
-------------------------------------
Request URL : https://ipapi.co/timezone
AT+HTTPPARA="URL","https://ipapi.co/timezone"

AT+HTTPPARA="URL","https://ipapi.co/timezone"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://ipapi.co/timezone
AT+HTTPPARA="URL","https://ipapi.co/timezone"

AT+HTTPPARA="URL","https://ipapi.co/timezone"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
Request URL : https://ipapi.co/timezone
AT+HTTPPARA="URL","https://ipapi.co/timezone"

AT+HTTPPARA="URL","https://ipapi.co/timezone"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,715,0
HTTP get failed ! error code = 715
-------------------------------------
Request URL : http://ip-api.com/json/23.158.104.183
AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,302,0
HTTP get failed ! error code = 302
Request URL : http://ip-api.com/json/23.158.104.183
AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,302,0
HTTP get failed ! error code = 302
Request URL : http://ip-api.com/json/23.158.104.183
AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

AT+HTTPPARA="URL","http://ip-api.com/json/23.158.104.183"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,302,0
HTTP get failed ! error code = 302
-------------------------------------
```

kimble4, 2024-11-21:
Well this is embarrassing:  Seems that the issue was down to using the wrong SIM.

Not really expecting anything, I tried the SIM from my phone, which is on a different network (O2, UK) to the one I was trying (Three, UK), and it worked fine.  I then realised that the SIM I had been using is one from a now-expired PAYG account, and not the one with some data credit that I thought it was.  Can confirm that the device works fine with a working Three SIM.

(I assume the HTTP 302 from the expired SIM is an attempt to send you to a "Please top up your credit" page.)

So yeah, it seems that you can get a 715 error because the cellular network is blocking the connection.

lewisxhe, 2024-11-22:
Okay, then problem solved.

closed 2025-01-06 by github-actions[bot]