Xinyuan-LilyGO/LilyGo-Modem-Series #183: MqttsBuiltlnHivemq.ino example code  fails on T-7670G after firmware update

0xMattijs, 2024-11-24:
In order to solve a problem with SNI when using HTTPS I upgraded the firmware for my T-7670G to version B04:

```
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B04SIM767XM5A_M
SIM767XM5_B04V01_241010
IMEI: XXXXXXXXXX
```

This fixed the problem for HTTPS, but now MQTT communication with SSL fails. This problem can be reproduced using the `MqttsBuiltlnHivemq.ino` example.

The `modem.mqtt_set_certificate(HivemqRootCA);` call results in the `AT+CCERTDOWN` command failing:

```
AT+CMQTTSTART
OK

+CMQTTSTART: 0

AT+CCERTDOWN="ca_cert.pem",5555
>
-----BEGIN CERTIFICATE-----
MIIFBzCCA++gAwIBAgISA3nTO8URQRBuVpc/TcIA/pSpMA0GCSqGSIb3DQEBCwUA
MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
EwJSMzAeFw0yMzEwMDQyMzE3NTNaFw0yNDAxMDIyMzE3NTJaMB8xHTAbBgNVBAMM
FCouczIuZXUuaGl2ZW1xLmNsb3VkMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIB
CgKCAQEAxKOySSNPeWJ6cIWbmm1Psc3VmL/YK+K5NZj2GlOJJaMKopb6y0A8V7wy
YM0kaiIe4fGfdLxSKwoFrSXe8gsmaOkgdU734OweOGCPaU3W0ZjIqZzRzhpT8FY9
zj8NlpZpV/QGbbmYCOy6YfAixpZ10L9L1g74CO9M6Guf706Dfg9E1O4N5IwWQr1i
Yv1hvdr1tTHlblj0DcFVu+nYaJ0RIPsiHt3lfldphXGuw1pIEpuv61D121furZqQ
D3onWw1oQ3eUhJYUI/nKqrNSyrYDzvYckdzOnnit/lCuJqAzmFdYvEmyHvmWwS7H
GOEYKiOYjtWkrfK6siEs2pqk/zONaQIDAQABo4ICKDCCAiQwDgYDVR0PAQH/BAQD
AgWgMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjAMBgNVHRMBAf8EAjAA
MB0GA1UdDgQWBBRaojgg6k2PoGxKl7oJ7YO/gDASFjAfBgNVHSMEGDAWgBQULrMX
t1hWy65QCUDmH6+dixTCxjBVBggrBgEFBQcBAQRJMEcwIQYIKwYBBQUHMAGGFWh0
dHA6Ly9yMy5vLmxlbmNyLm9yZzAiBggrBgEFBQcwAoYWaHR0cDovL3IzLmkubGVu
Y3Iub3JnLzAzBgNVHREELDAqghQqLnMyLmV1LmhpdmVtcS5jbG91ZIISczIuZXUu
aGl2ZW1xLmNsb3VkMBMGA1UdIAQMMAowCAYGZ4EMAQIBMIIBAgYKKwYBBAHWeQIE
AgSB8wSB8ADuAHUA2ra/az+1tiKfm8K7XGvocJFxbLtRhIU0vaQ9MEjX+6sAAAGK
/TK26AAABAMARjBEAiAA4Um+6x8PRGrFiXd7GJiKQZl60scTbzB14Bw7SJKr0QIg
KzsFvO7En3vGGIHySjp41rPrqzn1S09rUewDjQGmOA4AdQB2/4g/Crb7lVHCYcz1
h7o0tKTNuyncaEIKn+ZnTFo6dAAAAYr9MrcZAAAEAwBGMEQCIBZo5Q4CRfOVy+Bb
1iCSfXm0rnsu+LWtvYOmNPL/bmCXAiBCvws714wTDLIVSGWO5ufifl1a9eL7WTXC
yrcz/lzl1DANBgkqhkiG9w0BAQsFAAOCAQEAYJD1qbrZ+aZ0uGpFZ0rfdvqAVJwh
X1JyQ+DyTXGQMZQPASvY6wjhhYFt//aRGC2clgzrfJBexICaPRTMV1cfFAYUFQU4
3WEW3JXh7Q2lrFKXVNVv06pzo1aBGQGXJv6vW3Q/DaHGLSLQUPe4YV0slnaYSqPs
ZoBcQGLesyxA3cAJueHqz1+wOrpocqm1IIL7s2JDkP66ZEPG7I7QbwPeaY3Y5zei
osPZ2U8p36XkEFxQeztWyodtVcTmXLzh+syNGy4+8+kwAh3kJ6jbbmg2WXDy4snc
d7tNMgaOAD6LE7e24PUV5A4i49rsU5iQN2M9oUK4l0pethU9HaYC+a+/Aw==
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
MIIFFjCCAv6gAwIBAgIRAJErCErPDBinU/bWLiWnX1owDQYJKoZIhvcNAQELBQAw
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMjAwOTA0MDAwMDAw
WhcNMjUwOTE1MTYwMDAwWjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3Mg
RW5jcnlwdDELMAkGA1UEAxMCUjMwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEK
AoIBAQC7AhUozPaglNMPEuyNVZLD+ILxmaZ6QoinXSaqtSu5xUyxr45r+XXIo9cP
R5QUVTVXjJ6oojkZ9YI8QqlObvU7wy7bjcCwXPNZOOftz2nwWgsbvsCUJCWH+jdx
sxPnHKzhm+/b5DtFUkWWqcFTzjTIUu61ru2P3mBw4qVUq7ZtDpelQDRrK9O8Zutm
NHz6a4uPVymZ+DAXXbpyb/uBxa3Shlg9F8fnCbvxK/eG3MHacV3URuPMrSXBiLxg
Z3Vms/EY96Jc5lP/Ooi2R6X/ExjqmAl3P51T+c8B5fWmcBcUr2Ok/5mzk53cU6cG
/kiFHaFpriV1uxPMUgP17VGhi9sVAgMBAAGjggEIMIIBBDAOBgNVHQ8BAf8EBAMC
AYYwHQYDVR0lBBYwFAYIKwYBBQUHAwIGCCsGAQUFBwMBMBIGA1UdEwEB/wQIMAYB
Af8CAQAwHQYDVR0OBBYEFBQusxe3WFbLrlAJQOYfr52LFMLGMB8GA1UdIwQYMBaA
FHm0WeZ7tuXkAXOACIjIGlj26ZtuMDIGCCsGAQUFBwEBBCYwJDAiBggrBgEFBQcw
AoYWaHR0cDovL3gxLmkubGVuY3Iub3JnLzAnBgNVHR8EIDAeMBygGqAYhhZodHRw
Oi8veDEuYy5sZW5jci5vcmcvMCIGA1UdIAQbMBkwCAYGZ4EMAQIBMA0GCysGAQQB
gt8TAQEBMA0GCSqGSIb3DQEBCwUAA4ICAQCFyk5HPqP3hUSFvNVneLKYY611TR6W
PTNlclQtgaDqw+34IL9fzLdwALduO/ZelN7kIJ+m74uyA+eitRY8kc607TkC53wl
ikfmZW4/RvTZ8M6UK+5UzhK8jCdLuMGYL6KvzXGRSgi3yLgjewQtCPkIVz6D2QQz
CkcheAmCJ8MqyJu5zlzyZMjAvnnAT45tRAxekrsu94sQ4egdRCnbWSDtY7kh+BIm
lJNXoB1lBMEKIq4QDUOXoRgffuDghje1WrG9ML+Hbisq/yFOGwXD9RiX8F6sw6W4
avAuvDszue5L3sz85K+EC4Y/wFVDNvZo4TYXao6Z0f+lQKc0t8DQYzk1OXVu8rp2
yJMC6alLbBfODALZvYH7n7do1AZls4I9d1P4jnkDrQoxB3UqQ9hVl3LEKQ73xF1O
yK5GhDDX8oVfGKF5u+decIsH4YaTw7mP3GFxJSqv3+0lUFJoi5Lc5da149p90Ids
hCExroL1+7mryIkXPeFM5TgO9r0rvZaBFOvV2z0gp35Z0+L4WPlbuEjN/lxPFin+
HlUjr8gRsI3qfJOQFy/9rKIJR0Y/8Omwt/8oTWgy1mdeHmmjk7j1nYsvC9JSQ6Zv
MldlTTKB3zhThV1+XWYp6rjd5JW1zbVWEkLNxE7GJThEUG3szgBVGP7pSWTUTsqX
nLRbwHOoq7hHwg==
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
MIIFYDCCBEigAwIBAgIQQAF3ITfU6UK47naqPGQKtzANBgkqhkiG9w0BAQsFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTIxMDEyMDE5MTQwM1oXDTI0MDkzMDE4MTQwM1ow
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwggIiMA0GCSqGSIb3DQEB
AQUAA4ICDwAwggIKAoICAQCt6CRz9BQ385ueK1coHIe+3LffOJCMbjzmV6B493XC
ov71am72AE8o295ohmxEk7axY/0UEmu/H9LqMZshftEzPLpI9d1537O4/xLxIZpL
wYqGcWlKZmZsj348cL+tKSIG8+TA5oCu4kuPt5l+lAOf00eXfJlII1PoOK5PCm+D
LtFJV4yAdLbaL9A4jXsDcCEbdfIwPPqPrt3aY6vrFk/CjhFLfs8L6P+1dy70sntK
4EwSJQxwjQMpoOFTJOwT2e4ZvxCzSow/iaNhUd6shweU9GNx7C7ib1uYgeGJXDR5
bHbvO5BieebbpJovJsXQEOEO3tkQjhb7t/eo98flAgeYjzYIlefiN5YNNnWe+w5y
sR2bvAP5SQXYgd0FtCrWQemsAXaVCg/Y39W9Eh81LygXbNKYwagJZHduRze6zqxZ
Xmidf3LWicUGQSk+WT7dJvUkyRGnWqNMQB9GoZm1pzpRboY7nn1ypxIFeFntPlF4
FQsDj43QLwWyPntKHEtzBRL8xurgUBN8Q5N0s8p0544fAQjQMNRbcTa0B7rBMDBc
SLeCO5imfWCKoqMpgsy6vYMEG6KDA0Gh1gXxG8K28Kh8hjtGqEgqiNx2mna/H2ql
PRmP6zjzZN7IKw0KKP/32+IVQtQi0Cdd4Xn+GOdwiK1O5tmLOsbdJ1Fu/7xk9TND
TwIDAQABo4IBRjCCAUIwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAQYw
SwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5pZGVudHJ1
c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTEp7Gkeyxx
+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEEAYLfEwEB
ATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2VuY3J5cHQu
b3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0LmNvbS9E
U1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFHm0WeZ7tuXkAXOACIjIGlj26Ztu
MA0GCSqGSIb3DQEBCwUAA4IBAQAKcwBslm7/DlLQrt2M51oGrS+o44+/yQoDFVDC
5WxCu2+b9LRPwkSICHXM6webFGJueN7sJ7o5XPWioW5WlHAQU7G75K/QosMrAdSW
9MUgNTP52GE24HGNtLi1qoJFlcDyqSMo59ahy2cI2qBDLKobkx/J3vWraV0T9VuG
WCLKTVXkcGdtwlfFRjlBz4pYg1htmf5X6DYO8A4jqv2Il9DjXA6USbW1FzXSLr9O
he8Y4IWS6wY7bCkjCWDcRQJMEhg76fsO3txE+FiYruq9RUWhiF1myv4Q6W+CyBFC
Dfvp7OOGAN6dEOM4+qR9sdjoSYKEBpsr6GtPAQw4dy753ec5
-----END CERTIFICATE-----

ERROR
[ 14226][E][TinyGsmMqttA76xx.h:107] mqtt_connect(): [A76XX] Write ca_cert pem failed!
Failed!
```

0xMattijs, 2024-11-24:
I tried downgrading to B03, with the same result. Is it possible that the partition meant for storing the certificates is missing in the firmware package?

lewisxhe, 2024-11-25:
OK, I also encountered the same error, I need SIMCOM to provide a solution to the problem. I will update here if there is any update

lewisxhe, 2024-11-25:
All versions after B02 have issues. I have submitted the log to SIMCOM for analysis.

# 2374B03STM767XM5A_M
[image]

# 2374B02SIM767XM5A_M

[image]

0xMattijs, 2024-11-25:
Is there a download link for the B02 version, so at least I can restore the MQTT feature?

lewisxhe, 2024-11-25:
@0xMattijs https://drive.google.com/file/d/1-63xiw4TbGwHi3rQDzJpPKwp2SDNPXC1/view

0xMattijs, 2024-12-02:
@lewisxhe Is there any update on this issue? Currently there is no firmware available where both MQTT with TLS and HTTPS work simultaneously.

lewisxhe, 2024-12-03:
There is no reply yet. It usually takes a few weeks for simcom to have a patched firmware, so we still need to wait.

lewisxhe, 2024-12-06:
@0xMattijs  [SIM7670G B05](https://drive.google.com/file/d/1-akwKQJttLbtLD48ApagusfBvS9ixZ4F/view?usp=sharing) The firmware has fixed MQTTS, I tested HTTPS and MQTTS and both worked fine.

0xMattijs, 2024-12-10:
@lewisxhe The link you provided above still points to the B04 version.

lewisxhe, 2024-12-10:
https://drive.google.com/file/d/10VvMNzgKhAX25lHSIGHnD3ip7kWjFl0g/view?usp=drivesdk
😂

closed 2024-12-11 by 0xMattijs