Xinyuan-LilyGO/LilyGo-Modem-Series #184: Problem with SMS reception after deleted SMS in memory

toupetlu, 2024-11-26:
Hello

I begin with the use of A7670E R2 not GPS and I use only ATdebug example

I received SMS and the AT Monitor displayed : + SMS FULL.

So I deleted memory with the AT command : AT+CMGD=,1. Since this command, my module didn't receive message. (Phone call are still received)

How to correct this?

Thank you

Ludovic

lewisxhe, 2024-11-26:
You can try to send a text message to see if it can be sent normally.

toupetlu, 2024-11-26:
When I send a text message : "TEST" for example (test realized with 2 differents emmetor phone), nothing happend in the AT serial monitor and the AT command to ask list (AT+CMGL=ALL) displays no new received message.

lewisxhe, 2024-11-26:
When sending a message using AT+CMGS, the modem does not respond with "OK"? No response at all?

toupetlu, 2024-11-26:
The modem respond OK and i did this following example : I wrote the AT command : the message is sent and I received it on my personnal phone

AT+CMGS="336.......57" 
>HELLO LUDO
<ctrl-Z>

toupetlu, 2024-11-26:
I re-try this evening. I resent a text SMS to my A7670e module and I got a AT response at these new message and the all other message I send yesterday (the same number of answer as the same number of sms I sent yesterday)... so, surprising

lewisxhe, 2024-11-27:
So, now you can only send but not receive text messages, right?

toupetlu, 2024-11-27:
No, now I receive text messages (SMS) normally. (but it is surprising that I received all the text messages (I sent the day before) at the same time)

lewisxhe, 2024-11-27:
Well... surprising...

domilui, 2024-12-12:
hello, I have the same problem...

domilui, 2024-12-12:
by resetting the A76xx, the SMS are unlocked...

domilui, 2024-12-12:
it seems that unsubscribing and re-registering on the network unblocks the situation. but unfortunately not all the time. 
so I don't know what to do either...

lewisxhe, 2024-12-13:
@domilui You can try upgrading to the latest firmware

domilui, 2024-12-13:
thanks!  I don't know how to reinstall the firmware... I'll look at your example to do it and will come back to you if I can't do it

closed 2025-01-28 by github-actions[bot]