Xinyuan-LilyGO/LilyGo-Modem-Series #187: T-Call A7670 [H700] 4G compatibility

arneBnansen, 2024-12-03:
Tech spec on website says 2G sim card, but I think that only applies to the SIM800 version, please advise.
Please confirm if the T-Call A7670 [H700] can function for voice calls without a 2G network.

lewisxhe, 2024-12-04:
A7670 supports 4G and 3G. If 4G and 3G are online, calls can be made. However, if there is only 2G network, calls cannot be made without 2G network.

arneBnansen, 2024-12-04:
The case is that 2G network was shut down years ago in my country. From your reply I understand that voice call with the T-Call A7670 should be fine. I would suggest that Liligo update/clarify this on the website and aliexpress store

lewisxhe, 2024-12-05:
> I would suggest that Liligo update/clarify this on the website and aliexpress store

Can you tell me where it is? I can talk to the salesperson.

MrHyrex, 2024-12-05:
Sure I can give you the address to Liligos website: [https://lilygo.cc/products/t-call-v1-4](url). Here only 2G is specifed, but as mention I suspect that is only valid for the SIM800 version. A7670 version has no 4G/2G spec.

Your aliexpress store can be found here [https://www.aliexpress.com/item/1005006477044118.html#nav-specification](url). Here the product title says "...A7670 4G 2G..." but the tech spec does not mention 4G/2G.

lewisxhe, 2024-12-06:
I have given feedback, thank you.

closed 2025-01-24 by github-actions[bot]