Xinyuan-LilyGO/LilyGo-Modem-Series #189: Can we use the speaker connected to T-Call-A7670X for other usages?

Tsury, 2024-12-03:
I soldered a speaker to the SPK+/- and it works great for calls.

When not in a call, I would like to use the speaker for other things, like text-to-speech, etc...

Is it possible? If so, how to achieve it? Thanks.

lewisxhe, 2024-12-04:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/datasheet/A76XX/A76XX_Series_AT_Command_Manual_V1.12.pdf

[image]

Tsury, 2024-12-04:
Thanks, but what if I want to use it for anything else? Play a sound file etc? Is there a possibility?

Tsury, 2024-12-07:
I will check the manual then, thanks.

closed 2024-12-07 by Tsury

Tsury, 2024-12-15:
Thanks @lewisxhe 
Do you think it's possible to also play an audio file? I saw something in the manual but wasn't able to make it work, maybe you could?

Thanks in advanced and appreciate the help!

lewisxhe, 2024-12-16:
It has not been implemented yet. To play an audio file, you need to write the audio into fs before you can play it. I have not implemented this function, but it may be added later.

Tsury, 2024-12-16:
Oh wow, I see.

Well, I would _greatly_ appreciate if this was implemented - it really enhances the usability of the device and will spare me from connecting multiple speakers.

Thanks in any case of course.

DomoMKT, 2025-02-03:
you can easily convert your text into high-quality speech with extremely low latency at [www.nemesyslabs.com](https://www.nemesyslabs.com).

Tsury, 2025-02-08:
Hey, was there any progress with this? Can I somehow help test/develop it? I really need this, thanks!

lewisxhe, 2025-02-08:
I haven't implemented this function yet, but I can find time to test its possibility. First of all, I would like to state that this method will take up the internal space of the modem. There is only 4MB of space inside the A7670, so it cannot play audio files that are too large.

lewisxhe, 2025-02-08:
Already added, for detailed description, please refer to the AT command manual

Tsury, 2025-02-08:
Oh wow thank you so much, going to test it now. Yeah I don't need a lot, just some indications, basic short mono sounds, low quality.

Tsury, 2025-02-09:
@lewisxhe It works amazingly, thank you so much! Going to order 3 more now that I know I can finish the project.